Clon1998/mobileraker_companion #1: Update Manager don't work

GG-Style, 2022-06-04:
The Link for the Update Manager don't work in Mainsail. It tells unknown.
Der Update Manager zeigt den Status unbekannt. Scheint also irgendwas mit dem Code nicht zu passem

Clon1998, 2022-06-04:
Sadly this is broken for months now already ...
I didn't bother trying to fix this issue yet since the companion did not need any updates yet.
However, I will look into this issue.

Clon1998, 2022-06-12:
Fixed. Please update the config file with the new update_manager section!

```
[update_manager mobileraker]
type: git_repo
path: ~/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch:main
managed_services: mobileraker
env: ~/mobileraker-env/bin/python
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install-mobileraker-companion.sh
```

closed 2022-06-12 by Clon1998