Clon1998/mobileraker_companion #5: API Key not working

jangrewe, 2022-07-12:
I'm one of those people that have set `force_logins: True`, but the API key i'm giving to mobileraker_companion doesn't seem to work.
I've tried in `~/klipper_config/Mobileraker.conf` and `~/mobileraker_companion/Mobileraker.conf`, with the key both raw and quoted, yet still i get a 401:

```
Jul 12 23:54:33 voronabox python[12299]: Traceback (most recent call last):
Jul 12 23:54:33 voronabox python[12299]:   File "/home/pi/mobileraker-env/lib/python3.7/site-packages/websockets/legacy/client.py", line 598, in __aiter__
Jul 12 23:54:33 voronabox python[12299]:     async with self as protocol:
Jul 12 23:54:33 voronabox python[12299]:   File "/home/pi/mobileraker-env/lib/python3.7/site-packages/websockets/legacy/client.py", line 633, in __aenter__
Jul 12 23:54:33 voronabox python[12299]:     return await self
Jul 12 23:54:33 voronabox python[12299]:   File "/home/pi/mobileraker-env/lib/python3.7/site-packages/websockets/legacy/client.py", line 650, in __await_impl_timeout__
Jul 12 23:54:33 voronabox python[12299]:     return await asyncio.wait_for(self.__await_impl__(), self.open_timeout)
Jul 12 23:54:33 voronabox python[12299]:   File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
Jul 12 23:54:33 voronabox python[12299]:     return fut.result()
Jul 12 23:54:33 voronabox python[12299]:   File "/home/pi/mobileraker-env/lib/python3.7/site-packages/websockets/legacy/client.py", line 664, in __await_impl__
Jul 12 23:54:33 voronabox python[12299]:     extra_headers=protocol.extra_headers,
Jul 12 23:54:33 voronabox python[12299]:   File "/home/pi/mobileraker-env/lib/python3.7/site-packages/websockets/legacy/client.py", line 328, in handshake
Jul 12 23:54:33 voronabox python[12299]:     raise InvalidStatusCode(status_code, response_headers)
Jul 12 23:54:33 voronabox python[12299]: websockets.exceptions.InvalidStatusCode: server rejected WebSocket connection: HTTP 401
```

Clon1998, 2022-07-13:
Hey,
it should work without putting the API key in quotes.
H
To verify that the config/API key is used can you restart the service and post the new log here?
By default the companion should log the connection tries and if it is using an API key or not.
The logs are located in `tmp/`

jangrewe, 2022-07-13:
Okay, this is strange... i restarted all services involved, and not it seems to be able to connect, but without using the API key. I also got no push notification, yet.

```
2022-07-13 10:04:55,645 [mobileraker.py:info()] - [Voron_2.4] Trying to connect to: ws://127.0.0.1:7125/websocket api key <NO API KEY>
2022-07-13 10:04:55,696 [mobileraker.py:info()] - [Voron_2.4] WebSocket connected
2022-07-13 10:04:55,697 [mobileraker.py:info()] - [Voron_2.4] Fetching printer Objects Try#0
2022-07-13 10:04:55,700 [mobileraker.py:info()] - [Voron_2.4] Received Server Info
2022-07-13 10:04:55,701 [mobileraker.py:info()] - [Voron_2.4] Querying printer Objects
2022-07-13 10:04:55,884 [mobileraker.py:info()] - [Voron_2.4] print_state transition NONE -> standby
2022-07-13 10:04:55,885 [mobileraker.py:info()] - [Voron_2.4] Subscribing to printer Objects
2022-07-13 10:04:55,896 [mobileraker.py:info()] - [Voron_2.4] Sending to firebase fcm (https://mobileraker.eliteschw31n.de): {'printState': 'standby', 'tokens': ['<redacted>'], 'printerIdentifier': 'e43cb7d3-<redacted>'
```

Clon1998, 2022-07-13:
That's weired. Did you setup a trusted client alongside enforcing logins?

Regarding the notifications, you need to open the app once after a restart and you will only receive notifications if mobileraker isn't open. On Android the app can also be force closed while on iOs it is required that you don't force close the app. Sometimes the power saving constraints of iOS still manage to block the notifications.

jangrewe, 2022-07-13:
Yes, 127.0.0.1 is in the trusted clients, but that shouldn't matter because of both `force_logins: True` and an existing user - which, according to the moonraker docs, disables trusted clients?

Regarding the notifications: thanks, that part is working now! <3

jangrewe, 2022-07-13:
Yea, it does look like it, though that's quite unexpected behaviour. Any other clients that are also listed in trusted_clients are indeed required to provide the API key. Maybe it's also a hardcoded behaviour explicitly and exclusively for `127.0.0.1`?
My moonraker version `v0.7.1-597-gd37f91c`.

jangrewe, 2022-07-13:
Well, then something is wrong. I am providing an API key, mobileraker_companion says it is not using it, yet still it works! :-D

Clon1998, 2022-07-13:
> Well, then something is wrong. I am providing an API key, mobileraker_companion says it is not using it, yet still it works! :-D

Can you update the companion?
The logging was setup wrongly. Now it should actually show you that it is using an API key :stuck_out_tongue_winking_eye:

jangrewe, 2022-07-13:
Ah, that would explain it!
Yes, now it works as expected:
```
2022-07-13 12:23:40,258 [mobileraker.py:info()] - [Voron_2.4] Trying to connect to: ws://127.0.0.1:7125/websocket api key <redacted>##########################
```