Clon1998/mobileraker_companion #8: "Update Manger mobileraker: Unable to locate pip executable"

Henk72, 2022-07-20:
It seems the code to past in the moonraker.conf is not working.

I think this env path was not correct: ~/mobileraker-env/bin/python and tried with this valid path: /home/pi/envs/mobileraker.

Next file contains some unreadable content, even though it is a .txt file: requirements: scripts/mobileraker-requirements.txt

Clon1998, 2022-07-20:
Thanks for your hints.
I updated the installation script and it now uses `~/mobileraker-env/bin/python` as it virtual env.
Therefore, no changes to the update manager are required.

Henk72, 2022-07-20:
Wow you are fast!!!!! Amazing

I will try and let you know

Henk72, 2022-07-20:
It is working now, thank you!

Please note that I have 2 parallel Moonraker instances (kiahu installed), but the app seems to need the generic API key, not of the instances. It is working, just wanted to mention how I got it to work.

Issue can be closed.

Clon1998, 2022-07-20:
Awesome!
Thanks for your bug report.

What do you mean by
> The app needs the generic API key, not of the instances.

Can you give me an example 🤔

closed 2022-09-02 by Clon1998