Clon1998/mobileraker_companion #9: Used in conjunction with moonraker-timelapse causes excessive notifications (timelapse pauses on every layer)

adrianocr, 2022-09-20:
If you use this in conjunction with the moonraker-timelapse component you get tons and tons of notifications. Moonraker-timelapse pauses for a split second on every layer at a specific location to take the picture so the print head looks like it is stationary in the resulting timelapse. 

Ideally those notifications could be filtered out if mookraker-timelapse is installed or give users a way to manually filter notifications?

Clon1998, 2022-09-20:
Hey,
thanks for the hint. I am already working on a fix for this. I tried directly adding it to the companion but sadly that does not resolve the issue completely. Therefore I am currently testing changes in the App that give users the option to only enable the state change notification for their configured States.

Clon1998, 2022-09-20:
I just triggered a new beta build for IOS and Android.
It should be available in the next few hours in all stores and regions.
It includes the option to select the states that trigger an notification!

closed 2022-10-12 by Clon1998

LongDono, 2023-10-12:
Is it possible to update the app to ignore the undesirable pause notification while still allowing us to receive valid pause notifications?

https://github.com/mainsail-crew/moonraker-timelapse/issues/42
```The macro provides a Flag called printer[“gcode_macro TIMELAPSE_TAKE_FRAME“].is_paused and the designer of the bot should omit the notifications if that and printer.pause_resume. is_paused are set to true.```

Thanks for the great app!