Clon1998/mobileraker_companion #10: Application error

Varga-Levente, 2022-09-29:
Exception:
 Null check operator used on a null value

#0      PrinterService._updatePrintStat (package:mobileraker/service/moonraker/printer_service.dart:712)
#1      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:416)
#2      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:367)
#3      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:617)
#4      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:486)
#5      PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:1021)
<asynchronous suspension>
#6      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:166)
<asynchronous suspension>

Clon1998, 2022-09-29:
Hey,
this is the GitHub repo of the python companion program. The app is located here https://github.com/Clon1998/mobileraker.
Anyhow, can you provide me with the version number of Mobileraker you are currently using?
Does the error happen permanently or just occasionally?

Varga-Levente, 2022-09-29:
App version 2.0.0-67
This error happens permanently since 4 days.

Clon1998, 2022-09-29:
Can you provide me the content of your browser if you go to `http://<YOUR-PRINTER-IP>/printer/objects/query?print_stats`.

Varga-Levente, 2022-09-29:
{"result": {"status": {"print_stats": {"print_duration": 5769.2903197780015, "total_duration": 6055.732069565001, "filament_used": 4502.69614, "filename": "Queue/CE3_standsmooth.gcode", "state": "printing", "message": ""}}, "eventtime": 14268.93741815}}

Clon1998, 2022-09-29:
Thanks.
I will push a hot fix as soon as possible.

Varga-Levente, 2022-09-29:
Okay. Thank you.

closed 2022-09-29 by Varga-Levente