Clon1998/mobileraker_companion #11: Error message received from WebSocket-Server Namespace 'mobileraker' not found

curather, 2022-10-12:
I installed the mobileraker_companion and received no errors upon install.  When I tried to start the service. This is what the status had.  I did add 
```

mobileraker.service - Companion app to enable push notifications on mobileraker
     Loaded: loaded (/etc/systemd/system/mobileraker.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-10-12 13:44:21 CDT; 2min 59s ago
   Main PID: 1061 (python)
      Tasks: 1 (limit: 9391)
     Memory: 17.1M
        CPU: 492ms
     CGroup: /system.slice/mobileraker.service
             └─1061 /home/pi/mobileraker-env/bin/python /home/pi/mobileraker_companion/mobileraker.py -l /tmp/mobileraker.log

Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] INFO [Ender3V2] Fetching printer Objects Try#0
Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] INFO [Ender3V2] Received Server Info
Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] INFO [Ender3V2] Querying printer Objects
Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] INFO [Ender3V2] print_state transition NONE -> standby
Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] INFO [Ender3V2] Subscribing to printer Objects
Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] WARNING [Ender3V2] Error message received from WebSocket-Server Namespace 'mobileraker' not found
Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] WARNING [Ender3V2] Error message received from WebSocket-Server Namespace 'mobileraker' not found
Oct 12 13:44:21 3dprinter python[1061]: 2022-10-12 13:44:21 3dprinter Client[1061] WARNING [Ender3V2] Could not send to mobileraker-fcm, no printerIdentifier found!
Oct 12 13:44:22 3dprinter python[1061]: 2022-10-12 13:44:22 3dprinter Client[1061] WARNING [Ender3V2] Error message received from WebSocket-Server Namespace 'mobileraker' not found
Oct 12 13:44:22 3dprinter python[1061]: 2022-10-12 13:44:22 3dprinter Client[1061] WARNING [Ender3V2] Error message received from WebSocket-Server Namespace 'mobileraker' not found
```

At the end of my moonraker.conf file.  I added
```
[update_manager mobileraker]
type: git_repo
path: /home/pi/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch:main
managed_services: mobileraker
env: /home/pi/mobileraker-env/bin/python
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install-mobileraker-companion.sh
```

My Mobileraker.conf is

```
[printer Ender3V2]
moonraker_uri: ws://127.0.0.1:7125/websocket
# Define the uri to the moonraker instance.
# Default value: ws://127.0.0.1:7125/websocket
# Optional
moonraker_api_key: 840446b7864f4fc79669e8d41388b1d8
# Moonraker API key if force_logins or trusted clients is active!
# Default value: False
# Optional
```
I went into moonraker and saw that I had the following namespaces:

`{"result": {"namespaces": ["gcode_metadata", "history", "update_manager", "moonraker", "webcams", "announcements", "mainsail"]}}`

I tried to create one of the namespace key tokens to no success. 3dprinter:7125/server/database/item?namespace=mobileraker&key=fcmTokens&value=1
`{"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 654, in _process_http_request\n    result = await self._do_request(args, conn)\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 630, in _do_local_request\n    return await self.callback(\n\n  File \"/home/pi/moonraker/moonraker/components/database.py\", line 732, in _handle_item_request\n    val = await self.get_item(namespace, key)\n\n  File \"/usr/lib/python3.9/concurrent/futures/thread.py\", line 52, in run\n    result = self.fn(*self.args, **self.kwargs)\n\n  File \"/home/pi/moonraker/moonraker/components/database.py\", line 170, in func_wrapper\n    return command_func(*args)\n\n  File \"/home/pi/moonraker/moonraker/components/database.py\", line 333, in _get_impl\n    ns = self._get_record(namespace, key_list[0])\n\n  File \"/home/pi/moonraker/moonraker/components/database.py\", line 642, in _get_record\n    db = self._get_db(namespace)\n\n  File \"/home/pi/moonraker/moonraker/components/database.py\", line 617, in _get_db\n    raise self.server.error(f\"Namespace '{namespace}' not found\", 404)\n\nutils.ServerError: Namespace 'mobileraker' not found\n\n\nThe above exception was the direct cause of the following exception:\n\n\nTraceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1704, in _execute\n    result = await result\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 617, in get\n    await self._process_http_request()\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 656, in _process_http_request\n    raise tornado.web.HTTPError(\n\ntornado.web.HTTPError: HTTP 404: Not Found (Namespace 'mobileraker' not found)\n"}}`

Any help is appreciated.

Clon1998, 2022-10-12:
Hey @curather,
by default, Mobileraker's Android/IOS app creates the namespace and keys the companion requires to work.
Therefore, you will need to add your printer to your mobile app before starting the service.

curather, 2022-10-12:
I can not connect from the mobile app.  This is the error in the moonraker log.

```
2022-10-12 16:16:38,319 [app.py:log_request()] - 201 POST /server/files/upload (172.31.10.179) [_TRUSTED_USER_] 9.43ms
2022-10-12 16:17:37,287 [websockets.py:on_close()] - Websocket Closed: ID: 140262134914208 Close Code: None, Close Reason: None, Pong Time Elapsed: 7.43
2022-10-12 16:17:37,560 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 127.0.0.1
2022-10-12 16:17:37,560 [app.py:log_request()] - 101 GET /websocket (127.0.0.1) [_TRUSTED_USER_] 1.41ms
2022-10-12 16:17:37,560 [websockets.py:open()] - Websocket Opened: ID: 140262134911664, Proxied: False, User Agent: Python/3.9 websockets/10.1, Host Name: 127.0.0.1
2022-10-12 16:17:37,674 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Namespace 'mobileraker' not found
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 253, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 362, in func
    result = await callback(
  File "/home/pi/moonraker/moonraker/components/database.py", line 732, in _handle_item_request
    val = await self.get_item(namespace, key)
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/moonraker/moonraker/components/database.py", line 170, in func_wrapper
    return command_func(*args)
  File "/home/pi/moonraker/moonraker/components/database.py", line 333, in _get_impl
    ns = self._get_record(namespace, key_list[0])
  File "/home/pi/moonraker/moonraker/components/database.py", line 642, in _get_record
    db = self._get_db(namespace)
  File "/home/pi/moonraker/moonraker/components/database.py", line 617, in _get_db
    raise self.server.error(f"Namespace '{namespace}' not found", 404)
```

Clon1998, 2022-10-12:
This isn't a huge problem since the log just shows that the app might have requested an item which did not exist. The app initially needs to check if the namespace and items exist to sync stuff like settings and notification handling IDs. If it does not find anything it should create it. Therefore, just open mobileraker ensure it shows no errors and the remote settings like temperature presets are shown in the printer edit page. Force close it and reopen it again. Afterwards check if the namespace exist by querying the endpoint or by starting the companion again.

curather, 2022-10-12:
I found out what the issue was.  I was entering in the full url address to get to the printer from the app ie http://3dprinter.local
I deleted the printer and readded it.  Then I changed it to 3dprinter.local and it worked.

closed 2022-10-12 by curather