Clon1998/mobileraker_companion #12: Mobileraker.conf cannot appear

nosoymilk, 2022-10-24:
hi i just finish the companion but after i restart klipper or  restart the servise thae Mobileraker.conf is not there 
is there any way to add it mantually

Clon1998, 2022-10-27:
Hey,
AS i stated in the readme.
A config is only required if you want to configure multiple printers.
If that is the case you can create a Mobileraker.conf file in:

~/Mobileraker.conf
<mobileraker_companion DIR>/Mobileraker.conf
~/klipper_config/Mobileraker.conf

Experte9, 2022-12-28:
Sorry to ask stupid questions, but where exactly do I need to create the config?

Experte9, 2022-12-28:
> Hey, AS i stated in the readme. A config is only required if you want to configure multiple printers. If that is the case you can create a Mobileraker.conf file in:
> 
> * ~/Mobileraker.conf
> * <mobileraker_companion DIR>/Mobileraker.conf
> * ~/klipper_config/Mobileraker.conf

Sorry to ask stupid questions, but where exactly do I need to create the config?
I have 2 Printers on my Pi

Clon1998, 2022-12-28:
> 

Just add it to the installation directory of mobileraker_companion

Experte9, 2022-12-28:
Okay, I did this, but I still don't get any notifications.

closed 2022-12-28 by Clon1998