Clon1998/mobileraker_companion #14: Config file location is case sensitive

Clon1998, 2023-01-06:
The config file `Mobileraker.conf` should be case insensitive.
Currently, the parsing for the file only is case insensitive in `~/klipper_config/`.   
It should also allow for both `mobileraker.conf` and `Mobileraker.conf` in the locations:   
- `~/(M|m)obileraker.conf`
- `<mobileraker_companion DIR>/(M|m)obileraker.conf`

dhruvinsh, 2023-01-11:
Is this the cause of not getting any printing notification?

hubertron, 2023-02-09:
Which case is it expecting?

Clon1998, 2023-02-09:
> Is this the cause of not getting any printing notification?

Most likely not.
Please make sure you're using the official build and not any beta build.

Clon1998, 2023-02-09:
> Which case is it expecting?

I think currently it always works if you use a capital M.

voidtrance, 2023-02-16:
Is this fixed? I had an issue with my printer's timezone, which required me to create a config file to manually specify the timezone. My config file is `~/mobileraker_companion/mobileraker.conf` and it's working.

Clon1998, 2023-02-16:
> Is this fixed? I had an issue with my printer's timezone, which required me to create a config file to manually specify the timezone. My config file is `~/mobileraker_companion/mobileraker.conf` and it's working.

Greate!
I think I fixed it but I will leave it open for now to verify it later :)

closed 2023-03-21 by Clon1998