Clon1998/mobileraker_companion #16: Unable to update to v0.2.1-40

devsfan1830, 2023-02-16:
Mainsail shows an update from v0.2.1-12 to v0.2.1-40 but when I run the update script I get the following error:

"Error updating mobileraker_companion: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte"

Previous updates worked fine. This is the first time I've seen this error on an update.

BlackNet, 2023-02-17:
Same error on TWO machines I have.

BlackNet, 2023-02-17:
The only solution I have found is to uninstall mobileraker_companion with the scripts file, delete the directory and reinstall.

tommy536, 2023-02-17:
For me too. How can i uninstall mobileraker?

Clon1998, 2023-02-17:
Hey,
I am not too sure why this error occurs.
There were already multiple users reporting this issue.
To fix it without removing the companion you can run:

`cd ~/mobileraker_companion && git pull`
And than:
`./scripts/install-mobileraker-companion.sh`

After that it should work again!

BlackNet, 2023-02-17:
> Hey, I am not too sure why this error occurs. There were already multiple users reporting this issue. To fix it without removing the companion you can run:
> 
> `cd ~/mobileraker_companion && git pull` And than: `./scripts/install-mobileraker-companion.sh`
> 
> After that, the companion should be updated and I also just verified that the update manager is working again too!

I tried that on my two and that did not work.  Did not see any errors either.

Clon1998, 2023-02-17:
> I tried that on my two and that did not work.  Did not see any errors either.

What's the output after running the two lines?

closed 2023-06-09 by Clon1998