Clon1998/mobileraker_companion #17: can not remove companion

TItan371, 2023-02-19:
i use this script to remove-mobileraker-companion.sh

this is my command
cd ~/
cd mobileraker_companion
./scripts/remove-mobileraker-companion.sh

and it show this
-bash: ./scripts/remove-mobileraker-companion.sh: Permission denied

Clon1998, 2023-02-19:
You will need to either be SU or run the script using sudo:

sudo ./scripts/remove-mobileraker-companion.sh

closed 2023-02-19 by Clon1998

TItan371, 2023-02-21:
many thanks , but still can not remove.

pi@:~ $ cd ~/
pi@:~ $ cd mobileraker_companion
pi@:~/mobileraker_companion $ sudo ./scripts/remove-mobileraker-companion.sh
[sudo] password for pi:
sudo: ./scripts/remove-mobileraker-companion.sh: command not found
pi@:~/mobileraker_companion $ cd scripts/
pi@:~/mobileraker_companion/scripts $ ls
install-mobileraker-companion.sh  mobileraker-requirements.txt  remove-mobileraker-companion.sh