Clon1998/mobileraker_companion #18: many thanks , but still can not remove.

TItan371, 2023-02-24:
many thanks , but still can not remove.

pi@:~ $ cd ~/
pi@:~ $ cd mobileraker_companion
pi@:~/mobileraker_companion $ sudo ./scripts/remove-mobileraker-companion.sh
[sudo] password for pi:
sudo: ./scripts/remove-mobileraker-companion.sh: command not found
pi@:~/mobileraker_companion $ cd scripts/
pi@:~/mobileraker_companion/scripts $ ls
install-mobileraker-companion.sh  mobileraker-requirements.txt  remove-mobileraker-companion.sh

_Originally posted by @TItan371 in https://github.com/Clon1998/mobileraker_companion/issues/17#issuecomment-1438641666_

Clon1998, 2023-02-26:
Hey, 
sorry for my late response. Been busy learning the last few weeks.

I just double-checked the files and I missed setting the executable permission for the script.
Therfore,  you will need to run `chmod +x ~/mobileraker_companion/scripts/remove-mobileraker-companion.sh`,
Afterward, you should be able to run the remove script!

Clon1998, 2023-02-26:
I also fixed that in b9f22b8f1aba302bcd4bb02c2cdbb95befb0e366.

So instead of manually adding the executable permission, you can also just update the companion and then run the script!

closed 2023-02-26 by Clon1998

TItan371, 2023-03-19:
still can not remove!
pi@LittleBOYPi:~ $ chmod +x ~/mobileraker_companion/scripts/remove-mobileraker-companion.sh
pi@LittleBOYPi:~ $ cd/
-bash: cd/: No such file or directory
pi@LittleBOYPi:~ $ cd mobileraker_companion
pi@LittleBOYPi:~/mobileraker_companion $ sudo ./scripts/remove-mobileraker-companion.sh
This script must not run as root
pi@LittleBOYPi:~/mobileraker_companion $ sudo ./scripts/remove-mobileraker-companion.sh
This script must not run as root
pi@LittleBOYPi:~/mobileraker_companion $ cd scripts
pi@LittleBOYPi:~/mobileraker_companion/scripts $ chmod +x ~/mobileraker_companion/scripts/remove-mobileraker-companion.sh
pi@LittleBOYPi:~/mobileraker_companion/scripts $ remove-mobileraker-companion.sh
-bash: remove-mobileraker-companion.sh: command not found
pi@LittleBOYPi:~/mobileraker_companion/scripts $ sudo ./scripts/remove-mobileraker-companion.sh
sudo: ./scripts/remove-mobileraker-companion.sh: command not found
pi@LittleBOYPi:~/mobileraker_companion/scripts $ sudo ./remove-mobileraker-companion.sh
This script must not run as root
pi@LittleBOYPi:~/mobileraker_companion/scripts $

Clon1998, 2023-03-21:
> still can not remove!
>
> ```
> pi@LittleBOYPi:~ $ chmod +x ~/mobileraker_companion/scripts/remove-mobileraker-companion.sh
> pi@LittleBOYPi:~ $ cd/
> -bash: cd/: No such file or directory
> pi@LittleBOYPi:~ $ [...]

Please execute:
```shell
cd mobileraker_companion/scripts/
ls -l | grep remove-mobileraker-companion.sh
```

The result shoud show the permission for the `remove-mobileraker-companion.sh`. Please make sure it has the `x` permission.

[image]

If thats not the case please run
```shell
chmod +x remove-mobileraker-companion.sh
```

Afterwards, you are able to execute the `remove-mobileraker-companion.sh` by calling it directly without the sudo in front:

```shell
./remove-mobileraker-companion.sh 
```
Let me know if that helps or if you need more help.