Clon1998/mobileraker_companion #21: Compatibility with Sonic Pad?

Naemion, 2023-03-07:
I use a Sonic pad with my ender 3 S1 Pro and spent the last two days trying to figure out how to get this installed on the Sonic Pad to no avail. Any advice to getting it to work so that I can get notifications through Mobileraker? Really like the idea and would be really cool to get working.

Clon1998, 2023-03-07:
Hey, 
I will look into that. But as long as you have an ssh connection and root/sudo access you should be able to install the companion as described.

Naemion, 2023-03-08:
That's the thing though. Creality did a great job of unlocking ssh and root access, while also locking it down so you can do absolutely nothing with it.

redlinebikeparts, 2023-03-08:
This might be my issue to. 

The pad says it's installed but it won't do the config file

I'll try manually

Naemion, 2023-03-08:
> 

How did you manage that? I couldn't even get that far.

redlinebikeparts, 2023-03-08:
> > 
> 
> 
> 
> How did you manage that? I couldn't even get that far.

On the pad I went to advanced and then root user and activated that.

Record the password 

Then in terminal on my mac I went 

SSH ROOT@(what ever the ip address is) 

Then password is what ever you recorded from the tab itself

Naemion, 2023-03-08:
> > > 
> > 
> > 
> > How did you manage that? I couldn't even get that far.
> 
> On the pad I went to advanced and then root user and activated that.
> 
> Record the password
> 
> Then in terminal on my mac I went
> 
> SSH ROOT@(what ever the ip address is)
> 
> Then password is what ever you recorded from the tab itself

Yep, I was able to get that far, but could not get anything other than to download the files to work.

Clon1998, 2023-03-15:
Hey, so I had an interesting discussion with Quin of Octoeverywhere and we both ran into the same problem. Basically, Creality locked the pad down, so unless creality does sth. about it, I can not do much now....

Naemion, 2023-03-15:
That's what I was afraid of. I don't get it though, why allow access to root(unlocked with a previous update) if you don't allow root to be used? Makes no sense!

riverratdc, 2023-11-18:
Hi Clon1998,

Any plans to revisit Sonic Pad compatibilty with mobileraker_companion now that OctoEverywhere has released a successful plugin for the Sonic Pad? Recently installed the new OctoEverywhere plugin for the Sonic Pad and have been using it successfully with mobileraker. Would be nice to get the notifications along with everything else in mobileraker, but all attempts to run the install script for mobileraker_companion have failed. If I log in as root it fails because I logged in as root. If I log in as the unprivileged Creality user, the script fails on permission errors on files in /tmp.

Would be happy to test any changes you might make.

Thanks,
Kevin

Clon1998, 2023-11-18:
> Hi Clon1998,
>
>
>
> Any plans to revisit Sonic Pad compatibilty with mobileraker_companion now that OctoEverywhere has released a successful plugin for the Sonic Pad? Recently installed the new [...]

Hey,
I can have another look.
Afaik, Ocotoeverywhere advised users to use a separate PC/Raspberry Pi that connects to the printer that is running on the SonicPad?
This definitely is also a viable solution for mobileraker.

riverratdc, 2023-11-18:
That was the advice of Octoeverywhere until they recently released version 2.10.0 with a native Sonic Pad plugin. I dismissed the advice to use a third device to provide Sonic Pad connectivity out of hand as a not yet ready for prime time suggestion.

riverratdc, 2024-02-11:
Hi Clon1998,

Yesterday, I installed the Sonic Pad compatible mobileraker_companion currently under development in the installer branch to my SP running the latest Creality software. I looked through the install.sh script in the installer branch beforehand and cloned the installer branch from the /usr/share/ directory to avoid the branch being reset to main if the script performed its relocation:
```
root@spad-8876:/usr/share# git clone -b installer https://github.com/Clon1998/mobileraker_companion.git
```
The install script ran successfully to completion without error but put up a warning at the very end regarding the moonraker.asvc file. Here's the console output of the installation:

```root@spad-8876:/usr/share# ./mobileraker_companion/scripts/install.sh

MMMMMMMMMMMMMMMMMMMMMMWNX0xolldk0XWWMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMWNKkdl:;;:;,,',:ldk0XWMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMWNX0xoc;,;:ldxxxxdoc:,'',:lxOKNWWMMMMMMMMMM
MMMMMMMMMWNKOxl:;,;codxkkdooddxxkkdol:;''';coxOXWMMMMMMM
MMMMMMWXOdl:,,:loxkkkdoloodxkOOkxxxxkkxdlc;''',:lkXWMMMM
MMMMMWk:,,,,,:x0OdoloddxxxddxkkOOOOkxddxkOOl''''',cOWMMM
MMMMMWOc,,,,,,:lddxkkxdollllodddxxkO000kkxo:'''''':OWMMM
MMMMWXOkxo:;,,,,,;loddooodxO0KK0Okkkkkdl;,'''',:odkOXWMM
MMMWXo;:ldxxdl:,,',,,:clx0NWWWWNKkoc;,''''';ldxxoc;,lKMM
MMMMNx:,,,;coxkxoc;,,,,,;cokkkdl;,''''',:odxdl:,''':kNMM
MMMWXK0xl:,,,,:lxkkdl:,,,,,,'''''''';ldxxoc,'''';lx0KXWM
MMNkc:ok00koc,,,,;cdkkxoc;,''''',codxdl:,'''';lxOkdc;l0W
MMXo,',,;lxO0kdc;,,,;:oxOkxlccldxxoc,'''',:lxkko:,''',xW
MMWKo:,,,,,;ldO0Odl;,,,,;ldkkxdl;,'''',:oxkxo:,''''';dKW
MMMMNKxl;,,,,,;cdO00xo:,,,,,,,''''',coxkdl;'''''';lkKWMM
MMMMMMWNKxl:,,,,,,:ok00koc;,'''';ldkkdc,'''''';lxKWMMMMM
MMMMMMMMMMNKxo:,,,,,,:ok00Odlcoxkxo:,'''''';lkKNMMMMMMMM
MMMMMMMMMMMMWNKkl:,,,,,,;lxOOkxl:,'''''';lkKWMMMMMMMMMMM
MMMMMMMMMMMMMMMWNKkl:,,,,,,;;,''''''';lkKWMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMWNKko:,,,,'''''';lkKWMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMWNKko:,'',;lkKWMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMWKkddkKWMMMMMMMMMMMMMMMMMMMMMMM

              Mobileraker Companion
Companion for Mobileraker, enabling push notification for Klipper using Moonraker.

Installer Script based on OctoEverywhere for Klipper, courtesy of Quinn Damerell
Mobileraker works as a simple UI for Klipper on the phone.
The companion is required to enable reliable push notifications.

Print State Notifications
Progress Notifications
Custom Notifications via M117 or custom Macro enabling layer based notifications or heat up notifications.
Running in Sonic Pad OS mode
Checking required system packages are installed...
Package python3 (3.7.8-3) installed in root is up to date.
Package python3-pip (20.1.1-1) installed in root is up to date.
Requirement already satisfied: virtualenv in /usr/lib/python3.7/site-packages (20.25.0)
Requirement already satisfied: distlib<1,>=0.3.7 in /usr/lib/python3.7/site-packages (from virtualenv) (0.3.8)
Requirement already satisfied: importlib-metadata>=6.6; python_version < "3.8" in /usr/lib/python3.7/site-packages (from virtualenv) (6.7.0)
Requirement already satisfied: filelock<4,>=3.12.2 in /usr/lib/python3.7/site-packages (from virtualenv) (3.12.2)
Requirement already satisfied: platformdirs<5,>=3.9.1 in /usr/lib/python3.7/site-packages (from virtualenv) (4.0.0)
Requirement already satisfied: zipp>=0.5 in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (3.15.0)
Requirement already satisfied: typing-extensions>=3.6.4; python_version < "3.8" in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (4.7.1)

System package install complete.
Checking Python Virtual Environment For Mobileraker Companion...
No virtual environment found, creating one now.
created virtual environment CPython3.7.8.final.0-64 in 25537ms
creator CPython3Posix(dest=/usr/share/mobileraker-env, clear=False, no_vcs_ignore=False, global=True)
seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/usr/share/.local/share/virtualenv)
added seed packages: pip==23.3.2, setuptools==68.0.0, wheel==0.42.0
activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Updating PIP if needed... (this can take a few seconds or so)
Requirement already satisfied: pip in ./mobileraker-env/lib/python3.7/site-packages (23.3.2)
Collecting pip
Using cached pip-24.0-py3-none-any.whl.metadata (3.6 kB)
Using cached pip-24.0-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
Attempting uninstall: pip
[... 97 lines cut ...]
File "/usr/share/moonraker/moonraker/components/database.py", line 389, in _get_impl
  key_list[1:], ns)
KeyError: 'client'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/usr/share/moonraker/moonraker/common.py", line 541, in execute_method
  result = await callback(params)
File "/usr/share/moonraker/moonraker/websockets.py", line 115, in func
  ip_addr=sc.ip_addr, user=sc.user_info))
File "/usr/share/moonraker/moonraker/components/database.py", line 790, in _handle_item_request
  val = await self.get_item(namespace, key)
File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
  result = self.fn(*self.args, **self.kwargs)
File "/usr/share/moonraker/moonraker/components/database.py", line 221, in func_wrapper
  return command_func(*args)
File "/usr/share/moonraker/moonraker/components/database.py", line 396, in _get_impl
  f"Key '{key}' in namespace '{namespace}' not found", 404)
```

Can this installation be made functional or should I just go away quietly and wait until you are finished? I treat this install as experimental and have factory reset my SP several times to give it a fresh start. Thought you might find the info helpful, anyway

Thanks,
Kevin

Clon1998, 2024-02-11:
> Hi Clon1998,
> 
> Yesterday, I installed the Sonic Pad compatible mobileraker_companion currently under development in the installer branch to my SP running the latest Creality software. I looked through the install.sh script in the installer branch beforehand and cloned the installer branch from the /usr/share/ directory to avoid the branch being reset to main if the script performed its relocation:
> 
> ```
> root@spad-8876:/usr/share# git clone -b installer https://github.com/Clon1998/mobileraker_companion.git
> ```
> 
> The install script ran successfully to completion without error but put up a warning at the very end regarding the moonraker.asvc file. Here's the console output of the installation:
> 
> ```
> 
> MMMMMMMMMMMMMMMMMMMMMMWNX0xolldk0XWWMMMMMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMWNKkdl:;;:;,,',:ldk0XWMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMWNX0xoc;,;:ldxxxxdoc:,'',:lxOKNWWMMMMMMMMMM
> MMMMMMMMMWNKOxl:;,;codxkkdooddxxkkdol:;''';coxOXWMMMMMMM
> MMMMMMWXOdl:,,:loxkkkdoloodxkOOkxxxxkkxdlc;''',:lkXWMMMM
> MMMMMWk:,,,,,:x0OdoloddxxxddxkkOOOOkxddxkOOl''''',cOWMMM
> MMMMMWOc,,,,,,:lddxkkxdollllodddxxkO000kkxo:'''''':OWMMM
> MMMMWXOkxo:;,,,,,;loddooodxO0KK0Okkkkkdl;,'''',:odkOXWMM
> MMMWXo;:ldxxdl:,,',,,:clx0NWWWWNKkoc;,''''';ldxxoc;,lKMM
> MMMMNx:,,,;coxkxoc;,,,,,;cokkkdl;,''''',:odxdl:,''':kNMM
> MMMWXK0xl:,,,,:lxkkdl:,,,,,,'''''''';ldxxoc,'''';lx0KXWM
> MMNkc:ok00koc,,,,;cdkkxoc;,''''',codxdl:,'''';lxOkdc;l0W
> MMXo,',,;lxO0kdc;,,,;:oxOkxlccldxxoc,'''',:lxkko:,''',xW
> MMWKo:,,,,,;ldO0Odl;,,,,;ldkkxdl;,'''',:oxkxo:,''''';dKW
> MMMMNKxl;,,,,,;cdO00xo:,,,,,,,''''',coxkdl;'''''';lkKWMM
> MMMMMMWNKxl:,,,,,,:ok00koc;,'''';ldkkdc,'''''';lxKWMMMMM
> MMMMMMMMMMNKxo:,,,,,,:ok00Odlcoxkxo:,'''''';lkKNMMMMMMMM
> MMMMMMMMMMMMWNKkl:,,,,,,;lxOOkxl:,'''''';lkKWMMMMMMMMMMM
> MMMMMMMMMMMMMMMWNKkl:,,,,,,;;,''''''';lkKWMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMWNKko:,,,,'''''';lkKWMMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMMMMWNKko:,'',;lkKWMMMMMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMMMMMMMMWKkddkKWMMMMMMMMMMMMMMMMMMMMMMM
> 
>               Mobileraker Companion
> Companion for Mobileraker, enabling push notification for Klipper using Moonraker.
> 
> Installer Script based on OctoEverywhere for Klipper, courtesy of Quinn Damerell
> Mobileraker works as a simple UI for Klipper on the phone.
> The companion is required to enable reliable push notifications.
> 
> Print State Notifications
> Progress Notifications
> Custom Notifications via M117 or custom Macro enabling layer based notifications or heat up notifications.
> Running in Sonic Pad OS mode
> Checking required system packages are installed...
> Package python3 (3.7.8-3) installed in root is up to date.
> Package python3-pip (20.1.1-1) installed in root is up to date.
> Requirement already satisfied: virtualenv in /usr/lib/python3.7/site-packages (20.25.0)
> Requirement already satisfied: distlib<1,>=0.3.7 in /usr/lib/python3.7/site-packages (from virtualenv) (0.3.8)
> Requirement already satisfied: importlib-metadata>=6.6; python_version < "3.8" in /usr/lib/python3.7/site-packages (from virtualenv) (6.7.0)
> Requirement already satisfied: filelock<4,>=3.12.2 in /usr/lib/python3.7/site-packages (from virtualenv) (3.12.2)
> Requirement already satisfied: platformdirs<5,>=3.9.1 in /usr/lib/python3.7/site-packages (from virtualenv) (4.0.0)
> Requirement already satisfied: zipp>=0.5 in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (3.15.0)
> Requirement already satisfied: typing-extensions>=3.6.4; python_version < "3.8" in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (4.7.1)
> 
> System package install complete.
> Checking Python Virtual Environment For Mobileraker Companion...
> No virtual environment found, creating one now.
> created virtual environment CPython3.7.8.final.0-64 in 25537ms
> creator CPython3Posix(dest=/usr/share/mobileraker-env, clear=False, no_vcs_ignore=False, global=True)
> seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/usr/share/.local/share/virtualenv)
> added seed packages: pip==23.3.2, setuptools==68.0.0, wheel==0.42.0
> activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
> Updating PIP if needed... (this can take a few seconds or so)
> Requirement already satisfied: pip in ./mobileraker-env/lib/python3.7/site-packages (23.3.2)
> Collecting pip
> Using cached pip-24.0-py3-none-any.whl.metadata (3.6 kB)
> Using cached pip-24.0-py3-none-any.whl (2.1 MB)
[... 108 lines cut ...]
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/usr/share/moonraker/moonraker/common.py", line 541, in execute_method
>   result = await callback(params)
> File "/usr/share/moonraker/moonraker/websockets.py", line 115, in func
>   ip_addr=sc.ip_addr, user=sc.user_info))
> File "/usr/share/moonraker/moonraker/components/database.py", line 790, in _handle_item_request
>   val = await self.get_item(namespace, key)
> File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
>   result = self.fn(*self.args, **self.kwargs)
> File "/usr/share/moonraker/moonraker/components/database.py", line 221, in func_wrapper
>   return command_func(*args)
> File "/usr/share/moonraker/moonraker/components/database.py", line 396, in _get_impl
[... cut ...]
> Can this installation be made functional or should I just go away quietly and wait until you are finished? I treat this install as experimental and have factory reset my SP several times to give it a fresh start. Thought you might find the info helpful, anyway
> 
> Thanks, Kevin

Hey,
thanks for the feedback! I still do not possess a sonic pad so this is good to know. You definitely did the right thing with the `asvc ` file. I guess I need some more logic for the sonic pad here...

Regarding the error. This is not caused by the installer and rather by the application itself. Could you provide me the entire companion's log file?

Clon1998, 2024-02-11:
@riverratdc,
i just realized that the error you posted is not from the companions log file but rather from the moonraker log.

It is normal for the companion to run before the app has had a chance to set up the necessary database entries, which may result in an error being logged by moonraker. However, the companion should be able to handle this error and ignore it. Once the app on your phone has connected to the sonic pad, the database entry should be created and and the companion is able to send notifications.

riverratdc, 2024-02-11:
mobileraker on my phone still shows "Companion not found" under notifications after many program launches and restarts of the Sonic Pad since yesterday. Mobileraker works fine through octoeverywhere on my phone.

I'd be happy to send the companion log, where is it? I didn't see it while browsing the SP file system.

Clon1998, 2024-02-11:
> mobileraker on my phone still shows "Companion not found" under notifications after many program launches and restarts of the Sonic Pad since yesterday. Mobileraker works fine through octoeverywhere [...]

By default it should be located at the same location as the rest. The installer logs mention `Logs: /mnt/UDISK/printer_logs`.

riverratdc, 2024-02-11:
No companion log in ```/mnt/UDISK/printer_logs```
```
klippy.log
moonraker.log
octoeverywhere.log
qt.log
```
What's the filename? I can search the filesystem and see if it ended up elsewhere.

Clon1998, 2024-02-11:
> No companion log in `/mnt/UDISK/printer_logs`
> 
> ```
> klippy.log
> moonraker.log
> octoeverywhere.log
> qt.log
> ```
> 
> What's the filename? I can search the filesystem and see if it ended up elsewhere.

Mhhhh....  Guess more stuff to debug lol.

The file should be named `mobileraker.log`

riverratdc, 2024-02-11:
A search of the SP filesystem for mobileraker* yields:
```
/etc/init.d/mobileraker_service
/mnt/UDISK/printer_config/mobileraker.conf
/overlay/upper/etc/init.d/mobileraker_service
/overlay/upper/usr/share/mobileraker_companion/assets/Mobileraker-System.png
/overlay/upper/usr/share/mobileraker_companion/assets/Mobileraker-System_witthbg.png
/overlay/upper/usr/share/mobileraker_companion/mobileraker/client/mobileraker_fcm_client.py
/overlay/upper/usr/share/mobileraker_companion/mobileraker/mobileraker_companion.py
/overlay/upper/usr/share/mobileraker_companion/mobileraker.py
/overlay/upper/usr/share/mobileraker_companion/scripts/mobileraker-requirements.txt
/usr/share/mobileraker_companion/assets/Mobileraker-System.png
/usr/share/mobileraker_companion/assets/Mobileraker-System_witthbg.png
/usr/share/mobileraker_companion/mobileraker/client/mobileraker_fcm_client.py
/usr/share/mobileraker_companion/mobileraker/mobileraker_companion.py
/usr/share/mobileraker_companion/mobileraker.py
/usr/share/mobileraker_companion/scripts/mobileraker-requirements.txt
```
I can stop and start the mobileraker_service, but have never seen any indication of it running in the process list.

Clon1998, 2024-02-11:
> A search of the SP filesystem for mobileraker* yields:
>
> ```
> /etc/init.d/mobileraker_service
> /mnt/UDISK/printer_config/mobileraker.conf
> /overlay/upper/etc/init.d/mobileraker_service [...]

That might indicate that it is not running.
You could check the content of the `/overlay/upper/etc/init.d/mobileraker_service` file as this should also state the log location. If there is no log yet, I assume the service never started.

riverratdc, 2024-02-11:
Log file location as you said. Service not running.
```#!/bin/sh /etc/rc.common
# Copyright (C) 2006-2011 OpenWrt.org

START=80
STOP=1
DEPEND=moonraker_service
USE_PROCD=1
OOM_ADJ=-17

start_service() {
    procd_open_instance
    procd_set_param env HOME=/root
    procd_set_param env PYTHONPATH=/usr/share/mobileraker_companion
    procd_set_param oom_adj $OOM_ADJ
    procd_set_param command /usr/share/mobileraker-env/bin/python3 /usr/share/mobileraker_companion/mobileraker.py -l /mnt/UDISK/printer_logs -c /mnt/UDISK/printer_config/mobileraker.conf
    procd_close_instance
}
```
FWIW the contents of the mobileraker.conf file look to be correct also.

Clon1998, 2024-02-11:
> Log file location as you said. Service not running.
> 
> ```
> # Copyright (C) 2006-2011 OpenWrt.org
> 
> START=80
> STOP=1
> DEPEND=moonraker_service
> USE_PROCD=1
> OOM_ADJ=-17
> 
> start_service() {
>     procd_open_instance
>     procd_set_param env HOME=/root
>     procd_set_param env PYTHONPATH=/usr/share/mobileraker_companion
>     procd_set_param oom_adj $OOM_ADJ
>     procd_set_param command /usr/share/mobileraker-env/bin/python3 /usr/share/mobileraker_companion/mobileraker.py -l /mnt/UDISK/printer_logs -c /mnt/UDISK/printer_config/mobileraker.conf
>     procd_close_instance
> }
> ```
> 
> FWIW the contents of the mobileraker.conf file look to be correct also.

I guess in that case there is something wrong with the service file. I am not an expert on the OS the SP uses and I only replicated the content of Octoeverywhere. So there is a chance that the file itself contains an error.

riverratdc, 2024-02-11:
Thanks for all your replies. I've worked with Openwrt a lot over the years by installing it on every wireless router I've owned in the past ten years. Coming from a background of bash-shell based  linux distros, I bumped my head on the Openwrt ash shell (a little above Bourne, but way short of Bash) many times.

Let me know if I can supply any additional info or you would like anything tested going forward.

A happy Mobileraker Supporter,
Kevin

Clon1998, 2024-02-11:
> Thanks for all your replies. I've worked with Openwrt a lot over the years by installing it on every wireless router I've owned in the past ten years. Coming from a background of bash-shell based [...]

Well I am not sure why the service is not starting.
You could try to run the python script manually just to test if its the actual service file or the python script that is causing problems.

To run just the python script `/usr/share/mobileraker-env/bin/python3 /usr/share/mobileraker_companion/mobileraker.py -l /mnt/UDISK/printer_logs -c /mnt/UDISK/printer_config/mobileraker.conf` should be a good starting point.

riverratdc, 2024-02-11:
```
root@spad-8876:~# /usr/share/mobileraker-env/bin/python3 /usr/share/mobileraker_companion/mobileraker.py -l /mnt/UDISK/printer_logs -c /mnt/UDISK/printer_co
nfig/mobileraker.conf
Traceback (most recent call last):
  File "/usr/share/mobileraker_companion/mobileraker.py", line 7, in <module>
    from mobileraker.client.snapshot_client import SnapshotClient
  File "/usr/share/mobileraker_companion/mobileraker/client/snapshot_client.py", line 3, in <module>
    from PIL import Image
  File "/usr/share/mobileraker-env/lib/python3.7/site-packages/PIL/Image.py", line 103, in <module>
    from . import _imaging as core
ImportError: cannot import name '_imaging' from 'PIL' (/usr/share/mobileraker-env/lib/python3.7/site-packages/PIL/__init__.py)
```

Clon1998, 2024-02-11:
> ```
> root@spad-8876:~# /usr/share/mobileraker-env/bin/python3 /usr/share/mobileraker_companion/mobileraker.py -l /mnt/UDISK/printer_logs -c /mnt/UDISK/printer_co
> nfig/mobileraker.conf
> Traceback (most [...]

Thanks. Seems like some dependencies for pillow are missing. I will have a look!

riverratdc, 2024-02-12:
I changed line 3 in snapshot_client.py from
```
from PIL import Image
```
to
```
import PIL
```
and mobilerkaer_companion became fully opertaional. Notifications in Mobileraker are working fine now. Importing all of PIL is probably an inefficient solution, but it was my starting point in troubleshooting the issue. I'll leave it like this until you merge the new installer into main.

Clon1998, 2024-02-13:
> It seems to be somehow pillow version related. The default version of PIL on the Sonic Pad current release is version 8.3.2 in
> 
> ```
> /usr/lib/python3.7/site-packages/PIL/
> ```
> 
> If I launch the python interpreter with this version of python/PIL
> 
> ```
> from PIL import Image
> ```
> 
> Image imports cleanly. If I activate the mobileraker venv and launch the interpreter with pillow 9.5.0, the import fails with the error referenced in the posts above. I may remove 9.5.0 pillow from the venv and force an install of 8.3.2 to see the effect.

Interesting lol....
Thanks for the feedback! I will probably move to the raw import `import PIL` as this seems to resolve the version issue too?

Clon1998, 2024-02-13:
Weired.
I have tried running the companion on PIL `8.3.2, 9.5.0, 10.2.0` on my machine without any problems. 
However, I am currently using python 3.9 instead of the sonic pad's py3.7.

riverratdc, 2024-02-13:
It is wierd. I can't find any reason that Pillow 9.5.0 shouldn't work on this system with python 3.7.8. When I changed to ```import PIL``` in snapshot_client.py I also changed line 41 from
```
image = Image.open(BytesIO(res.content)).convert("RGB")
```
to
```
image = PIL.Image.open(BytesIO(res.content)).convert("RGB")
```
With these changes, the snapshot client still fails but logs a different message:
```
2024-02-13 09:27:35,120 mobileraker.cam INFO - Trying to take a snapshot from URI: http://127.0.0.1/webcam/?action=snapshot
2024-02-13 09:27:35,249 mobileraker.cam ERROR - Error while trying to process image request: module 'PIL' has no attribute 'Image'
```
After ```import PIL``` in the interpreter in the Mobileraker venv, ```help(PIL)``` shows ```Image``` available among the package contents. At this point I'm left suspecting that something ran amok during the original Pillow install into the venv and that is why I'm considering reinstalls or downgrades of Pillow in the venv as a next step? Notifications in Mobileraker are still working fine without the snapshot.

Clon1998, 2024-02-13:
> It is wierd. I can't find any reason that Pillow 9.5.0 shouldn't work on this system with python 3.7.8. When I changed to `import PIL` in snapshot_client.py I also changed line 41 from
>
> ```
> image = [...]

Tbh. I am not an Python expert. I am often surprised that I managed to get the companion at the point it is right now lol.
Id love to throw out PIL as it is only used to mirror/rotate the image and it creates a TON of trouble. But so far I had no time to find a better solution for that...

riverratdc, 2024-02-13:
You should be proud of both Mobileraker and the companion. They are both excellent pieces of software. As a Sonic Pad user, I appreciate both your and Quinn's efforts to expand the capabilities of your products to include the SP and K1. Thanks!

I'm feeling like my problem with the companion is a 'one-off', but it has become man vs. machine at this point :) I'll let you know what I end up doing.

riverratdc, 2024-02-14:
I found a solution, though it wasn't the one I was hoping for.

After reverting snapshot_client.py back to its original state, I uninstalled Pillow 9.5.0 from the Mobileraker venv. This forced the companion to use the globally installed Pillow 8.3.2 in /usr/lib and snapshots began to work successfully with Mobileraker notifications. I then reinstalled Pillow 9.5.0 into the Mobileraker venv and this again broke companion snapshots. I again uninstalled Pillow 9.5.0 from the venv and snapshot functionality was restored.

Not sure what to make of it. The Pillow website shows Pillow 9.5.0 and Python 3.7.8 as a valid pairing. Unfortunately it does not appear to be the case with Openwrt and the Sonic Pad. The Openwrt package repository contains all versions of Pillow up through the latest 10.x versions seemingly indicating support for it.

I'm going to leave the companion as it is using the global Pillow 8.3.2. When I get a little more time I'll experiment with different versions of Pillow in the venv because I'm curious. Thanks for all your help.

Clon1998, 2024-02-20:
> I'd be happy to. I'll do it tomorrow (got a long print job running on the SP) and report the results.
> 
> 
> 
> Should I clone main or the installer branch as I did initially?

Thanks! It is still on the installer branch. I might merge it later this week once I get more feedback from a couple testers.

riverratdc, 2024-02-21:
> 
> I updated the script and changed the requirement to just include `pillow` could you uninstall and retry the installer?
> 
> (To uninstall, run the installer script with `-uninstall`)

Unpinning ```Pillow``` from version 9.5.0 in ```mobileraker-requirements.txt``` did the trick (though I did capitalize ```Pillow``` in the file before running the install). Activating the mobileraker venv and checking the locally installed packages afterward shows no ```Pillow``` in the venv, meaning the requirement was satisfied by the ```Pillow 8.3.2``` in root. The companion with snapshot support is working fine.

The only remaining issue I saw during this install was the same asv file warning at the end of the install. Same situation as the first time, I manually added mobileraker to the active asvc file:
``` 
Adding service to moonraker asv file...
Moonraker asv file did not exist, created it and added service to it.
By default, moonraker should already provide this file, so this is unexpected.

                 Happy Printing!
                        <3

/mnt/UDISK/.moonraker_database/moonraker.asvc

klipper_mcu
webcamd
MoonCord
KlipperScreen
moonraker-telegram-bot
moonraker-obico
sonar
crowsnest
octoeverywhere
ratos-configurator

/mnt/UDISK/mobileraker_data/moonraker.asvc

mobileraker
```
Fix this issue and the Sonic Pad install of the Mobileraker-companion is good to go.

PR #77 (2024-02-24): Using new installer
Using the Installer by OE.
This ensures that the companion can be installed on K1 and SonicPad.
It also simplifies the installation for multi-printer setups.

Closes #68, Closes #48, Closes #21
closed 2024-02-24 by Clon1998