Clon1998/mobileraker_companion #22: Endless Log entry

FreezeManny, 2023-03-13:
The mobileraker companion on my Mainsail install writes almost endless log entries.
I only noticed, once my 32 gb SD Card was filled with 24gb of logs saying the same thing:
[image]
It keeps on entering this in my log.

Clon1998, 2023-03-13:
Hey,
thats definitely not intended.
Can you stop the service, delete the log file. Start it again wait a few seconds, stop it again and attach the newly created log file?

jangrewe, 2023-03-18:
I have the same issue, multiple times already. Not sure what causes it, but restarting Mobileraker "fixes" it ... until the next time this happens. ;-)

Clon1998, 2023-03-24:
Should be fixed with [8c2df47](https://github.com/Clon1998/mobileraker_companion/commit/8c2df4760d147287b5ebf0a3e9159777e89104e3)

closed 2023-03-24 by Clon1998

jangrewe, 2023-03-25:
Thanks, a quick test with restarting Klipper and adjacent services confirms that this is indeed fixed! 🥳

FreezeManny, 2023-03-27:
On my end it is not fixed. It does the exact same thing as before on a fresh install.

Clon1998, 2023-03-27:
> On my end it is not fixed. It does the exact same thing as before on a fresh install.

Alright, can you stop the service and send the first few log entries ?