Clon1998/mobileraker_companion #24: Support for authorization header

mnestor, 2023-03-19:
I have my instance behind authentication so I can view it when not at home. It would be great if the app supported an authorization header so that I could use the app when not at home as well since the mobile web is kinda painful having to reload everything everytime.

Just being able to set:
Authorization: Bearer XYZ

Or even just a bit more of being able to add any other headers that might be desired would open up a lot more abilities.

Clon1998, 2023-03-26:
Hey are you talking about the Ios/Android app or the python plugin for klipper/moonraker?

mnestor, 2023-03-26:
And that would be why I thought I was losing my mind. It looks like I put it in the wrong repo. 

The iOS/Android app. I've gotten things to work sort of by adding basic auth but that's just kinda nasty.

Clon1998, 2023-03-26:
> And that would be why I thought I was losing my mind. It looks like I put it in the wrong repo.
> 
> The iOS/Android app. I've gotten things to work sort of by adding basic auth but that's just kinda nasty.

Alright,
is there any reason to why you add an additional authentication layer? AFAIK moonraker already handles that for you.
Or did you expose the printer to the public internet and therefore added additional authentication ?

mnestor, 2023-03-26:
Exposed to the internet. Doing more of a zero trust setup instead of a vpn since my workplace blocks vpns.

Clon1998, 2023-03-26:
> Exposed to the internet. Doing more of a zero trust setup instead of a vpn since my workplace blocks vpns.
> 
> If I knew anything about app dev I'd happily work on a PR but all my experience is server and web based.

I see.
I already implemented custom header support for Octoeverywhere. Therefore, all that's missing is a UI element that allows entering the stuff manually. I will look into that, can you do me a favor and reopen an issue on the mobileraker repo instead of this one?

mnestor, 2023-03-26:
Closing as wrong repo. Sorry about that!

closed 2023-03-26 by mnestor