Clon1998/mobileraker_companion #25: Unable to receive notifications

frncistic, 2023-04-15:
[image]

Clon1998, 2023-04-15:
Hey,
by looking at the images, I assume you are located in China?
It seems like the GFW is blocking the connection to my servers. I will look into options in order to resolve this issue for you, as I recently got another user reporting problems while trying to open the app itself (Which I assume is also caused by the GFW  as I am relying on Google's firebase for push notifications).

frncistic, 2023-04-15:
yeah， so i  use  vpn    I try to change the network settings. Thank you

frncistic, 2023-04-18:
[image]
[image]

It will still cause the same problem

frncistic, 2023-04-18:
[image]

closed 2023-08-26 by Clon1998