Clon1998/mobileraker_companion #27: Extremely high CPU usage caused print failure

thatbluflame, 2023-05-12:
So, my 6 hour print just failed and come top find out that mobileraker's companion was taking 83 percent of my pi's memory and caused it to crash twice. Memory leak going on here? I noticed it never even sent any notifications to my phone even though i have it setup for a 50 percent notif and to tell me when a print is going to start.
[image]

Clon1998, 2023-05-12:
Are you running the latest version of Mobileraker companion or an earlier one?

thatbluflame, 2023-05-12:
> Are you running the latest version of Mobileraker companion or an earlier one? There was an bug in version 0.2.1 and below that caused the exact behavior.
> 
> If you are on the latest version please share the log, if it's to big just the last 100-200 lines so I can have a look at it.

I unfortunately nuked the folder for mobileraker in an attempt to get it off as quick as possible to finish what I was trying to print, but yes it was the latest version, just installed it last week.

Clon1998, 2023-05-12:
Ohh damn...
That definitely sux. I am sorry for you, next time you can also just shut down the mobileraker service.

I sadly without any more information I probably won't be able to reproduce this nor fix it.
I am not a python dev, so the probability that I have a memory leak is there.

mpaw, 2023-06-04:
@Clon1998 This just happened to me. Mobileraker companion killed a print as it used up all available memory. I'm using a Raspberry Pi 3 B+ with 1GB RAM, mobileraker companion was using 587MB

[image]

Here is the error from `klippy.log`
```
Receive: 97 23538.175008 23537.885451 14: seq: 10, analog_in_state oid=15 next_clock=1921514368 value=2900
Receive: 98 23538.396709 23538.392594 11: seq: 12, clock clock=1916357178
Receive: 99 23538.403040 23538.397230 12: seq: 13, shutdown clock=1916683819 static_string_id=Timer too close
Transition to shutdown state: MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Dumping gcode input 0 blocks
Dumping 20 requests for client 548174257792
```
And here is the mobileraker.log file
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/11646801/mobileraker.log)

mpaw, 2023-06-04:
I just installed the companion from this github repo yesterday, so I'm assuming I have the latest version.

thatbluflame, 2023-06-04:
> @Clon1998 This just happened to me. Mobileraker companion killed a print as it used up all available memory. I'm using a Raspberry Pi 3 B+ with 1GB RAM, mobileraker companion was using 587MB [...]

Yea that klippy error is the same one I got

Clon1998, 2023-06-05:
> @Clon1998 This just happened to me. Mobileraker companion killed a print as it used up all available memory. I'm using a Raspberry Pi 3 B+ with 1GB RAM, mobileraker companion was using 587MB [...]
I'm sorry to hear that you're having some trouble. I took a look at the logs. It seems like the companion consistently reported a print-error state. Can you give me some background on when the crash occurred? Was it at `2023-06-04 02:56:12,114` or at `2023-06-04 17:47:47,500`? These are the only two entries where the moonraker connection was terminated. Also, since you recently installed the app, may I ask if you installed it while a print was running or not?

mpaw, 2023-06-05:
@Clon1998 It's ok, memory leaks happen 😁 . The crash happened at `2023-06-04 02:56:12,114`. I installed it on a fresh boot of the raspberry Pi with no jobs running, and I restarted the raspberry pi again after it was done installing. I did receive a notification in Mobileraker when the print (that crashed) started but then nothing after that. Let me know if you need any other logs from me.

Clon1998, 2023-06-05:
> 

Alright. Are there by any chance any additional logs e.g. `mobileraker.1.log`?

Clon1998, 2023-06-05:
I just committed [584ac52](https://github.com/Clon1998/mobileraker_companion/commit/584ac526dc614b6c6db06058bfdafb0d2b325da5). It included a fix for an edge case that might have triggered to many evaluation calls if the printer was not in printing state but updated progress or other print_stats

xldyelx, 2023-07-27:
I too am having this issue I think

619 pi        20   0  728100 496480   3160 S   0.0  57.8  31:34.28 python

pi@raspberrypi:~ $ ps -ef | grep 619
pi         619     1  1 Jul25 ?        00:31:34 /home/pi/mobileraker-env/bin/python /home/pi/mobileraker_companion/mobileraker.py -l /tmp/mobileraker.log
pi       15400 15273  0 22:08 pts/0    00:00:00 grep --color=auto 619

Clon1998, 2023-07-27:
> I too am having this issue I think
>
>
>
>
>
> 619 pi        20   0  728100 496480   3160 S   0.0  57.8  31:34.28 python
>
>
>
>
>
> pi@raspberrypi:~ $ ps -ef | grep 619
>
> pi         619     1  1 Jul25 ? [...]

Could you please share the mobileraker log file with me?
For you it should still be located in /tmp/mobileraker.log

Clon1998, 2023-08-26:
v0.4.0 should have fixed this issue.
However, I will keep this issue open just in case it is still present.

Please let me know if the companion drains system resources again!

xldyelx, 2023-08-26:
I reinstalled it and it fixed it thank you so much!

woheedev, 2023-09-09:
Just experienced this as well, looks like some sort of memory leak that caused my print to fail at around 90% total ram usage as reported on mainsail and showed 60-65% mem usage in top when I checked after the print failed.

Heres a snippet of the klippy log:
`Receive: 82 72661.945596 72661.941488 18: seq: 12, tmcuart_response oid=2 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 83 72662.024620 72661.995697 14: seq: 13, analog_in_state oid=24 next_clock=2965394432 value=1660
Receive: 84 72662.073616 72662.068332 18: seq: 14, tmcuart_response oid=3 read=b'\n\xfa\xef-\x80&\x02\x08 \x8d'
Receive: 85 72662.082678 72662.078455 18: seq: 15, tmcuart_response oid=0 read=b'\n\xfa\xef-\x802\x02\x08 \x9f'
Receive: 86 72662.089199 72662.084329 18: seq: 16, tmcuart_response oid=1 read=b'\n\xfa\xef-\x802\x02\x08 \x9f'
Receive: 87 72662.095616 72662.091213 18: seq: 17, tmcuart_response oid=4 read=b'\n\xfa\xef-\x80\x1e\x02\x08 \xd5'
Receive: 88 72662.102134 72662.096917 18: seq: 18, tmcuart_response oid=3 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 89 72662.108645 72662.103493 18: seq: 19, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 90 72662.115154 72662.109843 18: seq: 1a, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 91 72662.121695 72662.116487 18: seq: 1b, tmcuart_response oid=4 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 92 72662.243878 72662.217527 15: seq: 1e, analog_in_state oid=16 next_clock=2991794432 value=23525
Receive: 93 72662.324535 72662.262923 14: seq: 11, analog_in_state oid=24 next_clock=3001394432 value=1659
Receive: 94 72662.449765 72662.448552 11: seq: 13, clock clock=2981308964
Receive: 95 72662.543667 72662.448552 15: seq: 13, analog_in_state oid=16 next_clock=3027794432 value=23527
Receive: 96 72662.623737 72662.448552 14: seq: 13, analog_in_state oid=24 next_clock=3037394432 value=1663
Receive: 97 72662.843884 72662.448552 15: seq: 13, analog_in_state oid=16 next_clock=3063794432 value=23524
Receive: 98 72662.924022 72662.448552 14: seq: 13, analog_in_state oid=24 next_clock=3073394432 value=1662
Receive: 99 72663.036573 72663.035411 12: seq: 14, shutdown clock=3051738802 static_string_id=Timer too close
Stats 72665.2: gcodein=0  mcu: mcu_awake=0.016 mcu_task_avg=0.000018 mcu_task_stddev=0.000020 bytes_write=34802954 bytes_read=11437461 bytes_retransmit=9 bytes_invalid=0 send_seq=716905 receive_seq=716905 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=16 upcoming_bytes=0 freq=119996747 rpi: mcu_awake=0.000 mcu_task_avg=0.000008 mcu_task_stddev=0.000007 bytes_write=291646 bytes_read=899779 bytes_retransmit=0 bytes_invalid=0 send_seq=48581 receive_seq=48581 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999839 adj=50001222 heater_bed: target=80 temp=79.9 pwm=0.723 raspberry_pi: temp=41.3 sd_pos=3237016 sysload=0.92 cputime=3217.745 memavail=55720 print_time=47739.650 buffer_time=2.089 print_stall=8 extruder: target=245 temp=244.9 pwm=0.708
Transition to shutdown state: MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Dumping gcode input 0 blocks
Dumping 20 requests for client 1964885688
Received 71949.165465: b'{"id": 1719196784, "method": "info", "params": {}}'
Received 71949.166257: b'{"id": 1719194408, "method": "objects/list", "params": {}}'
Received 71949.166257: b'{"id": 1719194888, "method": "gcode/help", "params": {}}'
Received 71949.325681: b'{"id": 1719195992, "method": "info", "params": {}}'
Received 71949.326620: b'{"id": 1719194120, "method": "objects/list", "params": {}}'
Received 71949.326620: b'{"id": 1719195416, "method": "gcode/help", "params": {}}'`

and I attached the mobileraker log.
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/12566685/mobileraker.log)

Clon1998, 2023-09-09:
> 

Thats a bummer...
I am still trying to understand what is going on. Thanks for providing your log files

Clon1998, 2023-09-09:
> Just experienced this as well, looks like some sort of memory leak that caused my print to fail at around 90% total ram usage as reported on mainsail and showed 60-65% mem usage in top when I checked after the print failed.
> 
> [mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/12566685/mobileraker.log)
> 
> Snippet of the klippy log:
> 
> ```
> Receive: 83 72662.024620 72661.995697 14: seq: 13, analog_in_state oid=24 next_clock=2965394432 value=1660
> Receive: 84 72662.073616 72662.068332 18: seq: 14, tmcuart_response oid=3 read=b'\n\xfa\xef-\x80&\x02\x08 \x8d'
> Receive: 85 72662.082678 72662.078455 18: seq: 15, tmcuart_response oid=0 read=b'\n\xfa\xef-\x802\x02\x08 \x9f'
> Receive: 86 72662.089199 72662.084329 18: seq: 16, tmcuart_response oid=1 read=b'\n\xfa\xef-\x802\x02\x08 \x9f'
> Receive: 87 72662.095616 72662.091213 18: seq: 17, tmcuart_response oid=4 read=b'\n\xfa\xef-\x80\x1e\x02\x08 \xd5'
> Receive: 88 72662.102134 72662.096917 18: seq: 18, tmcuart_response oid=3 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
> Receive: 89 72662.108645 72662.103493 18: seq: 19, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
> Receive: 90 72662.115154 72662.109843 18: seq: 1a, tmcuart_response oid=1 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
> Receive: 91 72662.121695 72662.116487 18: seq: 1b, tmcuart_response oid=4 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
> Receive: 92 72662.243878 72662.217527 15: seq: 1e, analog_in_state oid=16 next_clock=2991794432 value=23525
> Receive: 93 72662.324535 72662.262923 14: seq: 11, analog_in_state oid=24 next_clock=3001394432 value=1659
> Receive: 94 72662.449765 72662.448552 11: seq: 13, clock clock=2981308964
> Receive: 95 72662.543667 72662.448552 15: seq: 13, analog_in_state oid=16 next_clock=3027794432 value=23527
> Receive: 96 72662.623737 72662.448552 14: seq: 13, analog_in_state oid=24 next_clock=3037394432 value=1663
> Receive: 97 72662.843884 72662.448552 15: seq: 13, analog_in_state oid=16 next_clock=3063794432 value=23524
> Receive: 98 72662.924022 72662.448552 14: seq: 13, analog_in_state oid=24 next_clock=3073394432 value=1662
> Receive: 99 72663.036573 72663.035411 12: seq: 14, shutdown clock=3051738802 static_string_id=Timer too close
> Stats 72665.2: gcodein=0  mcu: mcu_awake=0.016 mcu_task_avg=0.000018 mcu_task_stddev=0.000020 bytes_write=34802954 bytes_read=11437461 bytes_retransmit=9 bytes_invalid=0 send_seq=716905 receive_seq=716905 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=16 upcoming_bytes=0 freq=119996747 rpi: mcu_awake=0.000 mcu_task_avg=0.000008 mcu_task_stddev=0.000007 bytes_write=291646 bytes_read=899779 bytes_retransmit=0 bytes_invalid=0 send_seq=48581 receive_seq=48581 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999839 adj=50001222 heater_bed: target=80 temp=79.9 pwm=0.723 raspberry_pi: temp=41.3 sd_pos=3237016 sysload=0.92 cputime=3217.745 memavail=55720 print_time=47739.650 buffer_time=2.089 print_stall=8 extruder: target=245 temp=244.9 pwm=0.708
> Transition to shutdown state: MCU 'mcu' shutdown: Timer too close
> This often indicates the host computer is overloaded. Check
> for other processes consuming excessive CPU time, high swap
> usage, disk errors, overheating, unstable voltage, or
> similar system problems on the host computer.
> Dumping gcode input 0 blocks
> Dumping 20 requests for client 1964885688
> Received 71949.165465: b'{"id": 1719196784, "method": "info", "params": {}}'
> Received 71949.166257: b'{"id": 1719194408, "method": "objects/list", "params": {}}'
> Received 71949.166257: b'{"id": 1719194888, "method": "gcode/help", "params": {}}'
> Received 71949.325681: b'{"id": 1719195992, "method": "info", "params": {}}'
> Received 71949.326620: b'{"id": 1719194120, "method": "objects/list", "params": {}}'
> Received 71949.326620: b'{"id": 1719195416, "method": "gcode/help", "params": {}}'```
> ```

Could you provide me a rough timeline of your printjob?

In the logs I spotted quit a big time frame without any logs:

```
2023-09-08 21:40:29,853 mobileraker._Default INFO - Snapshot passed threshold. LastSnap: PrinterSnapshot(klippy_ready=True, print_state=printing, m117=Waiting for bed temp 80c..., m117_hash=ea7076c6a7dec4c787a89fad343ef30a14eba293ef6586703f73a50461fe04fe, virtual_sdcard=VirtualSDCard (progress: 0.010999, file_position: 11914), print_stats=PrintStats (filename: Hotends-Seam-Test-Hollow_PETG.gcode, total_duration: 436.37985363499956, print_duration: 0.0, state: printing, message: ), current_file=Filename: Hotends-Seam-Test-Hollow_PETG.gcode, Modified: 1694226788.5921977, Size: 1083157, toolhead=<mobileraker.data.dtos.moonraker.printer_objects.Toolhead object at 0x756e1580>, gcode_move=<mobileraker.data.dtos.moonraker.printer_objects.GCodeMove object at 0x75562bc8>, gcode_response=3.1 /80.0 T0:26.6 /0.0, gcode_response_hash=aeb25a69f75b10ef78014da0a6f436dc625b81ec46a1f6de3e80935dd53517bb), NewSnap: PrinterSnapshot(klippy_ready=True, print_state=printing, m117=Waiting for bed temp 80c..., m117_hash=ea7076c6a7dec4c787a89fad343ef30a14eba293ef6586703f73a50461fe04fe, virtual_sdcard=VirtualSDCard (progress: 0.010999, file_position: 11914), print_stats=PrintStats (filename: Hotends-Seam-Test-Hollow_PETG.gcode, total_duration: 438.38594780099993, print_duration: 0.0, state: printing, message: ), current_file=Filename: Hotends-Seam-Test-Hollow_PETG.gcode, Modified: 1694226788.5921977, Size: 1083157, toolhead=<mobileraker.data.dtos.moonraker.printer_objects.Toolhead object at 0x75553b50>, gcode_move=<mobileraker.data.dtos.moonraker.printer_objects.GCodeMove object at 0x75562bc8>, gcode_response=3.2 /80.0 T0:26.6 /0.0, gcode_response_hash=b761a8f54d9e1eae51440b57ebae8335d4348fb27fbd51c13434c2a975cc7a2c)
2023-09-09 02:06:02,621 mobileraker.sync INFO - Moonraker's connection to Klippy has terminated/disconnected
```

Clon1998, 2023-09-09:
I think I know what is going on.
If a network call gets dropped I did not include any timeout in the code.
That could cause the program to wait for it indefinitely causing new messages to queue up/fill up the memory because the Python program was waiting for the network request.

Therefore, I made the following adjustments:
1. Added a timeout to the network calls that should ensure they are canceled if not answered within 10 sec.
2. Limited the maximum time a notification evaluation task (The task that is triggered whenever moonraker updates a value) can live to 60 sec. If it is not completed within 60 sec it will get deleted.

Still, I will keep that issue open in case there is still another leak going on. But I am confident I resolved the major one 🤞

Clon1998, 2023-12-05:
After 3 months I did not receive any negative feedback in regards to this issue after my fixes.
I therefore will close this issue as fixed.

closed 2023-12-05 by Clon1998