Clon1998/mobileraker_companion #30: The container fails to run with an error

Maeians, 2023-06-02:
#Traceback (most recent call last):
?  File "/opt/mobileraker/mobileraker.py", line 14, in <module>
/    from snapshot_client import SnapshotClient
B  File "/opt/mobileraker/snapshot_client.py", line 3, in <module>
    from PIL import Image
T  File "/opt/venv/lib/python3.11/site-packages/PIL/Image.py", line 103, in <module>
#    from . import _imaging as core
VImportError: libjpeg.so.62: cannot open shared object file: No such file or directory
#Traceback (most recent call last):
?  File "/opt/mobileraker/mobileraker.py", line 14, in <module>
/    from snapshot_client import SnapshotClient
B  File "/opt/mobileraker/snapshot_client.py", line 3, in <module>
    from PIL import Image
T  File "/opt/venv/lib/python3.11/site-packages/PIL/Image.py", line 103, in <module>
#    from . import _imaging as core
VImportError: libjpeg.so.62: cannot open shared object file: No such file or directory

Clon1998, 2023-06-02:
@mkuf can you comment on that?
I think there is a Linux package that is missing in the image?

mkuf, 2023-06-02:
Yes, the libjpeg62-turbo is missing in the runtime image. 
I created a PR for this: https://github.com/Clon1998/mobileraker_companion/pull/31

@Clon1998 Do you have a list of dependencies to linux packages that we should add to the runtime image?  
I used the install script for reference when I created the dockerfile, its probably a good idea to also add this dependency to it.

Clon1998, 2023-06-02:
I don't, I did not even know that the `Pillow` lib requires that!
Do you mind adding the package to the install script as well in your PR?

Clon1998, 2023-06-02:
@Maeians a new image of the container should be available [shortly](https://github.com/Clon1998/mobileraker_companion/actions/runs/5159307723).

closed 2023-06-02 by Clon1998

Maeians, 2023-06-03:
> @Maeians容器的新图像应该[很快就会](https://github.com/Clon1998/mobileraker_companion/actions/runs/5159307723)可用。

ok thank you very much looking forward to happy printing