Clon1998/mobileraker_companion #33: Re-downloaded from github, but I got container running error again

Maeians, 2023-06-03:
#Traceback (most recent call last):
?  File "/opt/mobileraker/mobileraker.py", line 14, in <module>
/    from snapshot_client import SnapshotClient
B  File "/opt/mobileraker/snapshot_client.py", line 3, in <module>
    from PIL import Image
T  File "/opt/venv/lib/python3.11/site-packages/PIL/Image.py", line 103, in <module>
#    from . import _imaging as core
XImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
#Traceback (most recent call last):
?  File "/opt/mobileraker/mobileraker.py", line 14, in <module>
/    from snapshot_client import SnapshotClient
B  File "/opt/mobileraker/snapshot_client.py", line 3, in <module>
    from PIL import Image
T  File "/opt/venv/lib/python3.11/site-packages/PIL/Image.py", line 103, in <module>
#    from . import _imaging as core
XImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

mkuf, 2023-06-03:
I think we need to add some more libraries to get pillow to work. 
Is there a way for me to test the image processing of the companion without having the "supporter" Version of the app @Clon1998?

Looking at the build requirements for pillow, the list of external libraries is quite extensive and I'd refrain from installing **all** the corresponding runtime libraries to not bloat the resulting image.   
https://pillow.readthedocs.io/en/stable/installation.html#external-libraries

Clon1998, 2023-06-03:
@mkuf Sure!

I can grant you the supporter status manually, however, that would require the anonymous user ID of your device. I don't have a reliable way of extracting that since I want to move to user accounts via either Apple, google/another SSO.
But I include it in the logs that you can find at the bottom of the App's settings. To extract it that way:
1. Force Close the app
2. Open it again
3. go to the settings and open debug logs
4. Copy the logs 
5. Search for `$RCAnonymousID:` since that's the prefix. An typical ID would look like `
$RCAnonymousID:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx`
6. Dm it to me.

Also feel free to reach out to me via discord: Pad#3489

mkuf, 2023-06-03:
Thanks for the access @Clon1998!

I can't seem to replicate the described behaviour. 
What architecture/computer are you running on @Maeians?
Please run `docker system info` on your machine and share the output here. 

It does seem that on `amd64` pillow does come with the necessary libraries, so I suspect that this is specific to any of the arm architectures. This does also explain why I can't replicate it in my dev setup. 

```
mobileraker@3e8178a4d32d:/opt$ find / -iname *libopenjp*
/opt/venv/lib/python3.11/site-packages/Pillow.libs/libopenjp2-78c47f58.so.2.5.0
```

Maeians, 2023-06-04:
@mkuf Ok, I call Docker related information, as shown below

Last login: Sat Jun  3 20:07:00 2023 from 192.168.100.146
root@onecloud:~# docker system info
Client: Docker Engine - Community
 Version:    24.0.2
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.10.5
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.18.1
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 4
  Running: 2
  Paused: 0
  Stopped: 2
 Images: 4
 Server Version: 24.0.2
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 3dce8eb055cbb6872793272b4f20ed16117344f8
 runc version: v1.1.7-0-g860f061
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 6.1.11-meson
 Operating System: Debian GNU/Linux 11 (bullseye)
 OSType: linux
 Architecture: armv7l
 CPUs: 4
 Total Memory: 990.5MiB
 Name: onecloud
 ID: 5YTZ:WFSQ:K27X:YYCT:C2C4:G3NG:CT4G:6ST4:OT6I:MSJS:M6KH:7CW3
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Registry Mirrors:
  https://rnvca2sh.mirror.aliyuncs.com/
 Live Restore Enabled: false

Therefore, my Armbian system version is : Welcome to Armbian 23.02.2 Bullseye with bleeding edge Linux 6.1.11-meson

[image]

This firmware comes from the armbian project address of Wankeyun: https://github.com/hzyitc/armbian-onecloud
S805 chip
So Docker has been successfully installed here, and qinglong, kodexplorer, mainsail, etc. in other containers are running perfectly. Unfortunately, mobileker_companion cannot run in the container.

[image]
[image]

mkuf, 2023-06-04:
This is only affecting the `armv7l` architecture as far as I can see.  
On my `amd64` and `aarch64` machines the issue is not occuring.  

https://github.com/Clon1998/mobileraker_companion/pull/35 adds the required packages for running on `armv7l`

Maeians, 2023-06-05:
@mkuf @Clon1998 Unfortunately, the container error is still displayed
When I reinstalled, the error date is as follows
[image]

I used this machine as a backup server at the time, and it could only run containers and the like. I didn’t install the kiauh family plug-in, and Mainsail using Docker alone, but the armv7l you mentioned will only affect the architecture. I can consider replacing the amd64 and aarch64 machines to run mobileker_companion, thanks you work hard

But the last sentence is wrong because of the Docker command syntax "Run the Companion in Docker"
What should be correctly commanded is
docker run -d \
--name mobileraker_companion \
-v "$(pwd)/mobileraker.conf:/opt/printer_data/config/mobileraker.conf" \
ghcr.io/clon1998/mobileraker_companion:latest

Maybe you can consider using the armv7l architecture, it runs perfectly in the container and kiauh, you can consider please try it

mkuf, 2023-06-05:
The PR I mentioned has not been merged yet, so you'll have to wait for that before trying again :)

closed 2023-06-05 by Clon1998

Maeians, 2023-06-05:
@mkuf ok thanks, i'll try it later＼⁠(⁠°⁠o⁠°⁠)⁠／

Maeians, 2023-06-29:
> @Maeians is it working now?

Not for the time being, I am still working on compatibility with all python dependent libraries related to armv7l, due to the artificial docker kernel, I am still working on it