Clon1998/mobileraker_companion #37: libjpeg62-turbo on Ubuntu 22.04 is not available

lettore, 2023-06-10:
Hello,
I tried to install the companion on my printer running on Ubuntu but I had to change the name of a package from libjpeg62-turbo to libjpeg62 because it wasn't found in any repositories.
Not totally sure if everything will work, I receive the notification through.

Clon1998, 2023-06-10:
Hey,
I think the package name is dependent on the Linux distro you are using. 
If you are receiving notifications now, everything should be working. Keep in mind that image notifications are currently only available for supporters.

Antunes43, 2023-06-14:
hey there, I'm running Armbian 23.02.2 Jammy with Linux 5.15.93-sunxi and I'm having the same problem, cannot finish installing of companion app 'cause of that, the app keeps telling me to install it, even without being able to install this damn package

Clon1998, 2023-06-15:
@mkuf I think we need to further look i to the package installation

Antunes43, 2023-06-15:
Well, i just removed it from the installation dependencies install part and got it up and running without problems, but, i'm not a supporter, so i'm not using the thing it enables

mkuf, 2023-06-17:
> @mkuf I think we need to further look i to the package installation

You'd probably want some flow control in that script that checks the OS and uses the correct installer to install a specific set of packages for that System.

```bash
function install_packages {
  source /etc/os-release
  case ${ID} in
    arch)
      pacman -S package1 package2-foo
    ;;
    debian)
      apt update
      apt install package1 package2-turbo
    ;;
    ubuntu)
      apt update
      apt install package1 package2-bar
    ;;
    rhel|fedora|centos)
      yum install package1 package2
    ;;
    *)
      echo "Unsupported Operating System detected, exiting."
      exit 1
    ;; 
}

sudo install_packages
```

alternatively, you may create install scripts for each platform,[like klipper does](https://github.com/Klipper3d/klipper/tree/master/scripts). 

Or, to save yourself the hassle just use the docker image. But I might be a bit biased in that regard 😄

Clon1998, 2023-06-17:
> You'd probably want some flow control in that script that checks the OS and uses the correct installer to install a specific set of packages for that System.
>
>
>
> ```bash
>
> function install_packages [...]

Thanks!
I already asked ChatGPT to generate me some script snippets to achieve exactly that. I am just unsure if it's sufficient to install Zlib and libjpeg-turbo in order to use Pillow.

I will probably push the changes tomorrow and test the installation script on different containers 😊

simpat1zq, 2023-06-22:
Hi. I was just wondering if this is fixed or not. I was getting this error last week and I just want to know if I should try again. I need this because I want to install this on my Home Assistant server to be the central point for all the printers.

Clon1998, 2023-06-22:
> Hi. I was just wondering if this is fixed or not. I was getting this error last week and I just want to know if I should try again. I need this because I want to install this on my Home Assistant server to be the central point for all the printers. 

Hey,
I am a bit the last few days. I think i got a fix ready but I want to test it on different distros before publishing it to make sure it works!

Clon1998, 2023-06-23:
@simpat1zq @Antunes43 @lettore,
I just committed the new install script. Please let me know if it is working now!

simpat1zq, 2023-06-23:
Also, I just tried it out on my Ubuntu box, and it seems to work just fine there. So this seems to be something with Debian.