Clon1998/mobileraker_companion #38: Custom Notifications

fuzelet, 2023-06-22:
I searched around issues and did not see this posted anywhere.  I am trying to use custom notifications, which seem to work fine from the console, etc. however I am NOT able to get them to work when they are in a gcode file.  For example, when I send the following command in the console, I get a notification on my phone.  If I put the same line in a gcode file, I do not get the notification.  I do get other notifications like status changed, progress %, etc. though fine.  Its just an issue with custom notifications in gcode files i am having issues with.

M117 $MR$:$printer_name Printer Status|Layer change. Progress $progress ETA $a_eta

Do custom notifications have any certain requirements in the actual gcode file to work?  I am using the latest android version on the play store, even updated it this morning with no change. (v2.4.0 - 6/22/23)

mobileraker
v0.4.0-15-g584ac52 

M82 ;absolute extrusion mode
START_PRINT BED_TEMP=65 EXTRUDER_TEMP=230
;Nozzle diameter = 0.4
;Filament type = PLA
;Filament name = PLA
;Filament weight = [4.31984818359375]
;M190 S65
;M109 S230
M117 $MR$:$printer_name Printer Status|Layer change. Progress $progress. ETA: $a_eta
M117 Printing
G92 E0
G92 E0
G10
;LAYER_COUNT:17
SET_PRINT_STATS_INFO TOTAL_LAYER=17

;LAYER:0
SET_PRINT_STATS_INFO CURRENT_LAYER=1
M117 $MR$:$printer_name Printer Status|Layer change. Progress $progress. ETA: $a_eta
M117 Layer 1 / 17 
M107
G1 F300 Z0.4
G0 F7500 X110.041 Y110.084 Z0.4
;TYPE:SKIRT
G1 F300 Z0.2

Clon1998, 2023-07-27:
Hey 👋,
I am SO sorry for my late reply somehow all of the recent issues for the companion went under the radar.

In case the issue still persits for you, could you please upload a Gcode file for me to have a look at?

fuzelet, 2023-07-30:
Will do. Right in the middle of printing some back to back parts for my wife that don't contain the code but I'll get something uploaded here shortly.

fuzelet, 2023-07-31:
[Untitled.zip](https://github.com/Clon1998/mobileraker_companion/files/12217255/Untitled.zip)

I am essentially trying to use custom notifications to alert me when layer changes happen for really long prints that do not really align to the %s i am able to select within the application(time estimates, layer counts, etc).  If there is a better way to alert of layer change using the application, please let me know.

I did not get any of the notifications in the gcode file as it progressed through the layers. I am able to send them from the console fine and get a notification every time.

[image]

[image]

[image]

Clon1998, 2023-07-31:
Thanks for that,
Based on my analysis, it seems that the issue with your GCode file is due to the presence of two `M117` lines. To resolve this, kindly delete the second `M117` and attempt the test again. It's worth noting that when I tested a GCode file with only one `M117` notification line, it worked as expected.

[image]

fuzelet, 2023-07-31:
Appreciate the response. I can confirm notifications from the gcode work when i remove the double M117.

That being said, removing the second M117 and only using the Custom Nofitication M117 breaks the layer information on my LCD Screen.

[image]

I was under the assumption two M17 messages with the same arguments caused issues, not that you can't have two M117s. 

https://github.com/Clon1998/mobileraker_companion/blob/main/docs/Custom_Notifications.md

fuzelet, 2023-07-31:
I am currently testing using SET_DISPLAY_TEXT MSG= to set the LCD Screen instead of M117. I will report back if that ends up working with the Custom Notifications as well.

Clon1998, 2023-08-01:
> Appreciate the response. I can confirm notifications from the gcode work when i remove the double M117.
> 
> 
> 
> That being said, removing the second M117 and only using the Custom Notification M117 line breaks the layer information on my LCD Screen.
> 
> 
> 
> [image]
> 
> 
> 
> I was under the assumption two M17 messages with the same arguments caused issues, not that you can't have two M117s. 
> 
> 
> 
> https://github.com/Clon1998/mobileraker_companion/blob/main/docs/Custom_Notifications.md
> 
> 
> 
> Is it possible to change the custom notification from M117 to something less used?

I also thought having two m117 would not cause any Problem. However, it seems like Klipper Just skips the first message in case there are two.

fuzelet, 2023-08-01:
So I have not been successful yet with using the klipper display text.  Here is my current GCODE file and my gcode macro in klipper.  The screen updates fine, however custom notifications don't work now.

[Untitled.zip](https://github.com/Clon1998/mobileraker_companion/files/12231833/Untitled.zip)

[gcode_macro DISPLAY_PRINT_TIME]
gcode:
  {% set cur_layer = printer.print_stats.info.current_layer|int %}
  {% set tot_layer = printer.print_stats.info.total_layer|int %}
  SET_DISPLAY_TEXT MSG="Layer {cur_layer} / {tot_layer}"
  
  
[image]
[image]

I am not sure what else to try to get my screen and custom notifications to work at the same time.

fuzelet, 2023-08-01:
I ended up flipping the section in the GCODE and got the notifications to work again, however now my screen is back to being messed up.

DISPLAY_PRINT_TIME
M117 $MR$:$printer_name|Layer changed. $progress $eta

So definitely something going on with multiple M117 commands. Not sure if its cause they are back to back, some "timeframe" that has to pass, etc

fuzelet, 2023-08-01:
Not sure what else to try at this point. Even this way the notification does not work.  I have yet to add dwell commands between the commands as I have no idea if that will do anything other than just add delays, etc.  DISPLAY_PRINT_TIME was renamed to DISPLAY_LAYER_INFO

[Untitled.zip](https://github.com/Clon1998/mobileraker_companion/files/12232748/Untitled.zip)
[image]

Clon1998, 2023-08-01:
> Not sure what else to try at this point. Even this way the notification does not work. I have yet to add dwell commands between the commands as I have no idea if that will do anything other than [...]

Interesting and also confusing...
Sadly, I don't have access to my own printer until the end of September ( I know... I am crazy enough to keep working on an app without access to a machine lol).

I will confirm with some other users if they have the same problem regarding custom notifications in GCode files since you are the first one that raises the issue.

Could you do me a favor and try this for me:
1. Prepare a 4-10 min print/GCode that includes only the `m117` for the custom notifications for the layer change 
2. Delete any existing `mobileraker.log` file
3. Restart the mobileraker companion
4. start the print, wait until its done
5. Provide me the entire `mobileraker.log`

Afterward do the same with a Gcode File that includes both `M117` lines, the one for the custom notifications and the one that you want to use for your LCD.

I'm hoping that the log will provide me with additional clues as to what's happening here.

Also, I still owe you some background as to why I am using `M117` for custom notifications. Basically, there are two ways to implement such a system. There are two methods for this type of system. The first is a pull-based approach, which I am not using. This method involves the companion asking the printer for updates at regular intervals, such as every second. The second method, which I am using, is a push-based approach. With this method, the companion subscribes for updates from the printer. Whenever an update is available, the printer itself notifies the companion.

However, this does not solve the problem of injecting custom data you as a user can provide to the companion. To achieve that I opted to use the M117 messages because the companion is notified whenever this value changes. However, it could very well be that Klipper ignores multiple `M117` codes and only uses the last one it reads until another

fuzelet, 2023-08-01:
Will do. I am not able to locate any mobileraker.log file on my klipper box.  Do I need to enable something first in order to generate a log file?

Clon1998, 2023-08-01:
> Will do. I am not able to locate any mobileraker.log file on my klipper box. Do I need to enable something first in order to generate a log file?

There are multiple locations the logs might be written to.
1. `~/printer_data/logs/`
2. `/tmp/`
3. `Location defined during the installation of the companion via MOBILERAKER_LOG_PATH`

Clon1998, 2023-08-01:
> > 
> 
> I was not able to find any mobileraker.log using the "locate" command on my debian klipper box.

Ohh wait, you are running mobileraker in a container do you?

fuzelet, 2023-08-01:
> > >
> 
> Ohh wait, you are running mobileraker in a container do you?

I am on debian and used KIAUH to get everything up and running.

Clon1998, 2023-08-02:
> > > >
> 
> > 
> 
> > Ohh wait, you are running mobileraker in a container do you?
> 
> 
> 
> I am on debian and used KIAUH to get everything up and running. I have a printer_data folder, various venv folders, etc.  I don't see any logs, etc in those locations.
> 
> 

Alright, I need to test the kihau installation once more but I thought I ensured the mobileraker log is written to the default log location of moonraker, Klipper, Mainsail...

fuzelet, 2023-08-02:
> Alright, I need to test the kihau installation once more but I thought I ensured the mobileraker log is written to the default log location of moonraker, Klipper, Mainsail...

I went ahead and completely uninstalled and reinstalled everything with KIAUH and I now have a log file for mobileraker in my printer folder.  Either having multiple printers before was causing an issue with logs being created OR since i have had KIAUH around for a while and while upgrading, adding new features, etc something got messed up.  I will go ahead and see about getting this tested for you and report back.

fuzelet, 2023-08-02:
Attached are both tests that i just ran today. I did not get any custom notifications for either of these prints.

[logs.zip](https://github.com/Clon1998/mobileraker_companion/files/12243854/logs.zip)
[gcode.zip](https://github.com/Clon1998/mobileraker_companion/files/12243856/gcode.zip)

[image]
[image]

Clon1998, 2023-08-03:
Thanks for your efforts!
I will have a look at them hopefully during the weekend.

Clon1998, 2023-08-03:
Nvm I already looked at the logs.
So I was right, klipper/moonraker skips multiple M117. The companion only received the last one.
Therefore, there is very little I can do to resolve your issue. However, you might want to have a look at klipper's [delayed_gcodes](https://www.klipper3d.org/Command_Templates.html#delayed-gcodes). It should enabled you to send the 2nd M117 after a delay of e.g. 2 sec.

fuzelet, 2023-08-04:
Appreciate the constant feedback and assistance on this. I'll look into the delayed gcode stuff and see what that provides. Thanks again! I'll go ahead and close this as it seems to be unrelated to mobileraker and is actually a klipper/mainsail limitation. Given your experience and expertise, please let me know if you feel I should bring this up to the attention of the Klipper or Mainsail folks.

closed 2023-08-04 by fuzelet