Clon1998/mobileraker_companion #40: Multiple Printer Same Computer

solo2424, 2023-06-26:
I have two printer on the same PC, They are separeated by port (7125 and 7126). I changed the websocket to the corect port for each printer and they show fine. But the issue I am having is the webcams. by default, it is pulling the url from the ws, which is adding the port.  ex. **http://192.168.0.93:7125/webcam/?action=stream** and **http://192.168.0.93:7126/webcam/?action=stream** but the url to my printers are as follow **http://192.168.0.93/webcam/?action=stream** and **http://192.168.0.93/webcam2/?action=stream**. 
Below is my mobileraker.comf (changed to txt) aLthough I question if the companion app is even working properly. Any assistance on how to change the detault camera path so it does not add thw ws port?

[mobileraker.txt.log](https://github.com/Clon1998/mobileraker_companion/files/11867493/mobileraker.txt.log)

Clon1998, 2023-06-26:
Hey,
are you talking about the webcam in the app or the screenshots that are taken via the companion and included in the notifications?

In case the latter is the case, could you make sure to add the port to the URI of the snapshot URL:

```config
[printer RedRover]
moonraker_uri: ws://192.168.0.93:7125/websocket
# Define the uri to the moonraker instance.
# Default value: ws://192.168.0.93:7125/websocket
moonraker_api_key: False
# Moonraker API key if force_logins or trusted clients is active!
snapshot_uri: http://192.168.0.93:7125/webcam/?action=snapshot

[printer WhiteLighting]
moonraker_uri: ws://192.168.0.93:7126/websocket
# Define the uri to the moonraker instance.
# Default value: ws://192.168.0.93:7126/websocket
moonraker_api_key: False
# Moonraker API key if force_logins is active!
snapshot_uri: http://192.168.0.93:7126/webcam/?action=snapshot
```

solo2424, 2023-06-26:
Well I am referring to both. in my setup when I add the port (7125 or 7126) the webcams do not show.  But when I remove the port they show. Here is a screenshot of the working url from my browser. 
[image]

Here is a screenshot with the port added.
[image]

The error:
{"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/solo/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1690, in _execute\n    result = self.prepare()\n\n  File \"/home/solo/moonraker/moonraker/app.py\", line 962, in prepare\n    raise tornado.web.HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}

Clon1998, 2023-07-27:
Hey,
I am sorry for my late response. I was quite busy the last few weeks and simply forgo to monitor the companion GitHub page.

Getting back to your issue with the ports. The ports you configured for Moonraker (You referred to them as websocket port) is only used for the webserver that provides moonraker. For the cameras, you won't need to configure a port because they are often automatically exposed via the default port trough either crownsnest or camerastreamer.

Therefore, my initial config suggestion was simply wrong and a mistake on my side since I mixed them up myself for quite a while now (Sorry about that).

Tl; Dr:

Ensure that for both the app and the companion the printers are configured as follow:

**RedRover**
websocket/moonraker url = ws://192.168.0.93:7125/websocket
webcam1 SNAPSHOT url = http://192.168.0.93/webcam/?action=snapshot
webcam1 STREAM url = http://192.168.0.93/webcam/?action=stream

**WhiteLighting**
websocket/moonraker url = ws://192.168.0.93:7125/websocket
webcam2 SNAPSHOT url = http://192.168.0.93/webcam2/?action=snapshot
webcam2 STREAM url = http://192.168.0.93/webcam2/?action=stream

In case you need any more assitant please reach out!

Clon1998, 2023-12-05:
Feel free to reopen the issue if it is still not resolved.

closed 2023-12-05 by Clon1998