Clon1998/mobileraker_companion #41: Pillow 9.5.0 cannot be installed on armv7l

Maeians, 2023-06-27:
Successful installation in Pillow 9.3.0, unsuccessful installation in Pillow 9.5.0

admin@onecloud:~/mobileraker_companion$ ./scripts/install-mobileraker-companion.sh

###### Installing python virtual environment...
Collecting coloredlogs==15.0.1
  Using cached coloredlogs-15.0.1-py2.py3-none-any.whl (46 kB)
Requirement already satisfied: requests==2.26.0 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from -r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 2)) (2.26.0)
Requirement already satisfied: websockets==10.1 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from -r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 3)) (10.1)
Requirement already satisfied: pytz==2022.7.1 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from -r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 4)) (2022.7.1)
Requirement already satisfied: tzlocal==4.3 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from -r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 5)) (4.3)
Collecting Pillow==9.5.0
  Using cached Pillow-9.5.0.tar.gz (50.5 MB)
Requirement already satisfied: humanfriendly>=9.1 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from coloredlogs==15.0.1->-r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 1)) (10.0)
Requirement already satisfied: certifi>=2017.4.17 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from requests==2.26.0->-r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 2)) (2023.5.7)
Requirement already satisfied: charset-normalizer~=2.0.0 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from requests==2.26.0->-r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 2)) (2.0.12)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from requests==2.26.0->-r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 2)) (1.26.16)
Requirement already satisfied: idna<4,>=2.5 in /home/admin/mobileraker-env/lib/python3.9/site-packages (from requests==2.26.0->-r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 2)) (3.4)
Requirement already satisfied: pytz-deprecation-shim in /home/admin/mobileraker-env/lib/python3.9/site-packages (from tzlocal==4.3->-r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 5)) (0.1.0.post0)
Requirement already satisfied: tzdata in /home/admin/mobileraker-env/lib/python3.9/site-packages (from pytz-deprecation-shim->tzlocal==4.3->-r /home/admin/mobileraker_companion/scripts/mobileraker-requirements.txt (line 5)) (2023.3)
Building wheels for collected packages: Pillow
  Building wheel for Pillow (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/admin/mobileraker-env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-bybax3sj/pillow_1cacc4dbbde04324b13d8d9997bed768/setup.py'"'"'; __file__='"'"'/tmp/pip-install-bybax3sj/pillow_1cacc4dbbde04324b13d8d9997bed768/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-_n1phbsb
       cwd: /tmp/pip-install-bybax3sj/pillow_1cacc4dbbde04324b13d8d9997bed768/
  Complete output (179 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.9
  creating build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/features.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/_version.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/_util.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/_tkinter_finder.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/_deprecate.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/_binary.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/__main__.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/__init__.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/XpmImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/XbmImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/WebPImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/WalImageFile.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/TiffTags.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/TiffImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/TgaImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/TarIO.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/SpiderImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/QoiImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PyAccess.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PpmImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PngImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PixarImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PdfParser.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PdfImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PcxImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PcdImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PalmImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PaletteFile.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/PSDraw.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/MspImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/MpoImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
  copying src/PIL/MpegImagePlugin.py -> build/lib.linux-armv7l-3.9/PIL
[... 299 lines cut ...]
        cmd_obj.run()
      File "/home/admin/mobileraker-env/lib/python3.9/site-packages/setuptools/command/build_ext.py", line 84, in run
        _build_ext.run(self)
      File "/usr/lib/python3.9/distutils/command/build_ext.py", line 340, in run
        self.build_extensions()
      File "/tmp/pip-install-bybax3sj/pillow_1cacc4dbbde04324b13d8d9997bed768/setup.py", line 809, in build_extensions
        raise RequiredDependencyException(f)
    __main__.RequiredDependencyException: zlib

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-bybax3sj/pillow_1cacc4dbbde04324b13d8d9997bed768/setup.py", line 1010, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:

    The headers or library files could not be found for zlib,
    a required dependency when compiling Pillow from source.

    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html

    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/admin/mobileraker-env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-bybax3sj/pillow_1cacc4dbbde04324b13d8d9997bed768/setup.py'"'"'; __file__='"'"'/tmp/pip-install-bybax3sj/pillow_1cacc4dbbde04324b13d8d9997bed768/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ztrc8lu0/install-record.txt --single-version-externally-managed --compile --install-headers /home/admin/mobileraker-env/include/site/python3.9/Pillow Check the logs for full command output.

mdewyer, 2023-08-22:
Getting the same error - Creality SonicPad running Debian (https://github.com/Jpe230/SonicPad-Debian)

Clon1998, 2023-08-22:
Sorry I can not do that much. I am not an expert in python nor with Linux distros. Every distro seems to require a different version of pillow and I thought I covered most of them in my script.
I suggest you guys disabled the section in the script that is responsible for install pillow and take care of installing pillow yourself.

mdewyer, 2023-08-22:
No worries @Clon1998, thanks for the direction. 

I was able to install Pillow manually using the code from their [Installation Instructions](https://pillow.readthedocs.io/en/latest/installation.html):
```
python3 -m pip install --upgrade pip
python3 -m pip install --upgrade Pillow
```
Then I removed the `Pillow==9.5.0` line from **~/mobileraker_companion/scripts/mobileraker-requirements.txt**

And finally I was able to run the **install-mobileraker-companion.sh** script successfully.

The mobile app isn't complaining about the companion app not being installed, so seems to have worked! Thank you for your help again!