Clon1998/mobileraker_companion #44: Multiple Printers - Mobileraker Log File Missing

fuzelet, 2023-08-05:
Upon doing some tests on another issue, it seems a multiple printer setup using KIAUH does not create any mobileraker.log file on the system.  When I used KIAUH and only installed one printer, I was able to find the mobileraker.log file in the ~/printer_data/ location. I have since gone back to a multiple printer setup and the mobileraker.log file does not exist in either /printer_data, /ender3_data, /ender6_data, or /tmp.

I used KIAUH to install two printers, named ender3 and ender6.
[image]
[image]

Other Issue describing issue finding logs.
https://github.com/Clon1998/mobileraker_companion/issues/38

Clon1998, 2023-08-12:
Hey, could you do me a favor and send me the content of your service file:

```bash
cat /etc/systemd/system/mobileraker.service
```

fuzelet, 2023-08-12:
Here you go.

#Systemd service file for mobileraker
[Unit]
Description=Companion app to enable push notifications on mobileraker
After=network-online.target moonraker.service

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
User=jeffrey
WorkingDirectory=/home/jeffrey/mobileraker_companion
ExecStart=/home/jeffrey/mobileraker-env/bin/python /home/jeffrey/mobileraker_companion/mobileraker.py
Restart=always
RestartSec=10

ShaunRoberts543, 2023-08-17:
Same issue here.  I have 3 printer instances.  Printer #1 is showing on app as it should, pinter #2 and #3 still say "need companion app".  I also have not been able to locate the log file at all.  I did create a Mobileraker.conf file and it is on the #1st instance that is working, but with the other 2 listed in the file.

Clon1998, 2023-08-17:
Hey,
I am currently not at home for another month.
However, a new/2nd printer will be available once I am back home to test this.

Clon1998, 2023-08-17:
> Same issue here. I have 3 printer instances. Printer #1 is showing on app as it should, pinter #2 and #3 still say "need companion app". I also have not been able to locate the log file at all. I [...]

Also could you share the mobileraker config file with me?

Clon1998, 2023-12-05:
If you still need help feel free to reopen the issue.

closed 2023-12-05 by Clon1998