Clon1998/mobileraker_companion #47: not running from service file

TomW1605, 2023-08-12:
i just installed this but it dosnt run when using the system service. the status stays on "Activating" and it never starts
[image]

when i run the command from the service file directly it works fine

Clon1998, 2023-08-12:
Can you please run `/home/pi/mobileraker-env/bin/python --version` and verify that you are running python 3.x?

TomW1605, 2023-08-12:
yeah its python 3.9.2
[image]

Clon1998, 2023-08-12:
Could you check if the app creates a `mobileraker.log` file in either `~/printer_data/logs/mobileraker.log` or in `/tmp/mobileraker.log`

TomW1605, 2023-08-12:
running the service does not but running the python script directly does

TomW1605, 2023-08-31:
hi, this is still an issue. while i was working on the notification stuff it wasnt much of a problem keeping a terminal open with this running since i was restarting it a lot anyway but now its getting a bit annoying. any ideas on what might be causing this? i updated to the latest version to get the rewrite stuff hoping that might fix it but it didnt

Clon1998, 2023-08-31:
Seems to be a problem with your distro

The service errors with 217/USER

https://www.shellhacks.com/systemd-service-not-starting-status217-user/

TomW1605, 2023-08-31:
its a stock raspberry pi os install but ill have a look and see if i can fix it from whats in the link

TomW1605, 2023-08-31:
ok, so turns out i have been accidentally overriding the `$USER` environment variable in my dotfiles for years and somehow this is the first time its ben an issue. thanks for your help and the great app. fixed the unit file and its all working now

closed 2023-08-31 by TomW1605