Clon1998/mobileraker_companion #50: No Notifications | HTTP Connection Timeout Error

fuzelet, 2023-08-15:
Sorry if I opened an issue that already exists, but the ones I found were for a much older version of the application or a not exactly what I have seen myself.  I have not been getting any notifications from the application and I get the following error message when I randomly open the application.  Not sure if the two are related or not?  The only thing I can think of recently that changed was I recently entered the Google Play Store beta along with the various updates to mainsail, mobileraker, etc

Not sure the logs attached help or not.

v2.5.2-243
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/12346190/mobileraker_2023-08-15T10-52-24.log)
[moonraker.log](https://github.com/Clon1998/mobileraker_companion/files/12346242/moonraker.log)
[Mobilerakerconf.zip](https://github.com/Clon1998/mobileraker_companion/files/12346273/Mobilerakerconf.zip)

[image]
[image]
[image]

fuzelet, 2023-08-15:
I think i found the issue with the notifications by looking at the log, however not sure if that is the reason for the HTTP issue?  I needed to %% my %p in the conf file?
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/12346567/mobileraker.log)

fuzelet, 2023-08-15:
After fixing the % issue above I am still not getting the basic notifications from the application.  I did attach another log file that included an error on my most recent print, etc.

2023-08-15 12:46:45,902 mobileraker.jrpc WARNING - Error message received from WebSocket-Server Item at key 'fcm.098c0ea4-6f24-4db1-a24c-d95aff6d2d38.snap' in namespace 'mobileraker'is not a dictionary object, cannot insert

[logs-20230802-125410.zip](https://github.com/Clon1998/mobileraker_companion/files/12346636/logs-20230802-125410.zip)

Clon1998, 2023-08-15:
Seems like there is a bug in 0.4.0 that prevents the companion from writing into it. V0.5.0 is currently in public testing and will should solve that issue.

fuzelet, 2023-08-15:
> The branch name is actually rewrite

Thanks. I followed the below instructions and I am now on v0.4.0-31-g77141313 as of this comment.

cd ~/mobileraker_companion
git pull 
git switch rewrite
./scripts/install-mobileraker-companion.sh

fuzelet, 2023-08-15:
Since updating to the rewrite branch I am still getting printer timeout errors within the application. Anything I can provide, etc?

Clon1998, 2023-08-15:
The timeout errors on the app are not related to the companion. 
You see the timeout error in the mobile app because either your network or printer is too slow to answer the network call of mobileraker which has a fixed timeout of 3 seconds. Since there are a few more users that report the same behavior I will add a new field to the printer config that allows you to increase the timeout e.g. to 10 sec. 

After looking into the logs of the python app / companion I don't see a reason to why you don't receive the notifications.
Could you please issue the following GCode command in the GCode Console : `M117 $MR$:Hey I am a notification`

Afterwards please provide me the logs of the `mobileraker_companion`. The rewrite branch improved the logging and should give me a better idea about what is happening.

fuzelet, 2023-08-16:
Appreciate the response.  Just ran the command from the console and did not get any notification. I am seeing my mobileraker log full of the following error and nothing else. 

[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/12358982/mobileraker.log)
[mobileraker1.log](https://github.com/Clon1998/mobileraker_companion/files/12358992/mobileraker1.log)
[mobileraker2.log](https://github.com/Clon1998/mobileraker_companion/files/12359004/mobileraker2.log)

--------------------MobilerakerCompanion Log Start--------------------
Git Version: v0.4.0-31-g7714131
2023-08-16 07:44:45,793 root INFO - MobilerakerCompanion version: v0.4.0-31-g7714131
2023-08-16 07:44:45,793 root INFO - Passed config file is: /home/jeffrey/Mobileraker.conf
2023-08-16 07:44:45,793 root INFO - Found configuration file at: /home/jeffrey/Mobileraker.conf
2023-08-16 07:44:45,793 root INFO - Read 1 printer config sections
2023-08-16 07:44:45,799 root INFO - Main section read, language:"en", timezone:"America/Detroit", eta_format:"%p Completed; ETA: %H:%M", include_snapshot:"False"
2023-08-16 07:44:45,800 root ERROR - Uncaught exception <class 'TypeError'>: Lock.__init__() got an unexpected keyword argument 'loop'
  File "/home/jeffrey/mobileraker_companion/mobileraker.py", line 86, in <module>
    main()

  File "/home/jeffrey/mobileraker_companion/mobileraker.py", line 69, in main
    client = MobilerakerCompanion(
             ^^^^^^^^^^^^^^^^^^^^^

  File "/home/jeffrey/mobileraker_companion/mobileraker/mobileraker_companion.py", line 53, in __init__
    self._evaulate_noti_lock: Lock = Lock(loop=loop)
                                     ^^^^^^^^^^^^^^^
NoneType: None

I also see the following happening which might be related to the timeout?
2023-08-15 14:19:57,108 mobileraker._Default INFO - Querying printer Objects
2023-08-15 14:19:57,355 mobileraker._Default INFO - Subscribing to printer Objects
2023-08-15 14:19:57,355 mobileraker._Default INFO - Client init completed
2023-08-15 14:19:57,355 mobileraker._Default INFO - Took a PrinterSnapshot: PrinterSnapshot(print_state=standby, m117=None, m117_hash=, progress=None, filename=, printing_duration=None)
2023-08-15 14:19:57,858 mobileraker._Default INFO - Took a PrinterSnapshot: PrinterSnapshot(print_state=standby, m117=None, m117_hash=, progress=None, filename=, printing_duration=None)
2023-08-15 14:41:50,452 websockets.client INFO - ! connect failed; reconnecting in 3.7 seconds
Traceback (most recent call last):
  File "/home/jeffrey/mobileraker-env/lib/python3.11/site-packages/websockets/legacy/client.py", line 598, in __aiter__
    async with self as protocol:
  File "/home/jeffrey/mobileraker-env/lib/python3.11/site-packages/websockets/legacy/client.py", line 633, in __aenter__
    return await self
           ^^^^^^^^^^
  File "/home/jeffrey/mobileraker-env/lib/python3.11/site-packages/websockets/legacy/client.py", line 650, in __await_impl_timeout__
    return await asyncio.wait_for(self.__await_impl__(), self.open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/home/jeffrey/mobileraker-env/lib/python3.11/site-packages/websockets/legacy/client.py", line 654, in __await_impl__
    transport, protocol = await self._create_connection()
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1085, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1069, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 634, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 674, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.0.1', 7125)
2023-08-15 14:41:54,176 mobileraker.jrpc INFO - WebSocket connected
2023-08-15 14:41:54,176 mobileraker._Default INFO - Client init started... Try#0

Clon1998, 2023-08-16:
Sorry that it took some time to get back to you.
I don't know why you are seeing all of these pthon errors. I still need to look into that, however, they are not the source of the problem.
Something went wrong when I added the new notification Icon for android. It is working just fine on my emulator and dev build but the production version is complaining about not being able to locate the icon which prevents the app from creating notifciations.

fuzelet, 2023-08-20:
> Sorry that it took some time to get back to you. I don't know why you are seeing all of these pthon errors. I still need to look into that, however, they are not the source of the problem. Something [...]

I can confirm that I am getting notifications again since the icon change on Android. I probably should always include the version and OS. My apologies on that.

I am still getting the timeout errors though given that is still at 3 seconds. My question though on that front, did something around timeouts change? I haven't touched anything in my network and I just started getting it recently. I'm running 1gbps+ everywhere which is surprising if that's not fast enough anymore.

Clon1998, 2023-08-20:
> I can confirm that I am getting notifications again since the icon change on Android. I probably should always include the version and OS. My apologies on that.
>
>
>
> I am still getting the timeout [...]

Network delay/speed is not the only thing that contributes to a connection timeout. The processing time in the Webserver in our case the raspberry pi or host computer that runs Moonraker also adds to the delay.
On the development branch I already included the option to set a different timeout. This should be included in the next release 😉

fuzelet, 2023-08-20:
> 

Are you aware if the system logs what caused the timeout anywhere or something I can go chase on my box? I ask because this is an i5-1240p(16CPU) box with NVMe drive, etc so I hope it's not adding a delay unnecessarily.

Clon1998, 2023-08-20:
> > 
> 
> 
> 
> Are you aware if the system logs what caused the timeout anywhere or something I can go chase on my box? I ask because this is an i5-1240p(16CPU) box with NVMe drive, etc so I hope it's not adding a delay unnecessarily.

Well in that case it should work just fine. Not to sure than what is causing your problem.

fuzelet, 2023-08-21:
Wanted to report back that on the rewrite branch the companion keeps failing at startup, therefore no notifications.  As soon as I switch back to main it works fine.

**cd ~/mobileraker_companion
git pull
git switch rewrite
./scripts/install-mobileraker-companion.sh**

--------------------MobilerakerCompanion Log Start--------------------
Git Version: v0.4.0-32-g420cd2e
2023-08-21 15:08:51,921 root INFO - MobilerakerCompanion version: v0.4.0-32-g420cd2e
2023-08-21 15:08:51,921 root INFO - Passed config file is: /home/jeffrey/Mobileraker.conf
2023-08-21 15:08:51,921 root INFO - Found configuration file at: /home/jeffrey/Mobileraker.conf
2023-08-21 15:08:51,921 root INFO - Read 1 printer config sections
2023-08-21 15:08:51,926 root INFO - Main section read, language:"en", timezone:"America/Detroit", eta_format:"Completed; ETA: %I:%M %p", include_snapshot:"False"
2023-08-21 15:08:51,927 root ERROR - Uncaught exception <class 'TypeError'>: Lock.__init__() got an unexpected keyword argument 'loop'
  File "/home/jeffrey/mobileraker_companion/mobileraker.py", line 86, in <module>
    main()

  File "/home/jeffrey/mobileraker_companion/mobileraker.py", line 69, in main
    client = MobilerakerCompanion(
             ^^^^^^^^^^^^^^^^^^^^^

  File "/home/jeffrey/mobileraker_companion/mobileraker/mobileraker_companion.py", line 53, in __init__
    self._evaulate_noti_lock: Lock = Lock(loop=loop)
                                     ^^^^^^^^^^^^^^^
NoneType: None

**cd ~/mobileraker_companion
git pull
git switch main
./scripts/install-mobileraker-companion.sh**

--------------------MobilerakerCompanion Log Start--------------------
Git Version: v0.4.0-21-g7bb9f8f
2023-08-21 15:09:02,238 root INFO - MobilerakerCompanion version: v0.4.0-21-g7bb9f8f
2023-08-21 15:09:02,238 root INFO - Passed config file is: /home/jeffrey/Mobileraker.conf
2023-08-21 15:09:02,238 root INFO - Found configuration file at: /home/jeffrey/Mobileraker.conf
2023-08-21 15:09:02,239 root INFO - Read 1 printer config sections
2023-08-21 15:09:02,250 root INFO - Main section read, language:"en", timezone:"America/Detroit", eta_format:"Completed; ETA: %I:%M %p", include_snapshot:"False"
2023-08-21 15:09:02,250 mobileraker.jrpc INFO - Trying to connect to: ws://127.0.0.1:7125/websocket api key <NO API KEY>
2023-08-21 15:09:02,252 mobileraker.jrpc INFO - WebSocket connected
2023-08-21 15:09:02,252 mobileraker._Default INFO - Client init started... Try#0
2023-08-21 15:09:02,253 mobileraker._Default INFO - Received Server Info

Clon1998, 2023-08-21:
> Wanted to report back that on the rewrite branch the companion keeps failing at startup, therefore no notifications. As soon as I switch back to main it works fine.
> 
> **cd ~/mobileraker_companion git pull git switch rewrite ./scripts/install-mobileraker-companion.sh**
> 
> --------------------MobilerakerCompanion Log Start-------------------- Git Version: v0.4.0-32-g420cd2e 2023-08-21 15:08:51,921 root INFO - MobilerakerCompanion version: v0.4.0-32-g420cd2e 2023-08-21 15:08:51,921 root INFO - Passed config file is: /home/jeffrey/Mobileraker.conf 2023-08-21 15:08:51,921 root INFO - Found configuration file at: /home/jeffrey/Mobileraker.conf 2023-08-21 15:08:51,921 root INFO - Read 1 printer config sections 2023-08-21 15:08:51,926 root INFO - Main section read, language:"en", timezone:"America/Detroit", eta_format:"Completed; ETA: %I:%M %p", include_snapshot:"False" 2023-08-21 15:08:51,927 root ERROR - Uncaught exception <class 'TypeError'>: Lock.**init**() got an unexpected keyword argument 'loop' File "/home/jeffrey/mobileraker_companion/mobileraker.py", line 86, in main()
> 
> File "/home/jeffrey/mobileraker_companion/mobileraker.py", line 69, in main client = MobilerakerCompanion( ^^^^^^^^^^^^^^^^^^^^^
> 
> File "/home/jeffrey/mobileraker_companion/mobileraker/mobileraker_companion.py", line 53, in **init** self._evaulate_noti_lock: Lock = Lock(loop=loop) ^^^^^^^^^^^^^^^ NoneType: None
> 
> **cd ~/mobileraker_companion git pull git switch main ./scripts/install-mobileraker-companion.sh**
> 
> --------------------MobilerakerCompanion Log Start-------------------- Git Version: v0.4.0-21-g7bb9f8f 2023-08-21 15:09:02,238 root INFO - MobilerakerCompanion version: v0.4.0-21-g7bb9f8f 2023-08-21 15:09:02,238 root INFO - Passed config file is: /home/jeffrey/Mobileraker.conf 2023-08-21 15:09:02,238 root INFO - Found configuration file at: /home/jeffrey/Mobileraker.conf 2023-08-21 15:09:02,239 root INFO - Read 1 printer config sections 2023-08-21 15:09:02,250 root INFO - Main section read, language:"en", timezone:"America/Detroit", eta_format:"Completed; ETA: %I:%M %p", include_snapshot:"False" 2023-08-21 15:09:02,250 mobileraker.jrpc INFO - Trying to connect to: ws://127.0.0.1:7125/websocket api key 2023-08-21 15:09:02,252 mobileraker.jrpc INFO - WebSocket connected 2023-08-21 15:09:02,252 mobileraker._Default INFO - Client init started... Try#0 2023-08-21 15:09:02,253 mobileraker._Default INFO - Received Server Info

Updated the code on rewrite, could you try it again?

fuzelet, 2023-08-21:
I am now able to load the companion on the rewrite branch and no error is present.

Is the timeout variable you mentioned being in development present on the rewrite branch or is it another one?

Clon1998, 2023-08-22:
> I am now able to load the companion on the rewrite branch and no error is present.
> 
> 
> 
> Is the timeout variable you mentioned being in development present on the rewrite branch or is it another one?

The timeout us not related to the companion. Rather to the app itself. I already included that change on the latest closed beta and TestFlight version.

closed 2023-08-26 by Clon1998

fuzelet, 2023-08-28:
I can open a new issue, but since v0.4 update it appears the notifications on Android does not respect Mobileraker.conf for notifications anymore?  Did adding the new layer information and rewriting the app need the have the conf updated in some way?  As the "eta_format" section in the conf file used to be the way to customize the notifications sent by the application at % updates.

**Mobileraker.conf**
[general]
language: en
timezone: America/Detroit
eta_format: **Completed; ETA: %%I:%%M %%p**
include_snapshot: False
[printer]
moonraker_uri: ws://127.0.0.1:7125/websocket
moonraker_api_key: False
snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
snapshot_rotation: 0

[logs-20230801-103445.zip](https://github.com/Clon1998/mobileraker_companion/files/12455465/logs-20230801-103445.zip)

023-08-27 19:09:48,953 mobileraker._Default INFO - Fetched 1 app Cfgs!
2023-08-27 19:09:48,953 mobileraker._Default INFO - Evaluate for machineID f33b124e-d1ef-433a-a26a-1dddf198536f, cfg.snap: NotificationSnap(progress=20, state=printing, m117=08f1b7de713fc6929322fc9fc708b72f87b3087bc0f6a1fed393f6975a6e351f), cfg.settings: NotificationSettings(progress_config=20, state_config=['printing', 'complete', 'error'], created=2023-08-19T19:47:36.961626, last_modified=2023-08-19T19:47:36.961626)
2023-08-27 19:09:48,953 mobileraker._Default INFO - ProgressNoti preChecks: cfg.progress.config: 40 - 20 = 20 < 20 RESULT: True
2023-08-27 19:09:48,953 mobileraker._Default INFO - ProgressNoti: **Print progress of Ender 6 - 40%, ETA: 19:36:41, Layer: 31/127**
2023-08-27 19:09:48,953 mobileraker._Default INFO - 1 Notifications for machineID: f33b124e-d1ef-433a-a26a-1dddf198536f: state: False, proggress: True, M117 False
2023-08-27 19:09:48,953 mobileraker._Default INFO - Updating snap in FCM Cfg for f33b124e-d1ef-433a-a26a-1dddf198536f: NotificationSnap(progress=40, state=printing, m117=08f1b7de713fc6929322fc9fc708b72f87b3087bc0f6a1fed393f6975a6e351f)

https://github.com/Clon1998/mobileraker_companion#companion---config
https://strftime.org/

fuzelet, 2023-08-30:
Just want to report back my findings.

I have both the Android Beta and iOS Beta of the applications installed now.  Since configuring the timeout in the iOS version to 10 seconds, I am not seeing timeout errors.  

android v2.5.3-249
ios v2.5.4-259