Clon1998/mobileraker_companion #52: Phantom Messages of a excluded printer

dhnsilva, 2023-08-24:
I am receiving messages of an excluded printer. 
I had a cartesian printer called N955 and I trasform it on a Corexy. 
Renewed the Klipper and reinstalled all needs. 
Reinstalled the Mobileraker as a new name QUAD300. 

Now I am receiving a duplicate messages of mobileraker of a N955 and QUAD300 as the same printer.

I don't know where I can find this duplicate or old entry to erase! 
Can someone help me?

Clon1998, 2023-08-24:
Hey,
mobileraker stores information about the printer+push notification device, inside Moonraker's database.
There are some rare edge cases in which the deletion of a printer does not fully remove the printer from this database which can cause these ghost notifications.
In order to resolve that issue you will need to delete the database entry of mobileraker. Currently, this is not possible from within the app but will be included in a future release.

In order to manually delete the database entry, you will need to issue the following HTTP-DELETE: `DELETE http://<PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm`. You can do that either via console or any other HTTP client. 

For example, you can run the following code from within your brower's development console (Press F12 to open it), please make sure your current browser window is showing your printers web UI (e.g. Mainsail or FLuidd):

```js
fetch("http://<PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm", { method: "DELETE"} )
```

dhnsilva, 2023-08-27:
Thank you!
Its Worked like a charm!

closed 2023-08-28 by Clon1998