Clon1998/mobileraker_companion #54: Customise notification strings via config file

Clon1998, 2023-08-28:
I can open a new issue, but since v0.4 update it appears the notifications on Android does not respect Mobileraker.conf for notifications anymore?  Did adding the new layer information and rewriting the app need the have the conf updated in some way?  As the "eta_format" section in the conf file used to be the way to customize the notifications sent by the application at % updates.

https://github.com/Clon1998/mobileraker_companion#companion---config
https://strftime.org/

**Mobileraker.conf**
[general]
language: en
timezone: America/Detroit
eta_format: **Completed; ETA: %%I:%%M %%p**
include_snapshot: False
[printer]
moonraker_uri: ws://127.0.0.1:7125/websocket
moonraker_api_key: False
snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
snapshot_rotation: 0

[logs-20230801-103445.zip](https://github.com/Clon1998/mobileraker_companion/files/12455465/logs-20230801-103445.zip)

023-08-27 19:09:48,953 mobileraker._Default INFO - Fetched 1 app Cfgs!
2023-08-27 19:09:48,953 mobileraker._Default INFO - Evaluate for machineID f33b124e-d1ef-433a-a26a-1dddf198536f, cfg.snap: NotificationSnap(progress=20, state=printing, m117=08f1b7de713fc6929322fc9fc708b72f87b3087bc0f6a1fed393f6975a6e351f), cfg.settings: NotificationSettings(progress_config=20, state_config=['printing', 'complete', 'error'], created=2023-08-19T19:47:36.961626, last_modified=2023-08-19T19:47:36.961626)
2023-08-27 19:09:48,953 mobileraker._Default INFO - ProgressNoti preChecks: cfg.progress.config: 40 - 20 = 20 < 20 RESULT: True
2023-08-27 19:09:48,953 mobileraker._Default INFO - ProgressNoti: **Print progress of Ender 6 - 40%, ETA: 19:36:41, Layer: 31/127**
2023-08-27 19:09:48,953 mobileraker._Default INFO - 1 Notifications for machineID: f33b124e-d1ef-433a-a26a-1dddf198536f: state: False, proggress: True, M117 False
2023-08-27 19:09:48,953 mobileraker._Default INFO - Updating snap in FCM Cfg for f33b124e-d1ef-433a-a26a-1dddf198536f: NotificationSnap(progress=40, state=printing, m117=08f1b7de713fc6929322fc9fc708b72f87b3087bc0f6a1fed393f6975a6e351f)

[image]
[image]
[image]

_Originally posted by @fuzelet in https://github.com/Clon1998/mobileraker_companion/issues/50#issuecomment-1695823051_

Clon1998, 2023-08-28:
@fuzelet,
the `eta_format` was never intended to be used for customizing the format of the progress notification. It's sole purpose is to customize the time format of the time format for eta and adaptive eta.

I initially planned to support custom formats, but it seems like I never got around to implementing them.
As a workaround, you can temporarily modify the [18n.py](https://github.com/Clon1998/mobileraker_companion/blob/main/mobileraker/util/i18n.py) file.

I am planning on making the format configurable via the config once I have got some more time :).

fuzelet, 2023-09-23:
By changing 
'print_progress_body': '$progress, ETA: **$a_eta**, Layer: $cur_layer/$max_layer',
to
'print_progress_body': '$progress, ETA: $eta, Layer: $cur_layer/$max_layer, Time Left: $remaining_avg',

it inherited what i have configured in the conf file. Any reason $a_eta was chosen over $eta? When using $remaining_avg I also only get a string of numbers outputted. 

2023-09-22 19:51:48,050 mobileraker._Default INFO - ProgressNoti: Print progress of Ender 6 - 25%, ETA: 10:58 PM, Layer: 59/657, Time Left: 11220

https://github.com/Clon1998/mobileraker_companion/blob/0bb58bdda928dbdfd4074fa0bac3910b241d8ada/docs/Custom_Notifications.md

Clon1998, 2023-09-23:
Adaptive ETA should only use the timestamp (HH:MM) if the print ends on the same day, while eta always uses the format defined in the config which by default is the date and time stamp.

Regarding remaining avg, i will have a look. Seems like it only output's the seconds hence I forgot to add the formatting in the cod.