Clon1998/mobileraker_companion #55: Moonraker warning.

StinkyTACO, 2023-09-01:
i just installed the companion with kiauh and all my instances show this error but looking it didnt yield much in the way of fixing this
[image]

Clon1998, 2023-09-02:
Hey,
by default my installation scripts looks for the `moonraker.asv` file in the following dir `~/printer_data/moonraker.asvc`.

I assume you are running multiple printers from a single host. Therefore, the install script was unable to find the correct file.
Could you do me a favor and dump the file structure of your host/klipper files?

Clon1998, 2023-09-02:
I think there is not much I can do.
By default, you can serve multiple printers with a single companion instance. Therefore, I can not add the service to every `moonraker.asv` file since I don't know the structure of these multi-printer setups.

StinkyTACO, 2023-09-03:
i am indeed running multiple instances but kiauh auto installs one for each instance
is there an easy and fast way to dump file structure? or do i need to manually type all that ?

Clon1998, 2023-09-03:
> i am indeed running multiple instances but kiauh auto installs one for each instance
> 
> is there an easy and fast way to dump file structure? or do i need to manually type all that ?

actually it's fine. 
I can imagine how the file structure is.
I will check on KIAUH to make sure it only installs once or properly supports multiple printer Installations.

If you want to fix that error, just add a line mobileraker to the mentioned file of the error.

StinkyTACO, 2023-09-03:
This can not be done through the mainsail filebrowser correct? I don't believe the .asv file can be edited through there. So where would the file be located?

StinkyTACO, 2023-09-03:
Nevermind you said it earlier i will try it out and get back to you.

closed 2023-12-05 by Clon1998