Clon1998/mobileraker_companion #56: M117 $MR$: Does not work now.

BlackNet, 2023-09-01:
Started a few days ago. Just stopped working. I did some entry testings and nothing. Have two printers doing this.

Clon1998, 2023-09-01:
I sadly can not reproduce it.
Custom notifications work just fine for me.

What version of Mobileraker are you using?
Could you try the new version that makes use of a [macro](https://github.com/Clon1998/mobileraker_companion/blob/main/docs/Custom_Notifications.md#using-the-mr_notify-custom-macro).

Clon1998, 2023-09-01:
Also did you restart the companion?

BlackNet, 2023-09-01:
Will look at the macro. Most are M117 $MR$.  I have rebooted the host several times etc. both machines.

Clon1998, 2023-09-01:
> Will look at the macro. Most are M117 MR. I have rebooted the host several times etc. both machines.
> 
> Version 2.5.2-246

Thanks could you please attach the log files of the companion and also the companions's version, both should be available on mainsail/fluidd

BlackNet, 2023-09-01:
ok restarting the service did work.  however, when there is an active print in place there is no notifications.

Clon1998, 2023-09-02:
Without the mobileraker.logs of the companion I can not look into that.
It is working fine for me.

BlackNet, 2023-09-02:
macro seems to work.  Not sure what happened. there were NO log entries while it was offline which makes me think it was the service itself.

closed 2023-09-02 by BlackNet

Clon1998, 2023-11-30:
Are you still experiencing this issue and need help?