Clon1998/mobileraker_companion #57: Possible memory leak in mobileraker.py

austinrdennis, 2023-09-09:
## Bug Report

### Description
I've noticed a slow and gradual increase in the amount of memory mobileraker.py uses that points to a possible memory leak.

### Steps to Reproduce
Boot the host machine and start the mobileraker service. Not sure what exactly the mechanism is.

### Expected Behavior
Memory usage remains relatively stable over time and is reasonable.

### Screenshots
[image]
[image]

### Version Information
- Mobileraker: v0.4.0-25-g2808855d
- Klipper: v0.11.0-271-g5f990f93
- Moonraker: v0.8.0-138-gfe120952

### Debug Logs
[mobileraker_2023-09-08T21-06-33.log](https://github.com/Clon1998/mobileraker/files/12566802/mobileraker_2023-09-08T21-06-33.log)
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/12566808/mobileraker.log)

### Additional Context
After about 26 hours of runtime, it's now using around 2.1 GB of system memory which seems like way too much.

austinrdennis, 2023-09-09:
I've also posted this as bug in the mobileraker repo as issue 251 as I wasn't sure which one to choose. Feel free to close this issue if it's not applicable this repo.

austinrdennis, 2023-09-09:
`sudo systemctl restart mobileraker.service` clears out the memory usage at least temporarily. I'll see if it builds up again.

Clon1998, 2023-09-09:
Hey, thanks for reporting!
This issue has been bogging me for quite a while now. However, I hope the fix I just pushed resolves it.

Duplicate of #27

Clon1998, 2023-09-09:
Please let me know if you observe any changes after updating to fa99513. 
Either in this issue or in #27

closed 2023-09-09 by Clon1998