Clon1998/mobileraker_companion #61: not picking up mobileraker.conf

faspina, 2023-10-01:
I have the mobileraker_companion installed. 

Looking at the moonraker.log it doesn't look like its finding my customer mobileracker.conf I have tried it in the 

~/ and 
~/printer_data

installed with Kiauh

I have forced logins and trusted clients on so I added change the url 
and add my api key to 
moonraker_api_key:

but according to moonraker.logs it its still looking at the local ip and not sending a api key

Clon1998, 2023-10-01:
Can you upload the mobileraker.log file?

faspina, 2023-10-01:
You can close, finally put it in the mobileraker_companion directory and I got it working. I also have mine behind a proxy so had to use wss:/// that seem to make it work

closed 2023-10-01 by faspina