Clon1998/mobileraker_companion #62: Multiple printer on multiple Pies, notifications only working on 1 printer

wikolii, 2023-10-03:
Hi

So we have 8 printer spread out on 3 Pies

The 1st Pi is the only one with the companion installed and set up to watch printers 1-6

Notifications are only working for printer #3

I can't find mobileraker.log anywhere
```
vic@raspberrypi1:~ $ find . -name mobileraker.log
vic@raspberrypi1:~ $ find . -name moonraker.log
./K1_data/logs/moonraker.log
./K3_data/logs/moonraker.log
./K2_data/logs/moonraker.log

```
[moonraker.log](https://github.com/Clon1998/mobileraker_companion/files/12791658/moonraker.log)
[Mobileraker.zip](https://github.com/Clon1998/mobileraker_companion/files/12791674/Mobileraker.zip)

Thanks!

Clon1998, 2023-10-03:
Since you are running multiple printers,I'd assume the log is located in /tmp could you check that and provide it to me?

wikolii, 2023-10-03:
That's right!
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/12792256/mobileraker.log)
[image]

Clon1998, 2023-10-03:
Thanks,
I had a look at the logs and it seems like you never connected the app to any other printer besides the `MK23-3`.
All other machines do not report an App-Cfg.
```log
..
2023-10-03 00:03:24,114 mobileraker.<MK23-3> INFO - Fetched 1 app Cfgs!
2023-10-03 00:03:24,114 mobileraker.<MK23-4> INFO - Fetched 0 app Cfgs!
2023-10-03 00:03:25,170 mobileraker.<MK23-5> INFO - Fetched 0 app Cfgs!
...
```

You might need to select all printers in the app once.

Clon1998, 2023-10-03:
Could you provide me the app logs to verify the app correctly tries to setup the app cfgs on all printers added to the app?

wikolii, 2023-10-04:
I think I got it. I may have installed the companion on the pi after adding printers 1 and 2, then I added the rest to the app.
Is it 1 companion per pi? That would explain why the rest of the printer (4 onwards) won't work, their pies don't have the companion installed.

I'll keep testing and update later

Clon1998, 2023-10-04:
I am sorry I didn't have time to look at the logs yet.

However, by default you won't need a companion per Pi since the companion should be able to connect to as many printers as you added. However, since I only own a single printer I never got around to test this properly 🙃.
But according to the mobileraker logs the companion was able to connect to the other printers. Just the info about the app was missing.
By default, this info should be set by the app automatically if the info is missing. I need to check the logs to see what is going on here.

wikolii, 2023-10-04:
Ok i can confirm that removing and re-adding (app level) the printers that were not installed prior to installing the Companion on the Pi, enabled notifications. Not sure if this is the correct "solution" but it worked in my case.
I'm very happy right now, thanks for your efforts!

closed 2023-10-04 by wikolii