Clon1998/mobileraker_companion #65: Companion via Docker is not found in Mobileraker

PatrickGlatz, 2023-11-29:
Hi 

although i have mobileraker_companion installed via Docker at ~/mobileraker_companion (mobileraker.conf is at ~/mobileraker_companion/mobileraker.conf) my iPhone does mention, that Companion is not found (see picture). The container is running 

`$ docker inspect -f '{{.State.Status}}' mobileraker_companion-mobileraker_companion-1
running`

`$ docker ps --all
CONTAINER ID   IMAGE                                           COMMAND                  CREATED          STATUS          PORTS                                       NAMES
4a7f86f523c4   ghcr.io/clon1998/mobileraker_companion:latest   "venv/bin/python mob…"   25 minutes ago   Up 19 minutes                                               mobileraker_companion-mobileraker_companion-1
fa49174fa15a   ghcr.io/donkie/spoolman:latest                  "uvicorn spoolman.ma…"   17 hours ago     Up 17 hours     0.0.0.0:7912->8000/tcp, :::7912->8000/tcp   spoolman-spoolman-1`

So the general app is working, i can see the webacm, control everything etc. But companion does not seem to connect. Do you have an advice where to look at? Didn't found any hints or description.

Thanks in advance
Patrick

[image]

Clon1998, 2023-11-29:
It might be worth to have a look at the log files of the companion.
By default they are located under /tmp or /printer_data/logs. Since you are running a container it should be /tmp rather than /printer_data

Clon1998, 2023-11-29:
You can also check if the companion itself is started within the container by attaching to the container's CLI and verifying running systemctl status mobileraker.service.

PatrickGlatz, 2023-11-29:
@Clon1998 That was a good hint. The docker container log mentions a wrong path of the config file, and i dont know how to correct that, see log output below. I thought that i have done everything correct.

As mentioned i have created the docker-compose.yml as well as the mobileraker.conf, then i started to install the container from inside of /home/user/mobileraker_companion/ using (not passing any config)
`docker compose up -d`

Do you have a hint, on how to resolve that issue? Whatever i try, it doesnt take the mobileraker.conf file. Also the config from the container mentions wrong addresses.

Honestly, this is my first docker project, but i thought it would be a good entry point. So maybe you can give me a guideline.

Thanks in advance
Patrick

/home/user/mobileraker_companion/docker-compose.yml
````
services:
  mobileraker_companion:
    image: ghcr.io/clon1998/mobileraker_companion:latest
    volumes:
    - /path/to/mobileraker.conf:/home/user/printer_data/config/mobileraker.conf
````

/home/user/printer_data/config/mobileraker.conf
```
[general]
language: en
# one of the supported languages defined in i18n.py#languages (de,en,...)
# Default: en
timezone: Europe/Berlin
# correct timezone e.g. Europe/Berlin for Berlin time or US/Central.
# For more values see https://gist.github.com/heyalexej/8bf688fd67d7199be4a1682b3eec7568
# Default: Tries to use system timezone
# Optional
eta_format: %%Y-%%m-%%d %%H:%%M:%%S
# Format used for eta and adaptive_eta placeholder variables
# For available options see https://strftime.org/
# Note that you will have to escape the % char by using a 2nd one e.g.: %d/%m/%Y -> %%d/%%m/%%Y
# Default: %%d.%%m.%%Y, %%H:%%M:%%S
# Optional
include_snapshot: False
# !! SUPPORTER ONLY - This feature requires beeing a supporter of Mobileraker as of now!
# Include a snapshot of the webcam in any print status/progress update notifications
# Default: True
# Optional

# Add a [printer ...] section for every printer you want to add
[printer Ender 3 S1 Plus]
moonraker_uri: ws://192.168.123.12:7125/websocket
# Define the uri to the moonraker instance.
# Default value: ws://127.0.0.1:7125/websocket
# Optional
moonraker_api_key: False
# Moonraker API key if force_logins or trusted clients is active!
# Default value: False
# Optional
snapshot_uri: http://192.168.123.12/webcam/?action=snapshot
# !! SUPPORTER ONLY - This feature requires beeing a supporter of Mobileraker as of now!
# The ABSOLUT url to the webcam, the companion should make a screenshot of.
# Default:
# Optional
snapshot_rotation: 0
# The rotation applied to the image. Valid values : 0, 90, 180, 270
# Default: 0
# Optional
```

$ docker logs c8a6423718fa
```
2023-11-29 21:08:03 root INFO MobilerakerCompanion version: ?
2023-11-29 21:08:03 root INFO Passed config file is: printer_data/config/mobileraker.conf
2023-11-29 21:08:03 root INFO Checking mobileraker_companion dir
[... 389 lines cut ...]
            "Networks": {
                "mobileraker_companion_default": {
                    "IPAMConfig": null,
                    "Links": null,
                    "Aliases": [
                        "mobileraker_companion-mobileraker_companion-1",
                        "mobileraker_companion",
                        "c8a6423718fa"
                    ],
                    "NetworkID": "f180c01181a87aaf9f420c02332e0dabac1c48383e4a01d230090616ed10f62e",
                    "EndpointID": "5d65d121ac3ac7adf54638a20fd3d6af8e530098170e11a779b277cbbd3f6868",
                    "Gateway": "172.19.0.1",
                    "IPAddress": "172.19.0.2",
                    "IPPrefixLen": 16,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "02:42:ac:13:00:02",
                    "DriverOpts": null
                }
            }
        }
    }
]
```

PatrickGlatz, 2023-11-30:
@Clon1998 FANTASTIC. I have simply replaced the wrong side of the volumes definition. So finally it ends up with this:

`cat /home/user/mobileraker_companion/docker-compose.yml`
```
services:
  mobileraker_companion:
    image: ghcr.io/clon1998/mobileraker_companion:latest
    volumes:
      - /home/user/printer_data/config/mobileraker.conf:/opt/printer_data/config/mobileraker.conf
 ```

I did a 
`docker down``
followed by
`docker compose up --build -d``

And directly my iPhone made "PING" :)

Thanks for pointing me in the right direction!

Regards and thanks for this tool!
Patrick

closed 2023-11-30 by PatrickGlatz