Clon1998/mobileraker_companion #66: Extra config search path

jelleh, 2023-11-30:
-snip-
The Companion searches for a Mobileraker.conf file in the following locations (in order of precedence):

1. ~/Mobileraker.conf
2. <mobileraker_companion DIR>/mobileraker.conf
3. ~/klipper_config/mobileraker.conf
-snip-

A lot of installs/howtos/kiauh use the following location to collect al the configs: ~/printer_data/config/

Maybe add ~/printer_data/config/mobileraker.conf ?

Just a suggestion.

Clon1998, 2023-11-30:
> -snip-
>
> The Companion searches for a Mobileraker.conf file in the following locations (in order of precedence):
>
>
>
> 1. ~/Mobileraker.conf
>
> 2. <mobileraker_companion DIR>/mobileraker.conf
>
> 3. [...]

I'll need to double check that. However, I think it already also looks for into the printer data folder?

Clon1998, 2023-11-30:
It looks at both the old klipper_config and new printer_data folders:

https://github.com/Clon1998/mobileraker_companion/blob/4d3666483bcf0a8b319384219c94b8354880906a/mobileraker/util/configs.py#L12-L17