Clon1998/mobileraker_companion #68: Error during install regarding "building wheel for backports.zoneinfo"

DDen87, 2023-12-04:
Hello @Clon1998,

During install I encounter the following issue: 
_Using cached certifi-2023.11.17-py3-none-any.whl (162 kB)
Using cached idna-3.6-py3-none-any.whl (61 kB)
Using cached urllib3-1.26.18-py2.py3-none-any.whl (143 kB)
Building wheels for collected packages: backports.zoneinfo
  error: subprocess-exited-with-error
  
  × Building wheel for backports.zoneinfo (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for backports.zoneinfo (pyproject.toml) ... error
  ERROR: Failed building wheel for backports.zoneinfo
Failed to build backports.zoneinfo
ERROR: Could not build wheels for backports.zoneinfo, which is required to install pyproject.toml-based projects_

Is this something I could resolve myself? Please keep in mind I'm a total newbie....

Clon1998, 2023-12-05:
Related to #22

Clon1998, 2023-12-09:
Hey,
I had a look at the changes that Quinn made to allow [OE](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/tree/master) to work on K1 and Sonic Pads.
I will need to adapt a lot of code to get this working/to install.
I can not promise any timeline, but I hope I get this done by the end of December.

DDen87, 2023-12-09:
@Clon1998 thanks for the effort! If I can help in any way please let me know.

Clon1998, 2024-01-14:
Hey,
I finally had some time to look into the stuff that Quin did for OE. 
The new installer that should support an installation on the K1 is currently available on the branch `installer`.

Can you do me a favor and try to run the install script? The K1 I ordered especially to test it was damaged during delivery...

To run the new installation. 
1. Go into the folder of the repo `mobileraker_companion`
2. Switch to the new branch:
```bash
cd ~/mobileraker_companion
git pull
git checkout installer
```
3. Run the new installer 
```bash
./scripts/install.sh
```

Please let me know if you encounter any problem during installation or if the companion does not start afterwards. Please note that currently the companion might not be shown as service in the UI (Fluidd/mainsail) for restart/stopping.

DDen87, 2024-01-15:
Hey @Clon1998,

I tried in the Creality account on the sonic pad an unfortunately ran into the following error while installing:
_creality@spad-7342:~$ cd ~/mobileraker_companion
creality@spad-7342:~/mobileraker_companion$ git pull
Already up to date.
creality@spad-7342:~/mobileraker_companion$ git checkout installer
Already on 'installer'
Your branch is up to date with 'origin/installer'.
creality@spad-7342:~/mobileraker_companion$ ./scripts/install.sh
./scripts/install.sh: line 7: of: command not found

**(DDen87: I removed the picture that is in the terminal)**

                  Mobileraker Companion
 Companion for Mobileraker, enabling push notification for Klipper using Moonraker.
 - Installer Script based on OctoEverywhere for Klipper, courtesy of Quinn Damerell

Mobileraker works as a simple UI for Klipper on the phone.
The companion is required to enable reliable push notifications.
  - Print State Notifications
  - Progress Notifications
  - Custom Notifications via M117 or custom Macro enabling layer based notifications or heat up notifications.

Running in Sonic Pad OS mode
Checking required system packages are installed...
Package python3 (3.7.8-3) installed in root is up to date.
Package python3-pip (20.1.1-1) installed in root is up to date.
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: virtualenv in /usr/lib/python3.7/site-packages (20.25.0)
Requirement already satisfied: importlib-metadata>=6.6; python_version < "3.8" in /usr/lib/python3.7/site-packages (from virtualenv) (6.7.0)
Requirement already satisfied: distlib<1,>=0.3.7 in /usr/lib/python3.7/site-packages (from virtualenv) (0.3.8)
Requirement already satisfied: filelock<4,>=3.12.2 in /usr/lib/python3.7/site-packages (from virtualenv) (3.12.2)
Requirement already satisfied: platformdirs<5,>=3.9.1 in /usr/lib/python3.7/site-packages (from virtualenv) (4.0.0)
Requirement already satisfied: typing-extensions>=3.6.4; python_version < "3.8" in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (4.7.1)
Requirement already satisfied: zipp>=0.5 in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (3.15.0)
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/share/.local/lib'
Check the permissions._

So I tried in root access as well but also ran into an issue there: 
_root@spad-7342:~# cd ~/mobileraker_companion
root@spad-7342:~/mobileraker_companion# git pull
Already up to date.
root@spad-7342:~/mobileraker_companion# git checkout installer
Branch 'installer' set up to track remote branch 'installer' from 'origin'.
Switched to a new branch 'installer'
root@spad-7342:~/mobileraker_companion# ./scripts/install.sh
./scripts/install.sh: line 7: of: command not found

**(DDen87: again I removed the picture)**

                  Mobileraker Companion
 Companion for Mobileraker, enabling push notification for Klipper using Moonraker.
 - Installer Script based on OctoEverywhere for Klipper, courtesy of Quinn Damerell

Mobileraker works as a simple UI for Klipper on the phone.
The companion is required to enable reliable push notifications.
  - Print State Notifications
  - Progress Notifications
  - Custom Notifications via M117 or custom Macro enabling layer based notifications or heat up notifications.

Running in Sonic Pad OS mode
Current path /root/mobileraker_companion

For the Creality devices the mobileraker_companion repo must be cloned into /usr/share/mobileraker_companion

Moving the repo and running the install again...
HEAD is now at 7297fbc Ensure eval is also triggered in case timelapse pause -> error
Already on 'main'
Your branch is up to date with 'origin/main'.
Already up to date.
./scripts/install.sh: line 139: ./install.sh: No such file or directory
ash: getcwd: No such file or directory_

Am I doing something wrong here?

Clon1998, 2024-01-15:
> Hey @Clon1998,
>
> I tried in the Creality account on the sonic pad an unfortunately ran into the following error while installing:
>
> ```
> _creality@spad-7342:~$ cd ~/mobileraker_companion [...]

Hey,
not at all. It seems like I have an issue in the install script since I did not expect yet to test it on the Sonic Pad.

For the Sonic Pad you will need to run the script in sudo/root.
Can you try to re-run the script since I still have an issue with it?

Clon1998, 2024-01-15:
Actually,I just noticed that the install script switches back to the main branch. Please rerun the git commands and then the install script as root.

DDen87, 2024-01-16:
@Clon1998 , I went into root.
Did the install as per below, unfortunately again a error saying **"./scripts/install.sh: line 140: ./scripts/install.sh: No such file or directory
ash: getcwd: No such file or directory"** 

root@spad-7342:~# cd ~/mobileraker_companion
-ash: cd: can't cd to /root/mobileraker_companion
root@spad-7342:~# git pull
fatal: not a git repository (or any of the parent directories): .git
root@spad-7342:~# git checkout installer
fatal: not a git repository (or any of the parent directories): .git
root@spad-7342:~# cd ~/mobileraker_companion
-ash: cd: can't cd to /root/mobileraker_companion
root@spad-7342:~# cd ~/
root@spad-7342:~# git clone https://github.com/Clon1998/mobileraker_companion.gi
t
Cloning into 'mobileraker_companion'...
remote: Enumerating objects: 799, done.
remote: Counting objects: 100% (185/185), done.
remote: Compressing objects: 100% (70/70), done.
remote: Total 799 (delta 143), reused 140 (delta 115), pack-reused 614
Receiving objects: 100% (799/799), 501.29 KiB | 463.00 KiB/s, done.
Resolving deltas: 100% (488/488), done.
root@spad-7342:~# cd mobileraker_companion
root@spad-7342:~/mobileraker_companion# git pull
Already up to date.
root@spad-7342:~/mobileraker_companion# git checkout installer
Branch 'installer' set up to track remote branch 'installer' from 'origin'.
Switched to a new branch 'installer'
root@spad-7342:~/mobileraker_companion# ./scripts/install.sh
./scripts/install.sh: line 7: of: command not found

MMMMMMMMMMMMMMMMMMMMMMWNX0xolldk0XWWMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMWNKkdl:;;:;,,',:ldk0XWMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMWNX0xoc;,;:ldxxxxdoc:,'',:lxOKNWWMMMMMMMMMM
MMMMMMMMMWNKOxl:;,;codxkkdooddxxkkdol:;''';coxOXWMMMMMMM
MMMMMMWXOdl:,,:loxkkkdoloodxkOOkxxxxkkxdlc;''',:lkXWMMMM
MMMMMWk:,,,,,:x0OdoloddxxxddxkkOOOOkxddxkOOl''''',cOWMMM
MMMMMWOc,,,,,,:lddxkkxdollllodddxxkO000kkxo:'''''':OWMMM
MMMMWXOkxo:;,,,,,;loddooodxO0KK0Okkkkkdl;,'''',:odkOXWMM
MMMWXo;:ldxxdl:,,',,,:clx0NWWWWNKkoc;,''''';ldxxoc;,lKMM
MMMMNx:,,,;coxkxoc;,,,,,;cokkkdl;,''''',:odxdl:,''':kNMM
MMMWXK0xl:,,,,:lxkkdl:,,,,,,'''''''';ldxxoc,'''';lx0KXWM
MMNkc:ok00koc,,,,;cdkkxoc;,''''',codxdl:,'''';lxOkdc;l0W
MMXo,',,;lxO0kdc;,,,;:oxOkxlccldxxoc,'''',:lxkko:,''',xW
MMWKo:,,,,,;ldO0Odl;,,,,;ldkkxdl;,'''',:oxkxo:,''''';dKW
MMMMNKxl;,,,,,;cdO00xo:,,,,,,,''''',coxkdl;'''''';lkKWMM
MMMMMMWNKxl:,,,,,,:ok00koc;,'''';ldkkdc,'''''';lxKWMMMMM
MMMMMMMMMMNKxo:,,,,,,:ok00Odlcoxkxo:,'''''';lkKNMMMMMMMM
MMMMMMMMMMMMWNKkl:,,,,,,;lxOOkxl:,'''''';lkKWMMMMMMMMMMM
MMMMMMMMMMMMMMMWNKkl:,,,,,,;;,''''''';lkKWMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMWNKko:,,,,'''''';lkKWMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMWNKko:,'',;lkKWMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMWKkddkKWMMMMMMMMMMMMMMMMMMMMMMM

                  Mobileraker Companion
 Companion for Mobileraker, enabling push notification for Klipper using Moonraker.
 - Installer Script based on OctoEverywhere for Klipper, courtesy of Quinn Damerell

Mobileraker works as a simple UI for Klipper on the phone.
The companion is required to enable reliable push notifications.
  - Print State Notifications
  - Progress Notifications
  - Custom Notifications via M117 or custom Macro enabling layer based notifications or heat up notifications.

Running in Sonic Pad OS mode
Current path /root/mobileraker_companion

For the Creality devices the mobileraker_companion repo must be cloned into /usr/share/mobileraker_companion

Moving the repo and running the install again...
HEAD is now at 7297fbc Ensure eval is also triggered in case timelapse pause -> error
remote: Enumerating objects: 66, done.
remote: Counting objects: 100% (66/66), done.
remote: Compressing objects: 100% (43/43), done.
remote: Total 66 (delta 40), reused 49 (delta 23), pack-reused 0
Unpacking objects: 100% (66/66), 27.17 KiB | 14.00 KiB/s, done.
From https://github.com/Clon1998/mobileraker_companion
   a20cd35..8c3d185  installer  -> origin/installer
Already up to date.
./scripts/install.sh: line 140: ./scripts/install.sh: No such file or directory
ash: getcwd: No such file or directory
root@spad-7342:(unknown)#

DDen87, 2024-01-17:
Hey @Clon1998,

Did it again in root and still got some "no such file or directory".:
Moving the repo and running the install again...
HEAD is now at 7297fbc Ensure eval is also triggered in case timelapse pause -> error
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 8 (delta 6), reused 8 (delta 6), pack-reused 0
Unpacking objects: 100% (8/8), 828 bytes | 3.00 KiB/s, done.
From https://github.com/Clon1998/mobileraker_companion
   8c3d185..7a9dbfd  installer  -> origin/installer
Already up to date.
Current path /usr/share/mobileraker_companion
./scripts/install.sh: line 141: /usr/share/mobileraker_companion/scripts/install.sh: **No such file or directory**
Cleaning up the old repo folder. The new repo is in /usr/share/mobileraker_companion
ash: getcwd: **No such file or directory**
root@spad-7342:(unknown)#

Clon1998, 2024-01-17:
Thanks,
can you give me the output of this bash commands:

```bash
cd /usr/share/mobileraker_companion
ls -l
cd /scripts
ls -l
```

I want to see if the install.sh file is missing in the new repo folder...

DDen87, 2024-01-17:
root@spad-7342:/usr/share/mobileraker_companion# cd /usr/share/mobileraker_compa
nion
root@spad-7342:/usr/share/mobileraker_companion# ls -l
-rw-r--r--    1 root     root           828 Jan 15 18:41 Dockerfile
-rw-r--r--    1 root     root          1072 Jan 15 18:41 LICENSE
-rw-r--r--    1 root     root          8371 Jan 15 18:41 README.md
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 assets
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 docs
drwxr-xr-x    6 root     root          1024 Jan 15 18:41 mobileraker
-rw-r--r--    1 root     root          2887 Jan 15 18:41 mobileraker.py
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 scripts
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 tests
root@spad-7342:/usr/share/mobileraker_companion# cd /scripts
-ash: cd: can't cd to /scripts
root@spad-7342:/usr/share/mobileraker_companion# ls -l
-rw-r--r--    1 root     root           828 Jan 15 18:41 Dockerfile
-rw-r--r--    1 root     root          1072 Jan 15 18:41 LICENSE
-rw-r--r--    1 root     root          8371 Jan 15 18:41 README.md
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 assets
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 docs
drwxr-xr-x    6 root     root          1024 Jan 15 18:41 mobileraker
-rw-r--r--    1 root     root          2887 Jan 15 18:41 mobileraker.py
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 scripts
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 tests
root@spad-7342:/usr/share/mobileraker_companion#

Clon1998, 2024-01-17:
Thanks,
and sorry the 2nd part was wrong: please once again :(
```bash
cd /usr/share/mobileraker_companion
ls -l
cd /scripts
ls -l

DDen87, 2024-01-17:
No problem!

see below:
root@spad-7342:/usr/share/mobileraker_companion# cd /usr/share/mobileraker_compa
nion
root@spad-7342:/usr/share/mobileraker_companion# ls -l
-rw-r--r--    1 root     root           828 Jan 15 18:41 Dockerfile
-rw-r--r--    1 root     root          1072 Jan 15 18:41 LICENSE
-rw-r--r--    1 root     root          8371 Jan 15 18:41 README.md
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 assets
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 docs
drwxr-xr-x    6 root     root          1024 Jan 15 18:41 mobileraker
-rw-r--r--    1 root     root          2887 Jan 15 18:41 mobileraker.py
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 scripts
drwxr-xr-x    2 root     root          1024 Jan 15 18:41 tests
root@spad-7342:/usr/share/mobileraker_companion# cd scripts
root@spad-7342:/usr/share/mobileraker_companion/scripts# ls -l
-rwxr-xr-x    1 root     root           116 Jan 15 18:41 fix-log-location.sh
-rwxr-xr-x    1 root     root          4749 Jan 15 18:41 install-mobileraker-companion.sh
-rw-r--r--    1 root     root            95 Jan 15 18:41 mobileraker-requirements.txt
-rwxr-xr-x    1 root     root          1131 Jan 15 18:41 remove-mobileraker-companion.sh
root@spad-7342:/usr/share/mobileraker_companion/scripts#

Clon1998, 2024-01-17:
> No problem!
>
> see below: root@spad-7342:/usr/share/mobileraker_companion# cd /usr/share/mobileraker_compa nion root@spad-7342:/usr/share/mobileraker_companion# ls -l -rw-r--r-- 1 root root 828 Jan 15 [...]

Thanks! There is the issue. Since we are using a branch right now, this part of the script is not working since it is on the `main` branch rather than the `installer` one.

So lets not worry about that! You should be able to run the installer directly via this commands. It switches the branch and calls the script:
```bash
cd /usr/share/mobileraker_companion
git pull
git checkout installer
./scripts/install.sh
```

DDen87, 2024-01-17:
Hey @Clon1998 this really went ok until this:
Multiple Moonraker Instances Detected.
Mobileraker Companion can manage multiple printers.
However, only one installation of the Companion is required.
If a Mobile Companion instance is already installed, the Installer will adjust the companions config to add the selected Moonraker instance if needed.

Sonic Pad/K1 Users - If you're only using one printer, select number 1

  1) moonraker_service [/mnt/UDISK/printer_config/moonraker.conf]
  2) moonraker_service.2 [/mnt/UDISK/printer_config2/moonraker.conf]
  3) moonraker_service.3 [/mnt/UDISK/printer_config3/moonraker.conf]
  4) moonraker_service.4 [/mnt/UDISK/printer_config4/moonraker.conf]

Enter the number for the config you would like to setup now: 1
Moonraker instance selected! [moonraker_service:/mnt/UDISK/printer_config/moonraker.conf]
Starting configuration...

Installer failed - Printer data folder path was not set.

Stack Trace:
Traceback (most recent call last):
  File "/usr/share/mobileraker_companion/installer/Installer.py", line 20, in Run
    self._run()
  File "/usr/share/mobileraker_companion/installer/Installer.py", line 124, in _run
    configure.run(context)
  File "/usr/share/mobileraker_companion/installer/Configure.py", line 43, in run
    context.printer_data_folder = os.path.join(Util.parent_dir(context.printer_data_folder), "mobileraker_data")
  File "/usr/share/mobileraker_companion/installer/Context.py", line 335, in printer_data_folder
    raise AttributeError("Printer data folder path was not set.")
AttributeError: Printer data folder path was not set.

Please contact our support team directly at support@octoeverywhere.com so we can help you fix this issue!

DDen87, 2024-01-17:
Hahahaha no problem! I'm happy to help!

See below:
root@spad-7342:/usr/share/mobileraker_companion# ./scripts/install.sh -debug

MMMMMMMMMMMMMMMMMMMMMMWNX0xolldk0XWWMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMWNKkdl:;;:;,,',:ldk0XWMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMWNX0xoc;,;:ldxxxxdoc:,'',:lxOKNWWMMMMMMMMMM
MMMMMMMMMWNKOxl:;,;codxkkdooddxxkkdol:;''';coxOXWMMMMMMM
MMMMMMWXOdl:,,:loxkkkdoloodxkOOkxxxxkkxdlc;''',:lkXWMMMM
MMMMMWk:,,,,,:x0OdoloddxxxddxkkOOOOkxddxkOOl''''',cOWMMM
MMMMMWOc,,,,,,:lddxkkxdollllodddxxkO000kkxo:'''''':OWMMM
MMMMWXOkxo:;,,,,,;loddooodxO0KK0Okkkkkdl;,'''',:odkOXWMM
MMMWXo;:ldxxdl:,,',,,:clx0NWWWWNKkoc;,''''';ldxxoc;,lKMM
MMMMNx:,,,;coxkxoc;,,,,,;cokkkdl;,''''',:odxdl:,''':kNMM
MMMWXK0xl:,,,,:lxkkdl:,,,,,,'''''''';ldxxoc,'''';lx0KXWM
MMNkc:ok00koc,,,,;cdkkxoc;,''''',codxdl:,'''';lxOkdc;l0W
MMXo,',,;lxO0kdc;,,,;:oxOkxlccldxxoc,'''',:lxkko:,''',xW
MMWKo:,,,,,;ldO0Odl;,,,,;ldkkxdl;,'''',:oxkxo:,''''';dKW
MMMMNKxl;,,,,,;cdO00xo:,,,,,,,''''',coxkdl;'''''';lkKWMM
MMMMMMWNKxl:,,,,,,:ok00koc;,'''';ldkkdc,'''''';lxKWMMMMM
MMMMMMMMMMNKxo:,,,,,,:ok00Odlcoxkxo:,'''''';lkKNMMMMMMMM
MMMMMMMMMMMMWNKkl:,,,,,,;lxOOkxl:,'''''';lkKWMMMMMMMMMMM
MMMMMMMMMMMMMMMWNKkl:,,,,,,;;,''''''';lkKWMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMWNKko:,,,,'''''';lkKWMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMWNKko:,'',;lkKWMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMWKkddkKWMMMMMMMMMMMMMMMMMMMMMMM

                  Mobileraker Companion
 Companion for Mobileraker, enabling push notification for Klipper using Moonraker.
 - Installer Script based on OctoEverywhere for Klipper, courtesy of Quinn Damerell

Mobileraker works as a simple UI for Klipper on the phone.
The companion is required to enable reliable push notifications.
  - Print State Notifications
  - Progress Notifications
  - Custom Notifications via M117 or custom Macro enabling layer based notifications or heat up notifications.

Running in Sonic Pad OS mode
Checking required system packages are installed...
Package python3 (3.7.8-3) installed in root is up to date.
Package python3-pip (20.1.1-1) installed in root is up to date.
Requirement already satisfied: virtualenv in /usr/lib/python3.7/site-packages (20.25.0)
Requirement already satisfied: distlib<1,>=0.3.7 in /usr/lib/python3.7/site-packages (from virtualenv) (0.3.8)
Requirement already satisfied: importlib-metadata>=6.6; python_version < "3.8" in /usr/lib/python3.7/site-packages (from virtualenv) (6.7.0)
Requirement already satisfied: platformdirs<5,>=3.9.1 in /usr/lib/python3.7/site-packages (from virtualenv) (4.0.0)
Requirement already satisfied: filelock<4,>=3.12.2 in /usr/lib/python3.7/site-packages (from virtualenv) (3.12.2)
Requirement already satisfied: zipp>=0.5 in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (3.15.0)
Requirement already satisfied: typing-extensions>=3.6.4; python_version < "3.8" in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (4.7.1)

System package install complete.
Checking Python Virtual Environment For Mobileraker Companion...
Updating PIP if needed... (this can take a few seconds or so)
Requirement already satisfied: pip in /usr/share/mobileraker-env/lib/python3.7/site-packages (23.3.2)
Installing or updating required python libs...
Python libs installed.
Bootstrap done. Starting python installer...
Starting Installer...
Os Type Detected: PlatformType.SONIC_PAD
DEBUG: Found config: {"REPO_DIR":"/usr/share/mobileraker_companion","ENV_DIR":"/usr/share/mobileraker-env","USERNAME":"root","USER_HOME":"/usr/share","CMD_LINE_ARGS":"-debug"}
DEBUG: Validating args
DEBUG: Starting discovery.
DEBUG: Discovery - Service Files
DEBUG:  /etc/init.d/S99swupdate_autorun
DEBUG:  /etc/init.d/adbd
DEBUG:  /etc/init.d/audio_setting
DEBUG:  /etc/init.d/avahi-daemon
DEBUG:  /etc/init.d/boot
DEBUG:  /etc/init.d/browser
DEBUG:  /etc/init.d/cron
[... 59 lines cut ...]
  1) moonraker_service [/mnt/UDISK/printer_config/moonraker.conf]
  2) moonraker_service.2 [/mnt/UDISK/printer_config2/moonraker.conf]
  3) moonraker_service.3 [/mnt/UDISK/printer_config3/moonraker.conf]
  4) moonraker_service.4 [/mnt/UDISK/printer_config4/moonraker.conf]

Enter the number for the config you would like to setup now: 1
Moonraker instance selected! [moonraker_service:/mnt/UDISK/printer_config/moonraker.conf]
Starting configuration...
DEBUG: Moonraker Service File Name: moonraker_service

Installer failed - Printer data folder path was not set.

Stack Trace:
Traceback (most recent call last):
  File "/usr/share/mobileraker_companion/installer/Installer.py", line 20, in Run
    self._run()
  File "/usr/share/mobileraker_companion/installer/Installer.py", line 124, in _run
    configure.run(context)
  File "/usr/share/mobileraker_companion/installer/Configure.py", line 43, in run
    context.printer_data_folder = os.path.join(Util.parent_dir(context.printer_data_folder), "mobileraker_data")
  File "/usr/share/mobileraker_companion/installer/Context.py", line 335, in printer_data_folder
    raise AttributeError("Printer data folder path was not set.")
AttributeError: Printer data folder path was not set.

Please contact our support team directly at support@octoeverywhere.com so we can help you fix this issue!

Clon1998, 2024-01-18:
> Hahahaha no problem! I'm happy to help!
> 
> See below:
> 
> ```shell
> root@spad-7342:/usr/share/mobileraker_companion# ./scripts/install.sh -debug
> 
> 
> 
> MMMMMMMMMMMMMMMMMMMMMMWNX0xolldk0XWWMMMMMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMWNKkdl:;;:;,,',:ldk0XWMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMWNX0xoc;,;:ldxxxxdoc:,'',:lxOKNWWMMMMMMMMMM
> MMMMMMMMMWNKOxl:;,;codxkkdooddxxkkdol:;''';coxOXWMMMMMMM
> MMMMMMWXOdl:,,:loxkkkdoloodxkOOkxxxxkkxdlc;''',:lkXWMMMM
> MMMMMWk:,,,,,:x0OdoloddxxxddxkkOOOOkxddxkOOl''''',cOWMMM
> MMMMMWOc,,,,,,:lddxkkxdollllodddxxkO000kkxo:'''''':OWMMM
> MMMMWXOkxo:;,,,,,;loddooodxO0KK0Okkkkkdl;,'''',:odkOXWMM
> MMMWXo;:ldxxdl:,,',,,:clx0NWWWWNKkoc;,''''';ldxxoc;,lKMM
> MMMMNx:,,,;coxkxoc;,,,,,;cokkkdl;,''''',:odxdl:,''':kNMM
> MMMWXK0xl:,,,,:lxkkdl:,,,,,,'''''''';ldxxoc,'''';lx0KXWM
> MMNkc:ok00koc,,,,;cdkkxoc;,''''',codxdl:,'''';lxOkdc;l0W
> MMXo,',,;lxO0kdc;,,,;:oxOkxlccldxxoc,'''',:lxkko:,''',xW
> MMWKo:,,,,,;ldO0Odl;,,,,;ldkkxdl;,'''',:oxkxo:,''''';dKW
> MMMMNKxl;,,,,,;cdO00xo:,,,,,,,''''',coxkdl;'''''';lkKWMM
> MMMMMMWNKxl:,,,,,,:ok00koc;,'''';ldkkdc,'''''';lxKWMMMMM
> MMMMMMMMMMNKxo:,,,,,,:ok00Odlcoxkxo:,'''''';lkKNMMMMMMMM
> MMMMMMMMMMMMWNKkl:,,,,,,;lxOOkxl:,'''''';lkKWMMMMMMMMMMM
> MMMMMMMMMMMMMMMWNKkl:,,,,,,;;,''''''';lkKWMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMWNKko:,,,,'''''';lkKWMMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMMMMWNKko:,'',;lkKWMMMMMMMMMMMMMMMMMMMM
> MMMMMMMMMMMMMMMMMMMMMMMMMWKkddkKWMMMMMMMMMMMMMMMMMMMMMMM
> 
>                   Mobileraker Companion
>  Companion for Mobileraker, enabling push notification for Klipper using Moonraker.
>  - Installer Script based on OctoEverywhere for Klipper, courtesy of Quinn Damerell
> 
> 
> Mobileraker works as a simple UI for Klipper on the phone.
> The companion is required to enable reliable push notifications.
>   - Print State Notifications
>   - Progress Notifications
>   - Custom Notifications via M117 or custom Macro enabling layer based notifications or heat up notifications.
> 
> 
> Running in Sonic Pad OS mode
> Checking required system packages are installed...
> Package python3 (3.7.8-3) installed in root is up to date.
> Package python3-pip (20.1.1-1) installed in root is up to date.
> Requirement already satisfied: virtualenv in /usr/lib/python3.7/site-packages (20.25.0)
> Requirement already satisfied: distlib<1,>=0.3.7 in /usr/lib/python3.7/site-packages (from virtualenv) (0.3.8)
> Requirement already satisfied: importlib-metadata>=6.6; python_version < "3.8" in /usr/lib/python3.7/site-packages (from virtualenv) (6.7.0)
> Requirement already satisfied: platformdirs<5,>=3.9.1 in /usr/lib/python3.7/site-packages (from virtualenv) (4.0.0)
> Requirement already satisfied: filelock<4,>=3.12.2 in /usr/lib/python3.7/site-packages (from virtualenv) (3.12.2)
> Requirement already satisfied: zipp>=0.5 in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (3.15.0)
> Requirement already satisfied: typing-extensions>=3.6.4; python_version < "3.8" in /usr/lib/python3.7/site-packages (from importlib-metadata>=6.6; python_version < "3.8"->virtualenv) (4.7.1)
> 
> System package install complete.
> Checking Python Virtual Environment For Mobileraker Companion...
> Updating PIP if needed... (this can take a few seconds or so)
> Requirement already satisfied: pip in /usr/share/mobileraker-env/lib/python3.7/site-packages (23.3.2)
> Installing or updating required python libs...
> Python libs installed.
> Bootstrap done. Starting python installer...
> Starting Installer...
> Os Type Detected: PlatformType.SONIC_PAD
> DEBUG: Found config: {"REPO_DIR":"/usr/share/mobileraker_companion","ENV_DIR":"/usr/share/mobileraker-env","USERNAME":"root","USER_HOME":"/usr/share","CMD_LINE_ARGS":"-debug"}
> DEBUG: Validating args
> DEBUG: Starting discovery.
> DEBUG: Discovery - Service Files
> DEBUG:  /etc/init.d/S99swupdate_autorun
[... 73 lines cut ...]
> Starting configuration...
> DEBUG: Moonraker Service File Name: moonraker_service
> 
> 
> Installer failed - Printer data folder path was not set.
> 
> 
> Stack Trace:
> Traceback (most recent call last):
>   File "/usr/share/mobileraker_companion/installer/Installer.py", line 20, in Run
>     self._run()
>   File "/usr/share/mobileraker_companion/installer/Installer.py", line 124, in _run
>     configure.run(context)
>   File "/usr/share/mobileraker_companion/installer/Configure.py", line 43, in run
>     context.printer_data_folder = os.path.join(Util.parent_dir(context.printer_data_folder), "mobileraker_data")
>   File "/usr/share/mobileraker_companion/installer/Context.py", line 335, in printer_data_folder
>     raise AttributeError("Printer data folder path was not set.")
> AttributeError: Printer data folder path was not set.
> 
> 
> 
> Please contact our support team directly at support@octoeverywhere.com so we can help you fix this issue!
> ```

Sorry for my late reply. I will have a look later today or during the weekend and figure out what is going on here :)

DDen87, 2024-01-18:
@Clon1998  Thanks!

Clon1998, 2024-01-19:
@DDen87 
Just fixed the installer. Please pull once again and re-run the installer in the path I mentioned before.

DDen87, 2024-01-20:
@Clon1998 I did like you mentioned:
cd /usr/share/mobileraker_companion
git pull
git checkout installer
./scripts/install.sh

Installer went ok until I got another error right after I needed to select the amount of printers, please see below:
Sonic Pad/K1 Users - If you're only using one printer, select number 1

  1) moonraker_service [/mnt/UDISK/printer_config/moonraker.conf]
  2) moonraker_service.2 [/mnt/UDISK/printer_config2/moonraker.conf]
  3) moonraker_service.3 [/mnt/UDISK/printer_config3/moonraker.conf]
  4) moonraker_service.4 [/mnt/UDISK/printer_config4/moonraker.conf]

Enter the number for the config you would like to setup now: 1
Moonraker instance selected! [moonraker_service:/mnt/UDISK/printer_config/moonraker.conf]
Starting configuration...
Enuring path and permissions [/mnt/UDISK/mobileraker_data]...
Dir doesn't exist, creating...

Installer failed - 'Context' object has no attribute 'UserName'

Stack Trace:
Traceback (most recent call last):
  File "/usr/share/mobileraker_companion/installer/Installer.py", line 20, in Run
    self._run()
  File "/usr/share/mobileraker_companion/installer/Installer.py", line 124, in _run
    configure.run(context)
  File "/usr/share/mobileraker_companion/installer/Configure.py", line 44, in run
    Util.ensure_dir_exists(context.printer_data_folder, context, True)
  File "/usr/share/mobileraker_companion/installer/Util.py", line 59, in ensure_dir_exists
    Logger.Info("Setting owner permissions to the service user ["+context.UserName+"]...")
AttributeError: 'Context' object has no attribute 'UserName'

Please contact our support team directly at support@octoeverywhere.com so we can help you fix this issue!

Clon1998, 2024-01-20:
Once again :)

DDen87, 2024-01-20:
@Clon1998 , it worked!!!
Thanks for your help!
  ~~~ Mobileraker Companion installation Complete ~~~
                   Happy Printing!
                          <3

Clon1998, 2024-01-20:
> @Clon1998 , it worked!!!
> 
> Thanks for your help!
> 
>   ~~~ Mobileraker Companion installation Complete ~~~
> 
>                    Happy Printing!
> 
>                           <3
> 
> 

Awesome!
Thanks for testing the Installer on the SonicPad.

Note that there might be some more changes in the future. E.g. the companion is currently not added to the options to restart/stop alongside klipper, moonraker...

PR #77 (2024-02-24): Using new installer
Using the Installer by OE.
This ensures that the companion can be installed on K1 and SonicPad.
It also simplifies the installation for multi-printer setups.

Closes #68, Closes #48, Closes #21
closed 2024-02-24 by Clon1998