Clon1998/mobileraker_companion #71: Remote acces through tailscale

Tollie46, 2023-12-23:
I have installed mobileraker and it works via WiFi, when I try to connect via tailscale I get the following error message jRpcError(code: -32602, message: Unauthorized}

Clon1998, 2023-12-23:
You might need to add the api key from Moonraker. Otherwise please post the log file here so I can have a more in depth look.

Tollie46, 2023-12-24:
okay, where can I find the api key? and which log file do you need moonraker? 
By the way, I followed this video https://youtu.be/vq-VE-9t13k?si=nVBL9v_Y5Cw95msm

mcristina422, 2023-12-30:
In your `moonraker.conf`, do you have the tailscale IP range defined? `100.64.0.0/10`

For example, my section looks like this
```
[authorization]
trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128
    100.64.0.0/10
```

https://tailscale.com/kb/1015/100.x-addresses