Clon1998/mobileraker_companion #73: Disable Notifications for Timelapse parking

Johnnytk36, 2023-12-29:
I just recently re-enabled time-lapse and I am getting a notification every time my printer goes to park for a time-lapse, because it uses the pause and resume command.

Is there a easy way to get the mobileraker companion to ignore them?

I have the companion installed on my NAS using docker and using a Sonic Pad with a Ender 5 S-1.

I might look at rewriting the timelapse macros not to use the standard pause and resume macros, but other than that, am I missing something easy that I didn't setup right?

[image]

Clon1998, 2023-12-29:
Hey,
you are able to disable the pause state in the App Settings. Afterward, you can modify the normal pause macro and add a custom notification trigger via M117/MR_NOTIFY.

Johnnytk36, 2023-12-29:
Thank you. I totally missed that option to select that in the actual app.

I might just go and modify the filament runout gcode to add a custom notification after runout is detected, as that is really the only pause command I am interested in, if I am not there

Thanks!

closed 2023-12-29 by Clon1998

Clon1998, 2024-01-02:
Hey @Johnnytk36, I just pushed an update to the companion that will allow users to keep the pause notification active while all timelapse pauses will be ignored.
Can you do me a favor and test that for me?

Johnnytk36, 2024-01-02:
@Clon1998 I will tonight on my next print.  I need to test my custom out of filament notification also.

Johnnytk36, 2024-01-03:
I updated the mobileraker companion using the docker command after shutting down the dock.
`sudo docker-compose pull && sudo docker-compose up -d` 

Note: I need sudo for it to work on my NAS. 

I then started a new print I am still getting the notifications.

[image]

commit 322b4d8a (2024-01-02): Fixes #73 - Ignore pauses for Timelapse plugins