Clon1998/mobileraker_companion #75: Wrong translation

Prepete, 2024-01-11:
Hi there!
I wanted to start by praising your work. The app is wonderful and I find it extremely intuitive to use! Great! I noticed a small error in the translation. In the dashboard it says "Filamemt", although it should say "End of printing" or "Completion". I have attached a photo so that the position is clear.
Best regards and keep up the good work![image]

Clon1998, 2024-01-11:
Hey there! This isn't the right place to report app-related issues. Please report them at [mobileraker](https://github.com/Clon1998/mobileraker). However, I wanted to let you know that the issue you were facing has been fixed in the latest app update (version 2.6.9).

Clon1998, 2024-01-11:
Related to https://github.com/Clon1998/mobileraker/issues/300

closed 2024-01-11 by Clon1998