Clon1998/mobileraker_companion #76: Persistent Notification for Progress

alfondehesa, 2024-01-12:
Not sure if this is the repo to add this in, or if this repo includes the app, but would be super cool to have a persistent notification with progress % and ETA instead of getting 20 notifications at each 5% at the most granular. I think Octoapp has an example of this for octoprint setups.

Clon1998, 2024-01-12:
Hey,
I assume you are talking about persistent progress notifications on Android?

It is on my ToDo list, as this was one of the features that I sadly had to remove due to a change in a library. However, luckily the library finally supports these kinds of notifications again. I just have some other stuff to work on right now.

alfondehesa, 2024-01-12:
Makes sense! Thanks for the quick reply.

fuzelet, 2024-02-29:
I can open an issue for this, but based on commits recently and my log file, it looks like this always making things a progress notification unless its a defined, start, stop, error event?

Is this intended behavior now and the % selection in the application doesn't control anything anymore?

https://github.com/Clon1998/mobileraker_companion/commit/e952712ec6f290d840f75dc86f5c66450e49d45f

[image]

Even if i set my App to use 5% i still cant get my old notifications back with images

Clon1998, 2024-02-29:
> I can open an issue for this, but based on commits recently and my log file, it looks like this always making things a progress notification unless its a defined event like start, stop, error?
> 
> Is this intended behavior now and the % selection in the application doesn't control anything anymore?
> 
> ![image](https://private-user-images.githubusercontent.com/70331495/308774225-1eda0ad7-5e64-4187-aaf0-609d3b9c0c85.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxOTkwNTUsIm5iZiI6MTcwOTE5ODc1NSwicGF0aCI6Ii83MDMzMTQ5NS8zMDg3NzQyMjUtMWVkYTBhZDctNWU2NC00MTg3LWFhZjAtNjA5ZDNiOWMwYzg1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI5VDA5MjU1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVhNzZlYzE0ODQ2YjU5ODE1OTg1MmE5MTk3NWVlMzgxZWE3ZGY1MWRiYTIzODUyYzU3N2NkMTUzYTRlY2U0OTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Co-GXbiZLeiz7b6GmAtcKScSqBEn1zXGDQEJmzsJUnA)
> 
> [e952712](https://github.com/Clon1998/mobileraker_companion/commit/e952712ec6f290d840f75dc86f5c66450e49d45f)
> 
> Even if i set my App to use 5% it still seems to use the new notification type? ![Screenshot 2024-02-28 210948](https://private-user-images.githubusercontent.com/70331495/308771620-17489855-3934-4b0c-ba57-d740fb0a560a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxOTkwNTUsIm5iZiI6MTcwOTE5ODc1NSwicGF0aCI6Ii83MDMzMTQ5NS8zMDg3NzE2MjAtMTc0ODk4NTUtMzkzNC00YjBjLWJhNTctZDc0MGZiMGE1NjBhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI5VDA5MjU1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVjMDU1NzNhN2FmZGE0NWJiNzQwY2Y5ZTk4ZmIxOGI3Yjk2YTZhOGM3ODIwNjEyNDM0ODU4MTk3YmVkY2RkMGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OGSQEQiVBJdgMSODSms_dmxCM161dCVzozE_byBmBm8)

Hey, are you using Android or IOS?
On Android, the Progress-Interval option has no effect anymore except if you disable it. While on IOS you should only get notifications in your defined progress interval