Clon1998/mobileraker_companion #78: Docker companion on separate host - no notifications after klipper host shutdown

pwlgrzs, 2024-01-27:
Hello,

I am running mobileraker_companion in docker on a separate host. Klipper host is a separate machine that I turn off during night time. I noticed that I will receive no notifications from companion until I restart the container, however in the logs I can see it's picking up the printer once klipper host is started.

Here's the last log from the container, as expected there's a good number of failed to connect errors ending with the companion reaching out to klipper host once I powered it on. I've done several prints since then and got 0 notifications. Then I restarted container and as visible in the logs I am getting notifications again.

[mobileraker_logs.txt](https://github.com/Clon1998/mobileraker_companion/files/14072430/mobileraker_logs.txt)

Johnnytk36, 2024-04-08:
Just letting you know, I am having the same issue. I have had to restart the docker container for mobileraker on my Synology NAS after restarting my sonic pad, or i don't get the push notifications.

Clon1998, 2024-04-08:
> ###
> [Apr 8, 2024](#issuecomment-2043633623)

Ill try to debug the container in my machine later this week. 
Can you confirm that it is working fine for one print, but as soon as you start another the container requires a restart?

Clon1998, 2024-04-08:
@Johnnytk36 

> > [Apr 8, 2024](#issuecomment-2043633623)
> 
> Ill try to debug the container in my machine later this week. Can you confirm that it is working fine for one print, but as soon as you start another the container requires a restart?

Johnnytk36, 2024-04-08:
@Clon1998 I'll try and do some testing with this for you. I haven't noticed that. Off the top of my head, as soon as i restart the sonic pad, mobileraker doesn't send any push notifications, till i restart the container.

Johnnytk36, 2024-09-05:
I just wanted to let you know this is still an issue.  As far as i can tell it doesn't work as soon as the sonic pad is rebooted.

Clon1998, 2024-09-07:
Sorry for the delay in looking into this. I got caught up with other stuff. I found the root cause of the problem and fixed it. Turns out, the issue wasn't with the docker image but with the websocket client not properly sending the disconnect event to all services. This meant that the data sync service, which receives state updates from the printer, wasn't re-subscribing to status updates.

closed 2024-09-07 by Clon1998

Johnnytk36, 2024-09-10:
@Clon1998 Sorry, i have been printing 24/7 lately. Just had a chance to try this. Your fix seems to have work. I can't recreate the error now.

Thank you!

commit 18064047 (2024-09-07): jrpc: Fix connection state propagation to listeners

Resolved an issue where the connection state was not properly propagating to the listeners. This caused the data_sync client to fail to subscribe to objects and receive status updates, resulting in the companion not issuing new notifications unless it or Klippy was restarted.

Fixes #78