Clon1998/mobileraker_companion #79: No connection to 2nd printer

ColdBeer72, 2024-01-31:
Hi! I've got this in my ~/Mobileraker.conf:                                                        
```
[general]
language: en
timezone: Europe/Madrid
eta_format: %%d/%%m/%%Y, %%H:%%M:%%S
include_snapshot: True

[printer FBG6]
moonraker_url: ws://127.0.0.1:7126/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
snapshot_rotation: 0

[printer CY300]
moonraker_url: ws://127.0.0.1:7125/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam2/?action=snapshot
snapshot_rotation: 0
```
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/14116867/mobileraker.log)
Attached my /tmp/mobileraker.log

But **no notification from "FBG6" printer (the one at 7126 port on moonraker)**  at all, and Mobileraker Android App shows this:
[image]

I installed MobileRaker through KIAUH when both printers were installed yet. After that, I tried a new reinstallation through:
`~/mobileraker_companion//scripts/install-mobileraker-companion.sh`
But everything is still the same.

Could I do something to make both printers work through Mobileraker_companion?

Clon1998, 2024-01-31:
Hey,
the logs state that the companion found a config file at `/home/coldbeer/Mobileraker.conf`. In that config file it was able tor read two printers. However, both printers are defined with the same IP:
```
2024-01-31 20:12:23,817 mobileraker.jrpc INFO - Trying to connect to: ws://127.0.0.1:7125/websocket api key false##########################
2024-01-31 20:12:23,818 mobileraker.jrpc INFO - Trying to connect to: ws://127.0.0.1:7125/websocket api key false##########################
```

ColdBeer72, 2024-01-31:
> Hey, the logs state that the companion found a config file at `/home/coldbeer/Mobileraker.conf`. In that config file it was able tor read two printers. However, both printers are defined with the [...]

Of course there are two printers on the same IP, but I didn't understand on your GitHub they should be different IPs... That share the same IP but uses different parts, so connections are totally different.

Clon1998, 2024-01-31:
> Of course there are two printers on the same IP, but I didn't understand on your GitHub they should be different IPs... That share the same IP but uses different parts, so connections are totally [...]

I am sorry, they are using the same port (Both use `7125`).

ColdBeer72, 2024-01-31:
> I am sorry, they are using the same port (Both use `7125`)

R U sure???
[image]

Clon1998, 2024-02-01:
> R U sure???
>
> [image]

It's what the companion wrote to the logs.
Let's try to change the non working URi to something completely different just to verify if the companion is picking up the correct config file.

Edit the FBG6 url to something like: 1.2.3.4:6060/test

Afterwards, restart the companion and have a look at the latest logs to verify if the connection attempt logs change. (Note to myself: Add the printer name to the logger of the JRPC client)

ColdBeer72, 2024-02-01:
> > > > > Hey, the logs state that the companion found a config file at `/home/coldbeer/Mobileraker.conf`. In that config file it was able tor read two printers. However, both printers are defined with the same IP:
> > 
> > 
> > > > >
> > 
> > 
> > > > > 2024-01-31 20:12:23,817 mobileraker.jrpc INFO - Trying to connect to: ws://127.0.0.1:7125/websocket api key false##########################
> > 
> > 
> > > > > 2024-01-31 20:12:23,818 mobileraker.jrpc INFO - Trying to connect to: ws://127.0.0.1:7125/websocket api key false##########################
> > 
> > 
> > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > Of course there are two printers on the same IP, but I didn't understand on your GitHub they should be different IPs... That share the same IP but uses different parts, so connections are totally different.
> > 
> > 
> > > 
> > 
> > 
> > > I am sorry, they are using the same port (Both use `7125`)
> > 
> > 
> > R U sure???
> > ![Screenshot_20240201_004947](https://private-user-images.githubusercontent.com/78320296/301342781-5f41d01a-fcc1-4e21-b0e9-3caebfe81c23.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY3OTk1NDQsIm5iZiI6MTcwNjc5OTI0NCwicGF0aCI6Ii83ODMyMDI5Ni8zMDEzNDI3ODEtNWY0MWQwMWEtZmNjMS00ZTIxLWIwZTktM2NhZWJmZTgxYzIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjAxVDE0NTQwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRmNzE5MTA4MDlkZTk0MjJhNjU3ZjU2ZDZjNGQzMTVmYTgwN2RjNjUzOTRmMmY1MTUxNWMyMGVjMzYyOTUxMmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QAGWYfxhN38SqtzFoCLhsoQ_4MqtlQC1ludnRoPZSNU)
> 
> It's what the companion wrote to the logs. Let's try to change the non working URi to something completely different just to verify if the companion is picking up the correct config file.
> 
> Edit the FBG6 url to something like: 1.2.3.4:6060/test
> 
> Afterwards, restart the companion and have a look at the latest logs to verify if the connection attempt logs change. (Note to myself: Add the printer name to the logger of the JRPC client)

Ok, here it's the new log after making exactly the change you said:
```
[printer FBG6]
moonraker_url: ws://1.2.3.4:6080/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
snapshot_rotation: 0

[printer CY300]
moonraker_url: ws://127.0.0.1:7125/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam2/?action=snapshot
snapshot_rotation: 0
```

[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/14127094/mobileraker.log)

ColdBeer72, 2024-02-01:
Ok, I applied exactly the changes you said:
```
[printer FBG6]
moonraker_url: ws://1.2.3.4:6080/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
snapshot_rotation: 0

[printer CY300]
moonraker_url: ws://127.0.0.1:7125/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam2/?action=snapshot
snapshot_rotation: 0
```
 And this is the new log:
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/14127132/mobileraker.log)

Clon1998, 2024-02-01:
Hey,
could you pull the latest version of the companion?
Afterwards, please restart the companion and provide me the logs once again.
Via #80 I improved the logging.

Also please provide me a `ls -l` dump of the content in `/home/coldbeer/`:
```bash
cd /home/coldbeer/
ls -l
```

ColdBeer72, 2024-02-01:
Ok, Here I am again with latest v0.4.0-42-g54581aa8 mobileraker version.

My home folder is:
```
coldbeer@3dprint:~$ ls -l
total 144
drwxrwxr-x 13 coldbeer coldbeer  4096 Feb  1 18:03 CY300_data
drwxrwxr-x 13 coldbeer coldbeer  4096 Feb  1 18:04 FBG6_data
-rw-rw-r--  1 coldbeer coldbeer 14682 Jan 26 21:23 IronMan_cuad.ansi.txt
-rwxr-xr-x  1 coldbeer coldbeer 31286 Jan 26 21:18 IronMan_cuad.jpg
drwxrwxr-x  9 coldbeer coldbeer  4096 Jan 30 08:48 KlipperScreen
-rw-r--r--  1 coldbeer coldbeer  1227 Jan 30 00:45 KlipperScreen.conf
-rw-rw-r--  1 coldbeer coldbeer  1903 Feb  1 15:55 Mobileraker.conf
-rwxrwxr-x  1 coldbeer coldbeer    35 Jan 26 20:33 beep.sh
drwxrwxr-x 10 coldbeer coldbeer  4096 Jan 26 17:09 crowsnest
drwxrwxr-x  7 coldbeer coldbeer  4096 Jan 26 13:54 kiauh
drwxrwxr-x  4 coldbeer coldbeer  4096 Jan 26 16:50 kiauh-backups
drwxrwxr-x 11 coldbeer coldbeer  4096 Jan 26 14:11 klipper
drwxrwxr-x  5 coldbeer coldbeer  4096 Jan 26 14:17 klippy-env
drwxrwxr-x  5 coldbeer coldbeer  4096 Jan 26 16:46 mainsail
drwxrwxr-x  3 coldbeer coldbeer  4096 Jan 26 16:50 mainsail-config
drwxrwxr-x  4 coldbeer coldbeer  4096 Jan 27 00:15 mobileraker-env
drwxrwxr-x  9 coldbeer coldbeer  4096 Feb  1 18:04 mobileraker_companion
drwxrwxr-x  8 coldbeer coldbeer  4096 Feb  1 18:03 moonraker
drwxrwxr-x  4 coldbeer coldbeer  4096 Jan 26 16:40 moonraker-env
drwxrwxr-x  8 coldbeer coldbeer  4096 Jan 26 17:23 moonraker-telegram-bot
drwxrwxr-x  4 coldbeer coldbeer  4096 Jan 26 17:29 moonraker-telegram-bot-env
drwxr-xr-x 12 coldbeer coldbeer  4096 Jan 31 20:42 moonraker-telegram-bot-timelapse
drwxrwxr-x  8 coldbeer coldbeer  4096 Jan 26 23:48 moonraker-timelapse
drwxr-x---  2 coldbeer coldbeer  4096 Jan 30 20:58 ovpns
drwxrwxr-x  9 coldbeer coldbeer  4096 Jan 30 00:17 pivpn
drwxrwxr-x  4 coldbeer coldbeer  4096 Jan 26 19:54 restore
-rw-r--r--  1 coldbeer coldbeer     0 Jan 26 23:23 variables.cfg
```
Anyway:
```
coldbeer@3dprint:~$ locate mobileraker.conf
coldbeer@3dprint:~$ locate Mobileraker.conf
/home/coldbeer/Mobileraker.conf
```
After that, the log with the bad config is:
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/14128576/mobileraker.log)

Anyway, I restored the good config and the -still bad- result log is:
[mobileraker.log](https://github.com/Clon1998/mobileraker_companion/files/14128608/mobileraker.log)

Clon1998, 2024-02-01:
You are using the wrong config keyword for the `URI`.
Currently, it is set to `moonraker_url` instead of `moonraker_uri`. Therfore, mobileraker always defaults to `ws://127.0.0.1:7125/websocket`.
Try this config:
```python
[general]
language: en
timezone: Europe/Madrid
eta_format: %%d/%%m/%%Y, %%H:%%M:%%S
include_snapshot: True

[printer FBG6]
moonraker_uri: ws://127.0.0.1:7126/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
snapshot_rotation: 0

[printer CY300]
moonraker_uri: ws://127.0.0.1:7125/websocket
moonraker_api_key: false
snapshot_uri: http://127.0.0.1/webcam2/?action=snapshot
snapshot_rotation: 0
```

Clon1998, 2024-02-01:
Woops, closed by accident.

ColdBeer72, 2024-02-01:
Personally, I think mobileraker is not taken the port in the connection procedure. It always shows as a default (7125) connection.

Clon1998, 2024-02-01:
> Personally, I think mobileraker is not taken the port in the connection procedure. It always shows as a default (7125) connection.

 It did not take the changed ip either. The reason for that is that you are not using `URI` as a suffix for `moonraker_`. Instead your config states `URL`. Since `moonraker_uri` is missing entierly the companion defaults to `ws://127.0.0.1:7125/websocket`.

ColdBeer72, 2024-02-01:
> You are using the wrong config keyword for the `URI`. Currently, it is set to `moonraker_url` instead of `moonraker_uri`. Therfore, mobileraker always defaults to `ws://127.0.0.1:7125/websocket`. Try this config:
> 
> ```ini
> [general]
> language: en
> timezone: Europe/Madrid
> eta_format: %%d/%%m/%%Y, %%H:%%M:%%S
> include_snapshot: True
> 
> [printer FBG6]
> moonraker_uri: ws://127.0.0.1:7126/websocket
> moonraker_api_key: false
> snapshot_uri: http://127.0.0.1/webcam/?action=snapshot
> snapshot_rotation: 0
> 
> [printer CY300]
> moonraker_uri: ws://127.0.0.1:7125/websocket
> moonraker_api_key: false
> snapshot_uri: http://127.0.0.1/webcam2/?action=snapshot
> snapshot_rotation: 0
> ```

Yeah!!! That was the error!!! Thank you very much for your patience with me... Now I go to cry in my loneliness...

closed 2024-02-01 by ColdBeer72

Clon1998, 2024-02-01:
> Yeah!!! That was the error!!! Thank you very much for your patience with me... Now I go to cry in my loneliness...

No worries