Clon1998/mobileraker_companion #81: Disable Notifications for Timelapse isn't working

Snuffy2, 2024-02-05:
Similar to #73, I recently setup moonraker-timelapse. I have it set to park the head at the end of each layer before taking the timelapse photo. With this enabled, I am now getting a Pause, immediately followed by a Print Notification for every layer. 

Not sure what information to provide, but https://github.com/Clon1998/mobileraker_companion/commit/322b4d8a90d2259746697aec17ea794af16c7bd4 doesn't seem to be working correctly at least for me.

[image]

Clon1998, 2024-02-06:
I finally had some time to install the timelapse plugin properly and debug this myself.

I just fixed it.

Please let me know if you still encounter this issue even after updating.
cc @Johnnytk36

Johnnytk36, 2024-02-06:
@Clon1998 Looks like it is from the test print I just did after updating. I also did a manual pause and resume and those notifications came though. 

Thanks!

Johnnytk36, 2024-02-06:
@Clon1998 I spoke too soon.  It made it to layer 9 with no issues, and then after I did a manual Pause and resume, I got a couple sets of pauses and resumes notifications from the timelapse function. I don't seem to get every layer now, jsut some random ones. I started the print fro work, so I cant stop and test again right now.

Clon1998, 2024-02-06:
> @Clon1998 I spoke too soon.  It made it to layer 9 with no issues, and then after I did a manual Pause and resume, I got the next couple of sets of pauses and resumes notifications from the timelapse function. I don't seem to get every layer now, just some random ones. I started the print from work, so I cant stop and test again right now. 

Thanks,
I'll look into that.

Johnnytk36, 2024-02-07:
I started a new print when I got home and with no intervention from me, i got random alerts when he timelapse paused/started. Not every time like before though.

commit cb6e9c37 (2024-02-20): Fixes #81 - Update timelapse pause handling in DataSyncService