Clon1998/mobileraker_companion #82: Disable Notifications for Timelapse Improved but still isn't fully working

Snuffy2, 2024-02-21:
Building on #73 and #81. As @Johnnytk36 had noted in #81, the notifications are now less when pausing and restarting to take a timelapse photo (not every layer), but are still happening, just intermittently. Often print will go 25-50% without any erroneous pause/start alerts then it will start showing these sometimes every few layers, sometimes for every layer until it completes.

If there's some type of logging you'd like to see or something to test, please let me know.

[image]
[image]

[image]
[image]
[image]
[image]
[image]

Clon1998, 2024-02-22:
Hey,
I have ran the current version of the companion + timelapse on two of my machines without any problem.

However, I am not using the parking mode.
Can you confirm to me that you are using the parking mode?

Feel free to upload the logs. They are either located in `/tmp/mobileraker.log` or in the same location as all other logs.

Johnnytk36, 2024-02-22:
I am using the parking mode.

[image]

Clon1998, 2024-02-22:
> I am using the parking mode.
> 
> <img alt="Screenshot 2024-02-22 at 12 13 14 PM" width="969" src="https://private-user-images.githubusercontent.com/133730955/307103409-892bcfc0-a9f5-45a4-a51a-75d56f89947b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2MzIzMTAsIm5iZiI6MTcwODYzMjAxMCwicGF0aCI6Ii8xMzM3MzA5NTUvMzA3MTAzNDA5LTg5MmJjZmMwLWE5ZjUtNDVhNC1hNTFhLTc1ZDU2Zjg5OTQ3Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMlQyMDAwMTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NmJlZjMwNjgyZTQyNmI1YWI3N2YwOTA2MTllMWE5ZjRiOWQ2ZmE3OGJmYjNhYzVkMjI5NjIzMTI0ZTdjMGMxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2hPDOKbnWCOHsW5cQEPBi3mnXO0B_eNjD1tPib5-zp4">

I will enable it on one of my printers to debug it!

Snuffy2, 2024-02-22:
Same here. The parking between layers before taking the timelapse is what is triggering those notices.

Snuffy2, 2024-03-07:
Logfile of recent print is attached. Timelapse with parking between layers is on. There were only 34 layers and it sent me Pause/Start alerts for 2 layers sometime between 25-50% complete (between layers 8-18).

[mobileraker_with_timelapse.log](https://github.com/Clon1998/mobileraker_companion/files/14518233/mobileraker_with_timelapse.log)

[image]

Snuffy2, 2024-07-02:
Any updates on getting this fixed?

Clon1998, 2024-07-02:
> Any updates on getting this fixed?

I lost track of this issue while working on 2.7.0.
I will take a look in the next days to see if I am able to reproduce this issue and hopefully fix it.

In the meantime, can you provide me the version of the companion you are using @Snuffy2 ?

Snuffy2, 2024-07-02:
No worries, totally understand and thank you very much for a great tool. I'm running:
Mobileraker app: 2.7.0-501
Mobileraker Companion: v0.4.0-58-g828e9a8c

Clon1998, 2024-07-03:
> No worries, totally understand and thank you very much for a great tool. I'm running: Mobileraker app: 2.7.0-501 Mobileraker Companion: v0.4.0-58-g828e9a8c

Thanks. The companion is a bit outdated but nothing major that would affect this issue.
Also, just was able to reproduce that behavior locally. Just need to identify the root cause for this issue!

closed 2024-07-03 by Clon1998

Snuffy2, 2024-07-04:
Working great so far after printing a couple of things. Thanks!!!

commit 074394a8 (2024-07-03): Fix #82 - Reset timelapse pause only after printer is back printing