Clon1998/mobileraker_companion #83: Moonraker warnings found

Clon1998, 2024-02-24:
If you are encountering this issue:

[image]

You will need to update the `install_script` property of the  `[update_manager mobileraker]` section:
(Red old value, Green new value. Ignore the +,- chars)
```diff
[update_manager mobileraker]
...# other properties
-install_script: scripts/install-mobileraker-companion.sh
+install_script: scripts/install.sh

```

Mechanikerlegion511, 2024-03-05:
After the update I had the same problem today.
I did as described in the post above, the problem disappeared

atong9033, 2024-04-23:
why not write it. what do I need to edit in moonraker.conf?

Clon1998, 2024-04-23:
> why not write it. what do I need to edit in moonraker.conf? do you comment on the lines? then why is it so difficult to specify the full path?

I don't understand what you're trying to ask?
The first post pretty clearly describes what you need to do to resolve the issue.

The origin of the issue is the fact that I removed the old install script in favor of a new one that uses the installer rather than a raw shell script to install the companion.

titanhusker, 2024-05-09:
> why not write it. what do I need to edit in moonraker.conf? do you comment on the lines? then why is it so difficult to specify the full path?

Found this post when I ran into the same issue. I thought I would shed a little light on it if you haven't figured it out yet. It is straight forward....Log into your printer>Under your configuration tab>modify moonraker.conf (the line item might not be the exact number but it will be close) change the line to the new one>save and restart and the error should be corrected. Hope that helps.

Clon1998, 2024-05-19:
> I just installed opennept4une today and I'm having the same problem.
> 
> In the configuration it looks like this : 
> 
> [update_manager mobileraker]
> 
> type: git_repo
> 
> path: ~/mobileraker_companion
> 
> origin: https://github.com/Clon1998/mobileraker_companion.git
> 
> primary_branch:main
> 
> managed_services: mobileraker
> 
> env: ~/mobileraker-env/bin/python
> 
> requirements: scripts/mobileraker-requirements.txt
> 
> install_script: scripts/install.sh
> 
> 
> 
>  Even if I rewrite it to look like your screen shot, the problem still persists.Please, what am I doing wrong?
> 
> I'm completely new to this.
> 
> Thank you

Did you restart moonraker after saving the file?
If you did please provide me a screenshot of the error.

MichalOrel, 2024-05-19:
Yes, I saved and rebooted. 
[image]
[image]

MichalOrel, 2024-05-20:
> Found this post when I ran into the same issue. I thought I would shed a little light on it if you haven't figured it out yet. It is straight forward....Log into your printer>Under your [...]

I dont understand:  (the line item might not be the exact number but it will be close) change the line to the new one> 
Please, could you elaborate a little bit?

Clon1998, 2024-05-20:
> I dont understand:  (the line item might not be the exact number but it will be close) change the line to the new one>
>
> Please, could you elaborate a little bit?

You'll need to locate the path of the mobileraker companion within the file system of the printer.
Afterwards, adjust the path in the install_script param to that local path.

MichalOrel, 2024-05-21:
> You'll need to locate the path of the mobileraker companion within the file system of the printer. Afterwards, adjust the path in the install_script param to that local path.

Thanks, but I'm completely lost.I have no idea where to look ...

titanhusker, 2024-05-23:
What do you mean? Locate the old line and change it to the new one, save and restart, you should be good from there. I was simply stating that the line # might differ depending on yours, don’t go by the number, go by the content.
Hope that helps.

MichalOrel, 2024-05-24:
> What do you mean? Locate the old line and change it to the new one, save and restart, you should be good from there. I was simply stating that the line # might differ depending on yours, don’t go by the number, go by the content. Hope that helps.

I am a complete amateur at this.I had no idea how to do it, what line and where to find it.
It's hard to solve such things without detailed guidance.
Anyway, I solved it via Kiauh update.

titanhusker, 2024-05-24:
Did you install Klipper yourself? When you log into your printer through the GUI, there should be tabs on the left side running down the left side. Go to the tab that has your config files, you will need to alter the moonraker config file. When you open the file look through the lines and locate the one that resembles what needs to be changed and change it (Onky change what needs to be changed). Then save the file and restart Klipper / Moonraker. The error should be gone if you did correctly. Hope that is a little more detailed for you. Good Luck

Prsmith3, 2024-06-22:
Hello, I installed openNept4une about three weeks ago. I had the exact same error messages.  I found this thread and went to the moonraker config to find it already set to the line change mentioned above.

[update_manager mobileraker]
type: git_repo
path: ~/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch:main
managed_services: mobileraker
env: ~/mobileraker-env/bin/python
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install.sh

There is one difference, and that is the 6th line. Mine doesn't say "virtualenv"
Changing that to match OP's doesn't change anything.  Yes, it was saved and rebooted multiple times.

[image]

So far this is the only place I can find that mentions this issue exactly as I'm seeing it , but the problem persists. 
I haven't done much printing since the install. just a few calibration prints. 'cause I just don't have enough Benchys 
yet!
That said things seem to be working.

Choccy-vr, 2024-07-09:
To those who installed Open Neptune and having trouble with this I found that I had this line `install_script: scripts/install.sh` instead of this `install_script: scripts/install-mobileraker-companion.sh`. change `install_script: scripts/install.sh` to `install_script: scripts/install-mobileraker-companion.sh` In `moonraker.conf`. Please note that this may change but at the time of posting this that is the answer for Open Neptune.

Prsmith3, 2024-07-10:
> To those who installed Open Neptune and having trouble with this I found that I had this line `install_script: scripts/install.sh` instead of this `install_script: [...]

Thank you! That worked perfectly.

geekredux, 2024-09-05:
> To those who installed Open Neptune and having trouble with this I found that I had this line `install_script: scripts/install.sh` instead of this `install_script: [...]

I just installed OpenNept4une and ran into the same problem, with the line already saying `install_script: scripts/install.sh`. 
Your change has at least made the warnings go away, and everything seems to be working.

May I ask how you figured this out?

johndsummers, 2024-12-22:
> To those who installed Open Neptune and having trouble with this I found that I had this line `install_script: scripts/install.sh` instead of this `install_script: [...]

As of 12/22/24, I changed it to the line above after doing a fresh install on 12/21/24. The error went away.

Nickfuentes92, 2025-01-13:
> If you are encountering this issue:
> 
> <img alt="image" width="1229" src="https://private-user-images.githubusercontent.com/10357775/307518334-508dfbf9-1477-4d06-aae9-8a7b61307bd8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3Mzg3NjMsIm5iZiI6MTczNjczODQ2MywicGF0aCI6Ii8xMDM1Nzc3NS8zMDc1MTgzMzQtNTA4ZGZiZjktMTQ3Ny00ZDA2LWFhZTktOGE3YjYxMzA3YmQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDAzMjEwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1ZmE2ODNkYzIzZTdmYTFlODAyNzVlZDg2ZWVhM2E4N2ZhNjI3NzhhNTBhYzc4MTI5ZTFkNDc2OTBiMDM3YWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.6vhKwkF2mnsyRRGS4oMpZZ-JHKn9dxUwpq-1K4n1nQk">
> You will need to update the `install_script` property of the `[update_manager mobileraker]` section: (Red old value, Green new value. Ignore the +,- chars)
> 
> ```diff
> [update_manager mobileraker]
> ...# other properties
> -install_script: scripts/install-mobileraker-companion.sh
> +install_script: scripts/install.sh
> ```
> 
> Result: <img alt="image" width="687" src="https://private-user-images.githubusercontent.com/10357775/307518658-44284b15-a036-4c3e-ac5a-6303184a70c2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3Mzg3NjMsIm5iZiI6MTczNjczODQ2MywicGF0aCI6Ii8xMDM1Nzc3NS8zMDc1MTg2NTgtNDQyODRiMTUtYTAzNi00YzNlLWFjNWEtNjMwMzE4NGE3MGMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDAzMjEwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiODA4ZDFiY2FlNDE2OTY0NTQxODUwOGViNjlkNDkyOTVjOGYzNDIxMDE5Mzk3MzRlNGM5YmY2YjVjOTRhY2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CFntp_LN-_uGbu1oRk0BsQ-mF0h5XcJCMC9_TYTFIpk">

where exactly do you go? new to all of this. What and where do you change it

Clon1998, 2025-01-16:
> > If you are encountering this issue:
> > <img alt="image" width="1229" src="https://private-user-images.githubusercontent.com/10357775/307518334-508dfbf9-1477-4d06-aae9-8a7b61307bd8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3Mzg3NjMsIm5iZiI6MTczNjczODQ2MywicGF0aCI6Ii8xMDM1Nzc3NS8zMDc1MTgzMzQtNTA4ZGZiZjktMTQ3Ny00ZDA2LWFhZTktOGE3YjYxMzA3YmQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDAzMjEwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1ZmE2ODNkYzIzZTdmYTFlODAyNzVlZDg2ZWVhM2E4N2ZhNjI3NzhhNTBhYzc4MTI5ZTFkNDc2OTBiMDM3YWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.6vhKwkF2mnsyRRGS4oMpZZ-JHKn9dxUwpq-1K4n1nQk">
> > You will need to update the `install_script` property of the `[update_manager mobileraker]` section: (Red old value, Green new value. Ignore the +,- chars)
> > [update_manager mobileraker]
> > ...# other properties
> > -install_script: scripts/install-mobileraker-companion.sh
> > +install_script: scripts/install.sh
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Result: <img alt="image" width="687" src="https://private-user-images.githubusercontent.com/10357775/307518658-44284b15-a036-4c3e-ac5a-6303184a70c2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3Mzg3NjMsIm5iZiI6MTczNjczODQ2MywicGF0aCI6Ii8xMDM1Nzc3NS8zMDc1MTg2NTgtNDQyODRiMTUtYTAzNi00YzNlLWFjNWEtNjMwMzE4NGE3MGMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDAzMjEwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiODA4ZDFiY2FlNDE2OTY0NTQxODUwOGViNjlkNDkyOTVjOGYzNDIxMDE5Mzk3MzRlNGM5YmY2YjVjOTRhY2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CFntp_LN-_uGbu1oRk0BsQ-mF0h5XcJCMC9_TYTFIpk">
> 
> where exactly do you go? new to all of this. What and where do you change it

By default the warning message tells you the path of the conf where you need to adjust it.

titanhusker, 2025-02-05:
[image]
[image]

I am stumped again. I updated Klipper and all of its components through the configuration page as everything was very out of date. I ended up with the error shown above. I know I corrected the issue before because I chimed in then, but something else has changed since that particular version. I have also reviewed all of the other notes. When I find a suitable solution I will repost for other to see. In the meantime, if @Clon1998 or anyone else knows how to clear this new error, your help is very much appreciated. 

I will add, I did not have a webcam before, nor do have one now for the printer if this camera related.

Clon1998, 2025-02-05:
> [image] [image]
>
> I am stumped again. I updated Klipper and all of its components through the configuration page as everything was very out of date. I ended up with the error shown above. I know I [...]

That is not related to mobileraker but crownnest. You will need to adjust the path for the crownnest script

FrankHermanns, 2025-02-26:
I had the same problem. After installing the printer, the new path (install_script: scripts/install.sh) was already entered in the moonranker.config. However, the old script is in the image file for the printer. Therefore I first had to replace the entry "install_script: scripts/install.sh" with "install_script: scripts/install-mobileraker-companion.sh". Restart the printer and the error is (for now) gone. Then update the system. The error is back. Then everything back to new in the moonranker.config. So replace the line "install_script: scripts/install-mobileraker-companion.sh" with "install_script: scripts/install.sh".