Clon1998/mobileraker_companion #84: Moonraker warnings found.

Festus141, 2024-02-24:
Moonraker warnings found.
[update_manager mobileraker]: Moonraker is not permitted to restart service 'mobileraker'. To enable management of this service add mobileraker to the bottom of the file /home/pi/printer_data/moonraker.asvc. To disable management for this service set 'is_system_service: False' in the configuration for this section.

Clon1998, 2024-02-24:
Hey,
you might want to re run the installer to make sure the service is properly added to the mentioned file.

Ru83nG, 2024-03-06:
Same thing here. clean install and a bunch of errors.   What exactly should I do?  Keep uninstalling and installing hoping it will catch on?   
[update_manager mobileraker]: Moonraker is not permitted to restart service 'mobileraker'. To enable management of this service add mobileraker to the bottom of the file /home/Pi4B/printer_data/moonraker.asvc. To disable management for this service set 'is_system_service: False' in the configuration for this section.

Besides, the file is already in where this states it should be.

Thanks

conlank, 2024-03-12:
I am using the following and not getting any of these moonraker errors.
```
[update_manager mobileraker]
type: git_repo
channel: dev
path: ~/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch: main
is_system_service: True
managed_services: mobileraker
virtualenv: ~/mobileraker-env
requirements: scripts/mobileraker-requirements.txt
```

thotland, 2024-03-15:
I also have this error. It's not related to the mobileraker companion, but to the fact that moonraker can't control the mobileraker service when it needs to. I've tried adding a line to moonraker.asvc but I don't know the syntax. I believe it's so moonraker can update mobileraker along with the other services.

Clon1998, 2024-03-15:
> Same thing here. clean install and a bunch of errors. What exactly should I do? Keep uninstalling and installing hoping it will catch on? [update_manager mobileraker]: Moonraker is not permitted to [...]

If the line is already in place I don't really have an idea what else could be causing this issue tbh.
By default adding `mobileraker` to the file in a new line should resolve the issue. Might be worth asking in the moonraker github about that issue.

Clon1998, 2024-03-15:
> I also have this error. It's not related to the mobileraker companion, but to the fact that moonraker can't control the mobileraker service when it needs to. I've tried adding a line to [...]

The asvc file is responsible for telling moonraker which services it is allowed to control.
Each lines contains a service. In my case, the file looks like:
```
klipper_mcu
webcamd
MoonCord
KlipperScreen
moonraker-telegram-bot
moonraker-obico
sonar
crowsnest
octoeverywhere
mobileraker
```

Ru83nG, 2024-03-16:
Ok, thank you soo much.
My fault, I did not read it correctly the first few times. 
It means to go into the asvc file and ADD the LINE with "mobileraker".
I did it and it works just fine now.
Sorry for the late reply.
No more errors.  
Thank you for the hard work!!

texaslonghorn75, 2024-03-18:
> Ok, thank you soo much. My fault, I did not read it correctly the first few times. It means to go into the asvc file and ADD the LINE with "mobileraker". I did it and it works just fine now. Sorry for the late reply. No more errors. Thank you for the hard work!!

Where do you find the asvc file?

Clon1998, 2024-03-18:
> > Ok, thank you soo much. My fault, I did not read it correctly the first few times. It means to go into the asvc file and ADD the LINE with "mobileraker". I did it and it works just fine now. Sorry for the late reply. No more errors. Thank you for the hard work!!
> 
> Where do you find the asvc file?

By default the installer of mobileraker should modify the file. In addition, the error message shown via Fluidd/Mainsail also mentions the path to the file (See the initial post of that thread)

Ru83nG, 2024-03-18:
Hello,

I used WinSCP to get to the file..

Navigate to:   /home/[your host name here]/printer_data/mmonrker.asvc

I edited (right-click to show other options) the file from within WinSCP to add the line at the bottom "mobileraker".
Save the file and reboot.
Hope this helps a bit.

Clon1998, 2024-03-23:
Hey guys,
I finally was able to reproduce that issue and resolve it.
If you are still encountering that issue. Please re run the installer as it will add the `mobileraker` to the `asvc` file.

closed 2024-03-23 by Clon1998

GrzesMis, 2024-08-14:
> I am using the following and not getting any of these moonraker errors.
>
> ```
> [update_manager mobileraker]
> type: git_repo
> channel: dev
> path: ~/mobileraker_companion
> origin: [...]

Did this and now moonraker cant boot... also klipper and fluid... Now i need to install Klipper all over again...

commit 2e3a1af1 (2024-03-23): Fix #84 - Ensured mobileraker is added as new line