Clon1998/mobileraker_companion #85: Install script does not work in NZDT

Deadleg, 2024-02-25:
Hi, down here in NZ we are in day light savings with UTC offset +13.

I've configured the system time zone via `timedatectl` to `Pacific/Auckland` which resolves to `GMT+13` in the install script, leading to the following error:

```
Installer failed - 'Etc/GMT+13'

Stack Trace:
Traceback (most recent call last):
  File "/home/biqu/mobileraker_companion/installer/Installer.py", line 20, in Run
    self._run()
  File "/home/biqu/mobileraker_companion/installer/Installer.py", line 126, in _run
    config.run(context)
  File "/home/biqu/mobileraker_companion/installer/Config.py", line 48, in run
    self._setup_mobileraker_conf(context)
  File "/home/biqu/mobileraker_companion/installer/Config.py", line 76, in _setup_mobileraker_conf
    config.set("general", "timezone", get_local_timezone())
  File "/home/biqu/mobileraker_companion/mobileraker/util/configs.py", line 32, in get_local_timezone
    local_tz = pytz.timezone(f'Etc/GMT{local_tz_offset//3600:+d}')
  File "/home/biqu/mobileraker-env/lib/python3.9/site-packages/pytz/__init__.py", line 188, in timezone
    raise UnknownTimeZoneError(zone)
pytz.exceptions.UnknownTimeZoneError: 'Etc/GMT+13'
```

I've manually hard coded the time zone to `Pacific/Auckland` as a work around.

I haven't dug into the code to see how the time zone is used but I wonder if the change daylight savings will have other side effects due to use of a fixed GMT + time zone offset.

Clon1998, 2024-02-25:
Hey,
thanks for letting me know.
I switched to yet another method to determine the timezone.
Can you do me a favor and remove the` mobileraker.conf` file and re-run the installer with the updated method?

It wont convert the tzinfo to a IANA timezone but it should run now without any errors.

Deadleg, 2024-03-09:
Just pulled the latest and removed `mobileraker.conf` before rerunning the installer which has added to following:

```toml
[general]
language: en
# one of the supported languages defined in i18n.py#languages (de,en,...)
# Default: en
timezone: NZDT
# The system's timezone e.g. Europe/Berlin for Berlin time or US/Central.
# For more values see https://gist.github.com/heyalexej/8bf688fd67d7199be4a1682b3eec7568
# Default: Tries to use system timezone
# Optional
eta_format: %%d.%%m.%%Y, %%H:%%M:%%S
# Format used for eta and adaptive_eta placeholder variables
# For available options see https://strftime.org/
# Note that you will have to escape the % char by using a 2nd one e.g.: %d/%m/%Y -> %%d/%%m/%%Y
# Default: %%d.%%m.%%Y, %%H:%%M:%%S
# Optional
include_snapshot: True
# !! SUPPORTER ONLY - This feature requires beeing a supporter of Mobileraker as of now!
# Include a snapshot of the webcam in any print status/progress update notifications
# Default: True
# Optional
```

NZDT is not in that gist so I've double checked what `dateutil.tz` does and it seems to resolve to _something_, but I'm not sure how that will behave once daylight savings changes.

```
>>> tz.gettz('NZDT')
tzlocal()
>>> tz.gettz('Pacific/Auckland')
tzfile('/usr/share/zoneinfo/Pacific/Auckland')
```

Clon1998, 2024-03-23:
> Just pulled the latest and removed `mobileraker.conf` before rerunning the installer which has added to following:
>
> ```toml
> [general]
> language: en
> # one of the supported languages defined in [...]

Thanks for letting me know.
I think it will have problems with dls. I need to investigate that further. The current code works on all platforms and I don't want to break it yet again..