Clon1998/mobileraker_companion #87: Unable to solve Mobileraker errors after Klipper update

fonsuak, 2024-03-31:
Hello, I am facing similar Moonraker issues after updateing Klipper as described in the last couple of reports. But I am not being able to solve them by following #83 

The errors I get are the following: 
[image]

My `[update_manager mobileraker]` section looks like this:
[image]

I have tried changing just the `install_script` line as recommended, and also changing all lines under `[update_manager mobileraker]` to match the picture shared in #83

Before all this I also tried:

- Get everything updated via Fluidd interface (Klipper, Fluidd,  Moonraker, Mobileraker, etc.)
- Updated the RPi mcu by:
  - ssh into pi
  - `cd ~/klipper/`
  - `make menuconfig`
  - select `linux process` under `microcontroller archetecture`
  - save and exit
  - `make clean`
  - `sudo service klipper stop`
  - `make flash`
  - ` sudo service klipper start`
  
- Uninstall Mobileraker via ssh with KIAUH
- Reinstall both Moonraker and Mobileraker via ssh with KIAUH

I have run out of ideas of things to try. I would really appreciate someone giving me a hand on this.
If there is any further information/documents that I can share to investigate this please let me know.

Thank you very much

Clon1998, 2024-03-31:
What version of moonraker are you using?
Did you try to verify that the path `/home/pi/mobileraker-env` exists in the file system via the cli?

fonsuak, 2024-03-31:
Hello, I just finally fixed it.
Looks like after tinkering so much with this last week I left mobileraker uninstalled, reason why #83 fix was not working.
After installing mobileraker and changing `install_script: scripts/install.sh` the warnings are gone.
Issue is fixed. Thank you for the quick response and sorry for making you lose time.

closed 2024-03-31 by fonsuak

Clon1998, 2024-03-31:
> Hello, I just finally fixed it. Looks like after tinkering so much with this last week I left mobileraker uninstalled, reason why #83 fix was not working. After installing mobileraker and changing [...]

No worries!