Clon1998/mobileraker_companion #89: 2 notification for almost same

cyberjak2k, 2024-04-28:
Is it supposed to have 2 of basically the same?
The persistent notification is useful however 
1 shows a progress bar , when tapped shows current layer of total

The other shows just the layer progress with not progress bar
And while idle you have 3 notice as shows idle state? 
Anyway to get this down to 1 notice that changes upon change of notice?

[image]
[image]
[image]

Clon1998, 2024-04-28:
Hey,
yes you can configure what kind of notifications you wish to receive.
The options are available on the app settings page.
If you disable the progress notifications via the drop-down, it should disable the progress notification while the normal progress-bar notification remains unless you disable it too.

The reason for that is that it is not possible to show a snapshot/screenshot of the webcam alongside the progress bar. That's why some users requested that I allow them to select both kinds of progress notifications.

cyberjak2k, 2024-04-28:
Ok that makes sense yeah just looking through the options it wasn't as
clear how that worked. Thank you

Clon1998, 2024-04-28:
> Ok that makes sense yeah just looking through the options it wasn't as clear how that worked. Thank you
> […](#)

Yes...
I need to rework that and might even consider removing the "0%" notifications entirely as it is redundant to print starting...

spitfire, 2024-05-16:
Sicce I've updated to the latest commit today (was maybe 2-3 commits behind), I'm getting multiple notifications for each event (with no other configuration changes:
[image]
[image]
[image]

Clon1998, 2024-05-16:
@spitfire please try to reset the printer's device registry.
It can happen that the app registers your device multiple times for notifications.

This can be done at the very bottom of the printer editing page.

spitfire, 2024-05-17:
> @spitfire please try to reset the printer's device registry. It can happen that the app registers your device multiple times for notifications.
> 
> This can be done at the very bottom of the printer editing page.

tried that, started a print just not, notifications are still tripled.

Clon1998, 2024-05-17:
> > @spitfire please try to reset the printer's device registry. It can happen that the app registers your device multiple times for notifications.
> 
> > 
> 
> > This can be done at the very bottom of the printer editing page.
> 
> 
> 
> tried that, started a print just not, notifications are still tripled.

Please provide me the mobileraker.log file.

spitfire, 2024-05-18:
It was still doing this after I've done that, but it eventually went back to normal ¯\_(ツ)_/¯

Clon1998, 2024-05-18:
> It was still doing this after I've done that, but it eventually went back to normal ¯_(ツ)_/¯

Maybe the companion was running as two separate instances, and one was terminated after a restart or similar. This happened to me as well! In any case, if the issue occurs again, please let me know and provide the log files!

Clon1998, 2024-07-03:
I'll close this. If the issue occurs again, feel free to reopen/create a new issue with the log files.

closed 2024-07-03 by Clon1998