Clon1998/mobileraker_companion #90: Installation Failed!

Gamerou, 2024-05-01:
I have tried it with both kiauh and the installer, I get the same error:

`Installer failed - invalid literal for int() with base 10: '7125                         # Port, auf dem der Server lauscht'

Stack Trace:
Traceback (most recent call last):
  File "/home/pi/mobileraker_companion/installer/Installer.py", line 20, in Run
    self._run()
  File "/home/pi/mobileraker_companion/installer/Installer.py", line 123, in _run
    configure.run(context)
  File "/home/pi/mobileraker_companion/installer/Configure.py", line 96, in run
    self._discover_moonraker_port(context)
  File "/home/pi/mobileraker_companion/installer/Configure.py", line 117, in _discover_moonraker_port
    port = config.getint("server", "port", fallback=None)
  File "/usr/lib/python3.9/configparser.py", line 818, in getint
    return self._get_conv(section, option, int, raw=raw, vars=vars,
  File "/usr/lib/python3.9/configparser.py", line 808, in _get_conv
    return self._get(section, conv, option, raw=raw, vars=vars,
  File "/usr/lib/python3.9/configparser.py", line 803, in _get
    return conv(self.get(section, option, **kwargs))
ValueError: invalid literal for int() with base 10: '7125                         # Port, auf dem der Server lauscht'

Please contact our support team directly at help@mobileraker.com so we can help you fix this issue!`

I use a Raspi 4B 4GB Ram with Msainsailos as the image

Clon1998, 2024-05-01:
Hey,
thanks for reporting that issue.
It seems like there is an bug in the moonraker port detection if the port config in the moonraker.conf contains a comment.

I'll adjust my installer to ignore the comments, in the mean time please trz removing the comment behind the port number and trz again.

Regards

Gamerou, 2024-05-01:
jup, that fixed it!

thank you

Gamerou, 2024-05-01:
Oh and one more thing I noticed: If you have a path in the domain in the mobile app, this is not taken into account in the webcam. For example, if the domain is https://example.com/printer, then the app tries to retrieve the webcam at https://example.com/webcam/?action=stream.

Clon1998, 2024-05-01:
> Oh and one more thing I noticed: If you have a path in the domain in the mobile app, this is not taken into account in the webcam. For example, if the domain is https://example.com/printer, then the app tries to retrieve the wecbcam at https://example.com/webcam/?action=stream, which should be https://example.com/printer/webcam/?action=stream.

That is intentional. Changing it now would potentially break to many running setups. You can always use an absolute url for the webcam tho.

closed 2024-05-01 by Gamerou