Clon1998/mobileraker_companion #91: Installer failing on newest firmware Creality K1 Max

scutzi128, 2024-05-09:
Getting this error during install and the companion never gets correctly installed.  Anyone else running into this or have a solution?

[image]

Clon1998, 2024-05-10:
Hey thanks for letting me know,
I will have a look at that. Seems like an issue with PIP in general.

scutzi128, 2024-05-10:
Thank you very much!

Shorezino, 2024-05-11:
First of all thank you for your time and help with this incredible app. I am also having same issue. My issueALSO occurs when trying to install octoeverywhere or moon raker time lapse. Seems like any plugins do this. Not sure if this is helpful information. I bought two brand new ender 3 v3 ke yesterday and tried this to no success after doing the same successfully to my other 2 KEs a few weeks ago. Let me know if I can do anything or test anything that might help. Again thank you

Shorezino, 2024-05-11:
Also i reset machine on the printer as well as the ssh terminal helper script reset option. Then tried the previous firmware with the same results

Clon1998, 2024-05-11:
It makes sense that the issue also occurs for other tools that use pip for dependency management.

There is the possibility that creality uses an outdated version of pip with an expired certificate.
Probably updating pip might solve the problem according to some quick Google searches about the pip problem in general.

If I have some time later I'll have a look to see if the issue also occurs on my test K1.

Shorezino, 2024-05-11:
Understood thank you for response. I'm not much of programmer I'm just following the guides and instructions. I couldn't figure out your update pip after doing some quick google searches myself. Anyways thank you. I subscribed to the paid version after seeing your quick response

Guilouz, 2024-05-11:
@Clon1998 pip 19.3.1 from /usr/lib/python3.8/site-packages/pip (python 3.8)

Clon1998, 2024-05-11:
> @Clon1998 pip 19.3.1 from /usr/lib/python3.8/site-packages/pip (python 3.8)

That's the current version on the K1?

Guilouz, 2024-05-11:
> > @Clon1998 pip 19.3.1 from /usr/lib/python3.8/site-packages/pip (python 3.8)
> 
> That's the current version on the K1?

Yes. After updating with entware: pip 23.3.1 from /opt/lib/python3.11/site-packages/pip (python 3.11)

Guilouz, 2024-05-11:
But same issue with updated pip version:

[image]

Clon1998, 2024-05-11:
> But same issue with updated pip version:
> 
> <img alt="Capture d’écran 2024-05-11 à 23 16 33" width="1060" src="https://private-user-images.githubusercontent.com/12702322/329813370-61befeef-0a6e-4996-8c09-77fd527c23e8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU0NjI1NTUsIm5iZiI6MTcxNTQ2MjI1NSwicGF0aCI6Ii8xMjcwMjMyMi8zMjk4MTMzNzAtNjFiZWZlZWYtMGE2ZS00OTk2LThjMDktNzdmZDUyN2MyM2U4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTExVDIxMTczNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhMGQ1ZWQ4YTlmNjg4ZGYyZTc5NzA3OGQ5OTgzZjM1NWUyMzU4NzQzZmY4NmNmODRkM2IxZGU1Y2JmOWFjMzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wJS-3P2Bna01lLHhAqpyRLhAHtRYyKmmhNxFzS2saF0">

Mhhhh...
It is actually working fine on my K1... with `pip 19.3.1 from /usr/lib/python3.8/site-packages/pip (python 3.8)`.

Guilouz, 2024-05-11:
I actually use Ender-3 V3 that use the same OS and same issue.

Obico have actually a similar issue: 

[image]

Guilouz, 2024-05-11:
@Clon1998 Which firmware version do you use ?

Clon1998, 2024-05-11:
> I actually use Ender-3 V3 that use the same OS and same issue.
> 
> Obico have actually a similar issue:
> 
> <img alt="Capture d’écran 2024-05-11 à 23 07 20" width="1055" src="https://private-user-images.githubusercontent.com/12702322/329813791-92d3fa4a-2ef2-4e16-9ada-947026827c9d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU0NjMzMTksIm5iZiI6MTcxNTQ2MzAxOSwicGF0aCI6Ii8xMjcwMjMyMi8zMjk4MTM3OTEtOTJkM2ZhNGEtMmVmMi00ZTE2LTlhZGEtOTQ3MDI2ODI3YzlkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTExVDIxMzAxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkMDg3YzBlOGE4ZTc5ODI5ODNmM2FjOWEzMzk5MDljNWU4NDIzYTZjNGU2ODZmYjQ4MGVkNzQyMTUzZWY4NGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.H9DydlCdSU8kg8bLcK1a-LZkV07oRQZ2eTcMThiBJ34">

Yeah thats to be expected as this problem is not caused by our installers but by PIP itself.

I wonder what the certificate looks like that is retrieved from "https://pypi.python.org"

Clon1998, 2024-05-11:
Can anyone get me the x508 cert from pypi.python.org via their printer using openssl or curl?
I just checked it via my browser and it is valid. However, using [SSL-Checker](https://www.sslshopper.com/ssl-checker.html#hostname=pypi.python.org) it states that the current cert became valid on the 9th of May. So I suspect that there is one that expired...

Shorezino, 2024-05-11:
Install failing on 1.1.0.12 and 1.1.0.13 and pre rooted 1.1.0.9 (destinal). However my other two V3 KEs installed fine a month or so ago on 1.1.0.12 and still running great after it upgraded to 1.1.0.13. I'm not much for this stuff but just trying to contribute.

Shorezino, 2024-05-11:
> Can anyone get me the x508 cert from pypi.python.org via their printer using openssl or curl?
>
> I just checked it via my browser and it is valid. However, using [...]

I can help if I get some dummy level instructions. I have putty for ssh

Clon1998, 2024-05-11:
> Can anyone get me the x509 cert from pypi.python.org via their printer using openssl or curl? I just checked it via my browser and it is valid. However, using [SSL-Checker](https://www.sslshopper.com/ssl-checker.html#hostname=pypi.python.org) it states that the current cert became valid on the 9th of May. So I suspect that there is one that expired...

Clon1998, 2024-05-11:
Gosh... There is no openssl available on the K1 FW ?
Fml. Let me google what I can use instead...

Guilouz, 2024-05-11:
> Gosh... There is no openssl available on the K1 FW ? Fml. Let me google what I can use instead...

You means openssl-util | 3.0.13-1 ?

https://bin.entware.net/mipselsf-k3.4/Packages.html

[openssl-util](https://bin.entware.net/mipselsf-k3.4/openssl-util_3.0.13-1_mipsel-3.4.ipk)	3.0.13-1

Clon1998, 2024-05-11:
```bash
opkg install openssl
echo | openssl s_client -connect pypi.python.org:443 -servername pypi.python.org 2>/dev/null | openssl x509 -noout -text
```
Should give me the cert data

Shorezino, 2024-05-11:
login as: root
root@192.168.50.198's password:
root@F005-A79D /root [#] opkg install openssl
-sh: opkg: not found
root@F005-A79D /root [#] echo | openssl s_client -connect pypi.python.org:443 -s
ervername pypi.python.org 2>/dev/null | openssl x509 -noout -text
-sh: openssl: not found
root@F005-A79D /root [#] login as: root
root@F005-A79D /root [#] opkg install openssl
-sh: opkg: not found
root@F005-A79D /root [#] echo | openssl s_client -connect pypi.python.org:443 -s
ervername pypi.python.org 2>/dev/null | openssl x509 -noout -text
-sh: openssl: not found
root@F005-A79D /root [#]
Password:

Shorezino, 2024-05-11:
login as: root
root@192.168.50.206's password:
root@F005-5DB4 /root [#] opkg install openssl
Unknown package 'openssl'.
Collected errors:
 * opkg_install_cmd: Cannot install package openssl.
root@F005-5DB4 /root [#]
root@F005-5DB4 /root [#] opkg install openssl
Unknown package 'openssl'.
Collected errors:
 * opkg_install_cmd: Cannot install package openssl.
root@F005-5DB4 /root [#] echo | openssl s_client -connect pypi.python.org:443 -servername pypi.python.org 2>/dev/null | openssl x509 -noout -text

Clon1998, 2024-05-11:
Woopsie....

It should state
```bash
opkg install openssl-util
echo | openssl s_client -connect pypi.python.org:443 -servername pypi.python.org 2>/dev/null | openssl x509 -noout -text
```

(Changed `openssl` -> `openssl-util`.

Shorezino, 2024-05-11:
login as: root
root@192.168.50.198's password:
root@F005-A79D /root [#] opkg install openssl-util
-sh: opkg: not found
root@F005-A79D /root [#] echo | openssl s_client -connect pypi.python.org:443 -s
ervername pypi.python.org 2>/dev/null | openssl x509 -noout -text
-sh: openssl: not found
root@F005-A79D /root [#] opkg install openssl-util
-sh: opkg: not found
root@F005-A79D /root [#] echo | openssl s_client -connect pypi.python.org:443 -s
ervername pypi.python.org 2>/dev/null | openssl x509 -noout -text
-sh: openssl: not found
root@F005-A79D /root [#]

i could be doing something wrong im not very good at this jus trying to help. 

i sshed thorugh putty. login:root password:Creality2023 (per my screen on v3 KE)

Shorezino, 2024-05-11:
fml i resest this printer let me try again

Shorezino, 2024-05-11:
ok check this out 

login as: root
root@192.168.50.198's password:
3root@F005-A79D /root [#] opkg install openssl-util
Installing openssl-util (3.0.13-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/openssl-util_3.0.13-1_mipsel-3.4.ipk
Installing zlib (1.3.1-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/zlib_1.3.1-1_mipsel-3.4.ipk
Installing libatomic (8.4.0-11) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libatomic_8.4.0-11_mipsel-3.4.ipk
Installing libopenssl (3.0.13-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libopenssl_3.0.13-1_mipsel-3.4.ipk
Installing libopenssl-conf (3.0.13-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libopenssl-conf_3.0.13-1_mipsel-3.4.ipk
Configuring libatomic.
Configuring zlib.
Configuring libopenssl.
Configuring libopenssl-conf.
Configuring openssl-util.
root@F005-A79D /root [#] echo | openssl s_client -connect pypi.python.org:443 -s
ervername pypi.python.org 2>/dev/null | openssl x509 -noout -text

Clon1998, 2024-05-11:
> ok check this out
>
> login as: root root@192.168.50.198's password: 3root@F005-A79D /root [#] opkg install openssl-util Installing openssl-util (3.0.13-1) to root... Downloading [...]

Mhhh👀 the output of the command is missing?

It should look something like:
[image]

I am primarily interested in the red boxed values haha

Clon1998, 2024-05-11:
@Guilouz do you mind giving the openssl-util a try ?

Shorezino, 2024-05-11:
login as: root
root@192.168.50.198's password:
root@F005-A79D /root [#] opkg install openssl-util
Package openssl-util (3.0.13-1) installed in root is up to date.
root@F005-A79D /root [#] echo | openssl s_client -connect pypi.python.org:443 -s
ervername pypi.python.org 2>/dev/null | openssl x509 -noout -text
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            04:7f:f8:9d:9e:6d:98:d7:23:93:9a:bc:4a:0c:62:b6:4a:5e
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = US, O = Let's Encrypt, CN = R3
        Validity
            Not Before: May  9 11:43:12 2024 GMT
            Not After : Aug  7 11:43:11 2024 GMT
        Subject: CN = python.org
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                Public-Key: (2048 bit)
                Modulus:
                    00:de:88:d5:b1:8c:dc:96:f0:38:f4:da:e6:16:1d:
                    16:6d:13:c6:4c:fd:3e:97:92:ba:fb:49:a7:de:86:
                    28:57:06:07:7d:22:45:9d:c4:9a:28:e0:1c:63:f7:
                    b7:c2:9b:69:f8:3d:ba:7b:b6:17:78:0d:3b:38:2b:
                    a6:28:e2:8f:86:f7:93:0d:6c:c6:7b:e6:00:bb:69:
                    84:8a:51:06:66:f7:c0:7a:b4:1b:7e:82:d8:57:79:
                    69:84:dc:e6:e2:7c:e2:ab:14:c7:5d:3f:ca:13:6a:
                    9d:9c:af:10:b9:42:c9:0a:e7:11:64:60:d4:cd:0d:
                    36:ad:7c:a4:85:9d:c6:18:5c:7d:19:87:a6:f0:82:
                    66:1d:76:49:1a:82:23:8c:97:68:5e:31:ef:e7:37:
                    78:7f:e2:8e:b7:d9:b9:9c:2d:62:93:e3:31:29:bb:
                    d0:52:06:92:ac:9c:51:35:eb:43:7c:78:53:58:4a:
                    05:3c:f1:6d:15:e4:1b:e6:3f:44:38:59:30:65:c5:
                    fb:37:7f:74:4e:98:54:b3:04:00:d9:2f:75:8b:e1:
                    21:09:1c:4c:2f:f1:f6:17:f6:6c:1a:fc:aa:22:ae:
                    72:20:12:15:f6:e1:be:d3:89:0a:c6:eb:4a:ef:75:
                    d1:18:95:09:89:ca:c8:c3:7c:44:fd:77:a6:f3:a7:
                    b2:09
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            X509v3 Key Usage: critical
                Digital Signature, Key Encipherment
            X509v3 Extended Key Usage:
                TLS Web Server Authentication, TLS Web Client Authentication
            X509v3 Basic Constraints: critical
                CA:FALSE
            X509v3 Subject Key Identifier:
                9A:7C:D8:7B:D2:4C:4F:C7:C7:17:1D:1C:71:D8:C2:7E:53:54:73:74
            X509v3 Authority Key Identifier:
                14:2E:B3:17:B7:58:56:CB:AE:50:09:40:E6:1F:AF:9D:8B:14:C2:C6
            Authority Information Access:
                OCSP - URI:http://r3.o.lencr.org
                CA Issuers - URI:http://r3.i.lencr.org/
            X509v3 Subject Alternative Name:
                DNS:*.python.org, DNS:python.org
            X509v3 Certificate Policies:
                Policy: 2.23.140.1.2.1
            CT Precertificate SCTs:
                Signed Certificate Timestamp:
                    Version   : v1 (0x0)
                    Log ID    : 48:B0:E3:6B:DA:A6:47:34:0F:E5:6A:02:FA:9D:30:EB:
                                1C:52:01:CB:56:DD:2C:81:D9:BB:BF:AB:39:D8:84:73
                    Timestamp : May  9 12:43:12.914 2024 GMT
                    Extensions: none
                    Signature : ecdsa-with-SHA256
                                30:45:02:21:00:91:9D:03:80:64:4E:0D:4F:D3:05:07:
                                A1:F4:D9:62:5A:FF:03:FF:07:BD:4A:C2:DC:08:C7:F9:
                                9D:E8:BF:D6:38:02:20:27:D7:99:D9:7D:F3:4E:EA:59:
                                73:74:E7:9B:B9:DF:51:B5:DD:BE:A8:1E:0A:85:F9:BC:
                                2A:63:1B:7A:D1:24:22
                Signed Certificate Timestamp:
                    Version   : v1 (0x0)
                    Log ID    : 76:FF:88:3F:0A:B6:FB:95:51:C2:61:CC:F5:87:BA:34:
                                B4:A4:CD:BB:29:DC:68:42:0A:9F:E6:67:4C:5A:3A:74
                    Timestamp : May  9 12:43:12.970 2024 GMT
                    Extensions: none
                    Signature : ecdsa-with-SHA256
                                30:45:02:21:00:89:D1:50:BB:AD:ED:95:4D:4F:98:DE:
                                82:50:A8:59:53:83:6B:BB:CB:3C:CC:FC:C7:65:5C:1E:
                                E4:2E:1A:8D:2B:02:20:28:DA:0D:FA:3E:CB:B8:A5:4C:
                                72:FD:24:5A:D7:FC:17:5F:1B:8C:10:F9:9B:55:96:13:
                                53:55:8B:DF:70:FF:44
    Signature Algorithm: sha256WithRSAEncryption
    Signature Value:
        7b:61:42:73:e2:3c:1e:f7:d5:68:09:70:c0:19:76:f8:20:d9:
        26:dc:72:60:d7:38:c7:50:15:b5:3f:db:03:59:17:3f:44:1f:
        d1:5e:21:00:c9:a8:47:00:81:6d:0f:40:08:60:ae:97:58:ee:
        9f:b8:6b:41:2b:5c:a1:31:14:7e:8f:ce:8e:fe:ae:57:a8:95:
        1f:df:7e:f0:8a:ce:19:2e:21:f9:02:4d:88:f8:65:58:2b:42:
        ac:0b:6c:c0:e0:7d:d7:c8:3f:be:d1:8d:16:af:ad:c3:d3:d0:
        bd:ff:f1:6d:c0:e7:4c:f9:de:74:c2:64:67:4f:fa:5a:fc:38:
        a7:29:d4:48:25:1f:e8:f4:37:6b:d9:e5:2b:ff:d7:de:79:7b:
        76:45:b0:34:81:56:74:f9:39:f6:8a:29:f7:b9:f6:52:d1:5f:
        5c:1d:66:af:13:93:9d:3b:99:0f:8b:fd:f2:7b:c1:32:fc:84:
        af:71:5c:14:a6:b4:bc:81:d3:a1:2f:d7:3e:03:e6:a8:b2:e0:
        d7:7f:d8:b6:fe:e9:cf:02:ea:a1:6b:da:c7:6f:23:f1:bb:80:
        30:9b:0b:48:da:94:f3:cc:01:99:84:4d:fd:8e:06:e4:5c:27:
        67:3f:d8:fe:be:3e:f2:ec:1d:0d:29:d6:7f:ac:db:eb:8e:67:
        d8:5b:04:cd

Clon1998, 2024-05-11:
> login as: root root@192.168.50.198's password: root@F005-A79D /root [#] opkg install openssl-util Package openssl-util (3.0.13-1) installed in root is up to date. root@F005-A79D /root [#] echo | [...]

THats the same cert.
Can you re run the installer can provide the error message once again?

Shorezino, 2024-05-11:
Using the helper script any plug in I now try to installs automatically installs SimplyPrint no matter which option I type in. I tried 12 for octoeverywhere and 15 for mobiler companion. It actually installed simply print also. Rebooted printer same issue now

Clon1998, 2024-05-11:
> Using the helper script any plug in I now try to installs automatically installs SimplyPrint no matter which option I type in. I tried 12 for octoeverywhere and 15 for mobiler companion. It actually installed simply print also. Rebooted printer same issue now

That might be something for @Guilouz. Installing openssl should not break that lol.

Shorezino, 2024-05-11:
> That might be something for @Guilouz. Installing openssl should not break that lol.

Can I uninstall now and try again?

Clon1998, 2024-05-11:
> Can I uninstall now and try again?

Uninstall the script or openssl?
You should be able to do that but Guilouz might be the better person to talk to about that.

Shorezino, 2024-05-11:
Right on well I'm not sure what else I can do right now. Thanks for your help. Trying not o get my new printers added for your app and octoeverywhere with companions

Clon1998, 2024-05-11:
> Right on well I'm not sure what else I can do right now. Thanks for your help. Trying not o get my new printers added for your app and octoeverywhere with companions 

Well adding it to the app should always work as long as there is the moonraker port available.

Just the push notifications require the companion.

Shorezino, 2024-05-11:
> Well adding it to the app should always work as long as there is the moonraker port available.
>
> Just the push notifications require the companion.

Cool I'll check that out I forgot about that. Thank you for your help. Hopefully we can revisit and get the companion going

Guilouz, 2024-05-11:
@Clon1998 Sorry i didn't follow the thread. What needed to test ?

Guilouz, 2024-05-11:
Same issue with openssl-util installed:

[image]

Shorezino, 2024-05-12:
I received this in the Octo everywhere GitHub. I started a similar conversation with them. "We are looking into this, it seems to be a problem with the date and time being set wrong. To work around it, try setting the date and time correctly."

I tried Sudo date reset and didn't work

Guilouz, 2024-05-12:
i have correct date and time and note working:

[image]

Guilouz, 2024-05-12:
When trying to install it outside environnement it's working:

[image]

Shorezino, 2024-05-12:
Can we implement this now? Guilloz is there a way we can support you

Guilouz, 2024-05-12:
@Clon1998 I have found a way to solve it.

I have installed all requierements manually in envirronement with this command : 

```/usr/data/mobileraker-env/bin/pip install --trusted-host pypi.python.org --trusted-host pypi.org --trusted-host=files.pythonhosted.org coloredlogs==15.0.1```

And re run installer and it's working:

[image]

With trusted host that do the job.

Shorezino, 2024-05-12:
right on. ill try it thank you. i see the helper script is currently disabled for some plugins. will i be able to perform this update then run helper script install? youre a gentleman and a scholar

Guilouz, 2024-05-12:
PR https://github.com/Clon1998/mobileraker_companion/pull/92 fix this issue.

closed 2024-05-13 by Clon1998

scutzi128, 2024-05-13:
Confirmed resolved for me. Thanks for all the help!

Shorezino, 2024-05-13:
> Confirmed resolved for me. Thanks for all the help!

Same thank you to clon and guilloz. I accidentally approved something last night idk if that messed with anything or even can mess with anything. New to this. Again thank you both so much.

Clon1998, 2024-05-13:
> > Confirmed resolved for me. Thanks for all the help!
> 
> Same thank you to clon and guilloz. I accidentally approved something last night idk if that messed with anything or even can mess with anything. New to this. Again thank you both so much.

No worries :)
Thanks again @Guilouz